espressif/esp-idf #11221: 保存音频

JuneSheng11, 2023-04-18:
### Is your feature request related to a problem?

使用双mic进行录音，希望可以保存双通道的原始音频，目前测试结果只是保存了单通道，如何才能保存双通道和回声消除中的参考信号的原始音频。

jason-mao, 2023-04-26:
@JuneSheng11  您使用的什么环境（IDF version 或其他的软件版本）？哪个example? 硬件codec或者开发板情况？

JuneSheng11, 2023-04-27:
4.4.4版本；esp-skainet中的vioce_communication;开发板是esp32-s3-Korvo-2的开发板

BlueSkyB, 2023-04-27:
vioce_communication中获取原始音频的地方是在feed_Task中，从I2S获取的是四通道音频，通过esp_get_feed_data接口后，会过滤成三通道音频。

JuneSheng11, 2023-04-28:
你好，这是我修改的代码，但是录制的音频很奇怪，存在以下情况：
1.  保存的原始音频并不是双通道的；
2.  录制的音频采样率不是16K;
我现在需要保存双麦原始音频，以及音频前端处理算法后的结果，特别是波束形成后的结果。我已将代码附上，您能帮忙修改一下吗

/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_wn_iface.h"
#include "esp_wn_models.h"
#include "dl_lib_coefgetter_if.h"
// #include "esp_afe_sr_iface.h"
#include "esp_afe_sr_models.h"
#include "esp_mn_iface.h"
#include "esp_mn_models.h"
#include "esp_board_init.h"
#include "driver/i2s.h"
#include "model_path.h"
#include "ringbuf.h"

#define DEBUG_SAVE_PCM      1  // 0
#if DEBUG_SAVE_PCM
#define FILES_MAX           3
ringbuf_handle_t rb_debug[FILES_MAX] = {NULL};
FILE * file_save[FILES_MAX] = {NULL};
#endif

int detect_flag = 0;
static esp_afe_sr_iface_t *afe_handle = NULL;
static esp_afe_sr_data_t *afe_data = NULL;
static volatile int task_flag = 0;  //0

void feed_Task(void *arg)
{
    esp_afe_sr_data_t *afe_data = arg;
    int audio_chunksize = afe_handle->get_feed_chunksize(afe_data);
    int nch = afe_handle->get_total_channel_num(afe_data);
    int feed_channel = esp_get_feed_channel();
    assert(nch <= feed_channel);
    int16_t *i2s_buff = malloc(audio_chunksize * sizeof(int16_t) * feed_channel);
    assert(i2s_buff);
    printf("channel %d\n",nch);
    while (task_flag) {
        esp_get_feed_data(i2s_buff, audio_chunksize * sizeof(int16_t) * feed_channel); // audio_chunksize 256 feed_channel 4

        afe_handle->feed(afe_data, i2s_buff);

    #if DEBUG_SAVE_PCM
        if (rb_bytes_available(rb_debug[0]) < audio_chunksize * nch * sizeof(int16_t)) {
            printf("ERROR! rb_debug[0] slow!!!\n");
        }

        rb_write(rb_debug[0], i2s_buff, audio_chunksize * nch * sizeof(int16_t), 0);
    #endif
    }
    if (i2s_buff) {
        free(i2s_buff);
        i2s_buff = NULL;
    }
    vTaskDelete(NULL);
[... 98 lines cut ...]
        printf("create_from_config fail!\n");
        return;
    }

#if DEBUG_SAVE_PCM
   // rb_debug[0] = rb_create(afe_handle->get_total_channel_num(afe_data) * 4 * 16000 * 2, 1);   // 4s ringbuf
    rb_debug[0] = rb_create(afe_handle->get_total_channel_num(afe_data) * 4 * 16000 * 2, 1);   // 4s ringbuf
    file_save[0] = fopen("/sdcard/feed_1.pcm", "w");
    if (file_save[0] == NULL) printf("can not open file\n");
    
	rb_debug[1] = rb_create(afe_handle->get_total_channel_num(afe_data) * 4 * 16000 * 2, 1);   // 4s ringbuf
    file_save[1] = fopen("/sdcard/feed_2.pcm", "w");
    if (file_save[1] == NULL) printf("can not open file\n");
	
    rb_debug[2] = rb_create(1 * 4 * 16000 * 2, 1);   // 4s ringbuf
    file_save[2] = fopen("/sdcard/fetch6.pcm", "w");
    if (file_save[2] == NULL) printf("can not open file\n");

    xTaskCreatePinnedToCore(&debug_pcm_save_Task, "debug_pcm_save", 2 * 1024, NULL, 5, NULL, 1);
#endif

    task_flag = 1;
    xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void*)afe_data, 5, NULL, 0);
    xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void*)afe_data, 5, NULL, 1);
}

BlueSkyB, 2024-09-25:
voice_communication 并不经过 BSS，speech recognition 那边，当双麦时，才会经过 BSS。
所以需要将 
`afe_handle = (esp_afe_sr_iface_t *)&ESP_AFE_VC_HANDLE;`
改为:
`afe_handle = (esp_afe_sr_iface_t *)&ESP_AFE_SR_HANDLE;`

同时将音频通道改为:
`afe_config.pcm_config.total_ch_num = 3;
afe_config.pcm_config.mic_num = 2;
afe_config.pcm_config.ref_num = 1;`

因为 skainet 中默认的korvo 2板子驱动，获得音频是有回踩信号的。这里的配置需要与实际音频相匹配。如果是自己的音频驱动，配置也需要与喂入AFE的实际音频情况相匹配。
fetch 获取的音频是算法处理后的单通道音频。要获取原始音频，保存喂给 afe 的音频。