espressif/esp-idf #11224: build active configuration taking long time

nileshkoladiya, 2023-04-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, 
My  build process stuck for infinite time with below status. i have to terminate IDE from task manager and reopen the ESP IDE.

[image]

igrr, 2023-04-19:
@nileshkoladiya which version of  the IDE you are using? If that is 2.9.0, please  upgrade to  2.9.1, the issue has been fixed in that release. Otherwise please report the issue at https://github.com/espressif/idf-eclipse-plugin, adding the information requested there in the issue template.

nileshkoladiya, 2023-04-20:
Hello igrr,
i update my IDE and now that problem is resolved. 
Thanks for your support.

Regard,
Nilesh koladiya

closed 2023-04-20 by igrr