espressif/esp-idf #11226: BLE connection problem with gatt_security_server esp32s3 and Android 13

twomas, 2023-04-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Install app nRF Connect in an Android 13 phone. I use a Google pixel.
Build and install gatt_security_server for ESP-IDF v4.4.4 for ESP32-S3-DevKitC-1 v1.1
First connection is fine from nRF Connect  but then connect problem.

I (211567) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (211817) SEC_GATTS_DEMO: remote BD_ADDR: 51031485e53b
I (211827) SEC_GATTS_DEMO: address type = 1
I (211827) SEC_GATTS_DEMO: pair status = success
I (211827) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_BOND
I (211837) SEC_GATTS_DEMO: Bonded devices number : 1

I (211837) SEC_GATTS_DEMO: Bonded devices list : 1

I (211847) SEC_GATTS_DEMO: 51 03 14 85 e5 3b
W (211917) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3d
I (211917) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x3d
I (211927) SEC_GATTS_DEMO: advertising start success
I (226217) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (226467) SEC_GATTS_DEMO: remote BD_ADDR: 51031485e53b
I (226477) SEC_GATTS_DEMO: address type = 1
I (226477) SEC_GATTS_DEMO: pair status = success
I (226477) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_BOND
I (226487) SEC_GATTS_DEMO: Bonded devices number : 1

I (226487) SEC_GATTS_DEMO: Bonded devices list : 1

I (226497) SEC_GATTS_DEMO: 51 03 14 85 e5 3b
W (226567) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3d
I (226567) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x3d
I (226577) SEC_GATTS_DEMO: advertising start success
I (303317) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (303567) SEC_GATTS_DEMO: remote BD_ADDR: 51031485e53b
I (303567) SEC_GATTS_DEMO: address type = 1
I (303567) SEC_GATTS_DEMO: pair status = success
I (303577) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_BOND
I (303587) SEC_GATTS_DEMO: Bonded devices number : 1

I (303587) SEC_GATTS_DEMO: Bonded devices list : 1

I (303597) SEC_GATTS_DEMO: 51 03 14 85 e5 3b
W (303667) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3d
I (303667) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x3d
I (303677) SEC_GATTS_DEMO: advertising start success

See also attached Android 13 logs via adb. Maybe log "LogMsg: btif_gatt_set_encryption_cb() - Encryption failed (1)" gives a hint

[Android13-logs.txt](https://github.com/espressif/esp-idf/files/11271423/Android13-logs.txt)

ztefanjo, 2023-04-24:
@xiewenxiang : Would it be possible for you to provide a patch for this for ESP-IDF v4.4.4? It is a severe error that we would like to have solved ASAP.

esp-cjh, 2023-05-08:
Hi @ztefanjo 
Apply the patch in components/bt/controller/lib_esp32c3_family/.
[0001-fix-ble-mic-err-v4.4.patch](https://github.com/espressif/esp-idf/files/11419871/0001-fix-ble-mic-err-v4.4.patch)
[0001-fix-ble-mic-err-v4.3.patch](https://github.com/espressif/esp-idf/files/11419874/0001-fix-ble-mic-err-v4.3.patch)

ztefanjo, 2023-05-08:
Hi @esp-cjh ,
Thanks. Would it be possible for you to also provide patches on top of our "co-exist SDK" that we are using? It appears that they change in the same lib file.
https://glab.espressif.cn/technical_support/husqvarnagroup/esp-idf-patch/commits/master

So, we would require a patch on b6701d79af943af2a5ce4be4a36e6554831f1404 (v4.4.4) and a patch on d9d4c09406a6f54fdbcb92322c5f39e208a51d13 (v4.4.3).

Perhaps we need updated versions of the "co-exist SDK"?

esp-cjh, 2023-05-10:
OK

esp-cjh, 2023-11-20:
This issue will be closed and that they can reopen anytime.

ztefanjo, 2023-11-20:
Yes, this issue is solved and can be closed.

closed 2023-11-21 by Alvin1Zhang