espressif/esp-idf #11227: Unable to use UART0 pins (Rx/GPIO20) in ESP32-C3-DevKitM-1 when powered with out USB cable

aggaddam, 2023-04-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

**Hardware: ESP32-C3-DevKitM-1**

I am configuring GPIO20 as RX and GPIO21 as TX pins of the UART1 port. Everything works fine if the hardware ESP32-C3-DevKitM-1 is powered with a Micro USB port. Data has been received without any issues on both Tx & Rx pins.

The issue happens when the hardware ESP32-C3-DevKitM-1 is powered either with 3.3V and GND or 5V and GND. In this case, I am not receiving data on the RX pin ie. GPIO20 but tx pin works fine. Any suggestions?

andylinpersonal, 2023-04-23:
Hi @aggaddam:

There might be two problems in your configuration. 

1. It seems like your on-board CP2102N usb-uart bridge is interfering your connection. 

See [1], VBUS pin is connected to CP2102N, and blocked by the diode D1 from the net VCC_5V. See [2] p.28, VBUS signal is required to indicate a USB connection. 
If you power from the pin header (VCC_5V or VCC_3V3), VBUS does not connect to any 5V signal and is pulled low to GND by [1] R18, the USB connection is not established (and CP2102N is under an undefined state). If you want to power your board and connect to U0RXD and U0TXD externally, you may have to desolder R1 and R4 to prevent the interference from CP2102N.

2. You cannot enter download mode by connecting to UART0 alone. 

CP2102N expose two extra signals, RTS and DTR. Espressif use these two signals to implement an auto-program circuit ([1] R10, R12, Q1, Q2) to toggle CHIP_PU and GPIO9 automatically.

So you may have to
a. Implement this circuit by yourself.
b. Flashing your board from the USB Serial/JTAG (GPIO 18/19) instead.
c. Enter download mode manually by pressing boot button and then reset button before flashing your board from UART0.

[1] https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITM-1_V1_20200915A.pdf
[2] https://www.silabs.com/documents/public/data-sheets/cp2102n-datasheet.pdf

aggaddam, 2023-04-24:
@andylinpersonal Thanks for the update.

Regarding point b - Flashing your board from the USB Serial/JTAG (GPIO 18/19) instead. 

It was mentioned only U0TXD and U0RXD allow the serial download mode. Is it possible to use (GPIO 18/19) to flash the firmware on to the board?

andylinpersonal, 2023-04-24:
@aggaddam 

Yes it’s perfectly ok. 

If you connected via GPIO18/19 to the computer via a breakout cable, your PC would show up a virtual COMx/ttyACMy port. This is the CDC-ACM component of USB Serial/JTAG Controller. You can just flash into that port using `idf.py flash -p PORT`

[1] https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html