espressif/esp-idf #11231: `idf_tools.py install` / `install.sh` help is broken.

AccidentallyTheCable, 2023-04-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

`install.sh`command help is broken. Executing `install.sh -h` or `install.sh --help` gives the following error:

```
idf_tools.py install: error: argument -h/--help: ignored explicit argument ','
```

`./tools/idf_tools.py install -h` does work correctly though.

### What is the actual behavior?

Expected to see same help output as `./tools/idf_tools.py install -h`

### Steps to reproduce.

1. Download Fresh copy of esp-idf (v5.0.1)
2. Per instructions, use `install.sh`, but check the help, with `./install.sh -h` or `./install.sh --help`

### Build or installation Logs.

```plain
$: ./install.sh -h
Detecting the Python interpreter
Checking "python3" ...
Python 3.9.2
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
usage: idf_tools.py install [-h] [--targets TARGETS] [TOOL ...]
idf_tools.py install: error: argument -h/--help: ignored explicit argument ','
```
```

### More Information.

No build done, because this is at tool installation steps.

dobairoland, 2023-04-20:
Hi @AccidentallyTheCable. This is not broken. It was never supported. 

The [Get Starting Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html) demonstrates the use of `install.sh`. The documentation about `idf_tools.py` can be found [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html). I hope this helps.

dobairoland, 2023-04-20:
I've marked this as feature request. We plan to address this in the future.

closed 2023-05-21 by espressif-bot