espressif/esp-idf #11232: Wrong condition

eiffel31, 2023-04-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/btc/profile/std/battery/battery_prf.c

This condition is useless as is it always true:
        if (char_uuid == GATT_UUID_SYSTEM_ID | GATT_UUID_MODEL_NUMBER_STR | GATT_UUID_PNP_ID |
                GATT_UUID_SERIAL_NUMBER_STR | GATT_UUID_FW_VERSION_STR | GATT_UUID_HW_VERSION_STR |
                GATT_UUID_SW_VERSION_STR | GATT_UUID_MANU_NAME | GATT_UUID_IEEE_DATA) {
            switch (char_uuid) { [...]

Note 1: bitwise or operator has lower priority than == operator
Note 2: adding parenthesis is not a solution because test will be always false.

Suggested change:
 if (char_uuid == GATT_UUID_SYSTEM_ID | char_uuid == GATT_UUID_MODEL_NUMBER_STR |
 char_uuid == GATT_UUID_PNP_ID | [...]

eiffel31, 2023-04-22:
More thoughts: is there any interest in keeping this condition? Why not just removing the full condition and adding an empty default case to the switch?

Weijian-Espressif, 2023-04-22:
@eiffel31 Thank you for the reminder, I will fix this error as soon as possible, thank you

Weijian-Espressif, 2023-05-11:
@eiffel31  I have fixed this issue, please update to the latest master and retest

eiffel31, 2023-05-11:
Hello @Weijian-Espressif , Thank you for the commit.

On more change you can do in the very same part: replace the "if" by a new case in the switch.

if (char_uuid == GATT_UUID_BATTERY_LEVEL) {
     bas_AddCharDescr(service_id, p_data->add_result.attr_id);
}

switch (char_uuid) {
case GATT_UUID_SYSTEM_ID:
    dis_cb.dis_attr[0].handle = service_id; break;
case GATT_UUID_MODEL_NUMBER_STR:

**Replace by**
switch (char_uuid) {
case GATT_UUID_BATTERY_LEVEL:
     bas_AddCharDescr(service_id, p_data->add_result.attr_id); break;
case GATT_UUID_SYSTEM_ID:
    dis_cb.dis_attr[0].handle = service_id; break;
case GATT_UUID_MODEL_NUMBER_STR:

closed 2023-08-10 by Alvin1Zhang