espressif/esp-idf #11237: Ability to switch between idf versions like python virtual environments

moefear85, 2023-04-20:
### Is your feature request related to a problem?

Over time I've accumulated many projects, each compiled/linked to different esp versions/commits. For some people, docker is not a solution, not just because of the learning curve or that it requires espressif to rebuild a new docker for every version. So switching between esp versions, especially with multiple simultaneously opened projects, is cumbersome.

### Describe the solution you'd like.

It would be nice if instead one could clone the esp-idf repo once, then locally simply checkout the version one currently wants to use (and create multiple temporary clones locally if one needs different versions/commits simultaneously).

Then the install script would install to the same IDF_TOOLS_PATH folder, but use the same dist & tools folders, and only installing new tool versions if they don't preexist. Likewise, each commit creates its own python virtual environment, and the environment includes the commit as part of its folder name, to differentiate. This way only the virtual environments are redundant.

Ideally it would be nice if different esp-idf commits were aware whether or not they need their own virtual environment, such that the virtual environment takes an additional special version number that indicates whether or not any libraries within have changed. Then each commit can determine based on this version along with the esp-idf version and python version, whether it can use a specific virtual environment or must create a new one. If this is too difficult to implement, then each esp-idf commit should simply look for its own python environment with a matching commit, and create it if it doesn't preexist.

For the Visual Studio Extension, it can have a menu in the status bar at the bottom, (in the same way how the python extension allows selecting different python interpreters hence virtual environments), that allows selecting which esp-idf version and commit one wants to use for the given project, and then searches within IDF_PATH and IDF_TOOLS_PATH for folders explicitly labelled with that commit. Then during setup of the extension, one can create multiple configurations, each having its own commit as the label or as part of the label of that configuration, which shows up then in the status bar.

### Describe alternatives you've considered.

Currently, I just clone multiple copies of the esp-idf repo, each having its commit as part of the folder name. Then I install each into its own espressif folder, also having its own commit in the folder name. Then I copy the VSCode folder, and give each folder copy a commit as a suffix. Then I create links from the currently needed commit as "esp-idf" "espressif" and "Code" respectively.
I wrote a python script to automate the switching. And when I switch to a new commit for the first time, I rerun the esp-idf extension configuration so it gets adjusted to its specific commit.

The downside of this approach, is that all these folders have to remain permanent, and the sizes of each folder is considerable, and there is a lot of redundancy. It allows me to use two or more different simultaneous commits or versions from the command line by setting the environment path variables differently for each, but the only practical way to work on a project is from within the extension, and I'm not aware of an easy way to make multiple instances of VSCode use different configuration folders, hence I can only use one specific commit/version at a time.

igrr, 2023-04-20:
@moefear85 could you please add, what issue are you facing with using a single copy of IDF, and using `git checkout` + `git submodule update` + `install.sh` + `export.sh` to switch?

moefear85, 2023-04-20:
I would then often face various compile errors in VSCode and end up having to reconfigure the extension on every switch. The setup menu of the extension also automatically popups even before I hit a compilation error, which is also problematic. I don't see why one has to choose a specific version of esp-idf, tools folder, and python environment if/since each of these and their paths is and can be selected per project in the .vscode folder. Instead the extension should only take a set of possibilities and remember them, then upon creation of a new project, it should ask which specific combination one wants for that project, and place those settings into the .vscode folder. 

Currently, every once in a while, I update the master branch. But then in VSCode, compilation often ends up failing, because between commits, some python packages have changed that aren't compatible, and usually running install.sh ends up overwriting the virtual environment, meaning the installed versions get changed on every switch. This means I need to redownload various packages on every switch, and I often don't have an internet connection at switching time.

I've never tried "git submodule switch", although I've tried the rest. Then when compiling an arbitrary project in VSCode, I either run into various errors (something about cmake and so on), or submodules get re-downloaded. Anything that would still need an internet connection everytime I switch back and forth between any two commits, is also not a solution from my perspective. Ideally everything should behave like python environments, where switching between them neither overwrites or changes anything, nor requires or performs redownloading/reconfiguration. It only updates links and environment variables where necessary so that different pre-setup configurations get used.

dobairoland, 2023-04-21:
Having multiple copies of ESP-IDF is not recommended. At least I often handle issues which are caused by multiple ESP-IDF installations interfering between each other. 

> This way only the virtual environments are redundant.

As pointed out, we would have also redundant ESP-IDF copies. So more redundancy as we currently have.

It the ESP-IDF is cloned with the full history (not a shallow copy, or single branch mode) then it is very fast to switch between versions. Activating the environment selects the right tools. Switching between them is also very fast (version from the ESP-IDF we are switching from doesn't have to uninstalled - next time will be available without reinstallation).

I'm sorry but the described switching mechanism seems very complex to me with a lot of potential errors.  I'm sorry, honestly I don't think that the proposed change would be an improvement over the current state. This statement is regarding the console-based ESP-IDF without any IDE.

dobairoland, 2023-04-21:
It looks like the described situation is more related to the usage of ESP-IDF with IDEs. @kolipakakondal Could you please take a look and see if we can or should improve anything regarding this?