espressif/esp-idf #11240: UUID inconsistency

eiffel31, 2023-04-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

BLE_UUID16_INIT : this macro requires an id in the form 0xabcd, where abcd are written in MSB order
BLE_UUID128_INIT : this macro requires a list of bytes written in LSB order

1) It is not documented that there is a danger here (or at least I was not able to find de documentation)
2) The UUID in the code should appear in the same way as in any official BLE documentation, that is in MSB order. Of course the message will be composed in LSB order. It is the work of the encoding macro/function to generate in respect of the documentation order. UUID128 has to be reversed as it is done for UUID16.

zhp0406, 2023-11-17:
Thank you for your feedback; we will improve the relevant documentation in the future.

esp-zhp, 2024-12-03:
fixed
bce1715dfdfe791ab950f7fcb94d537b74c90025 
f81286382722e67bbcafe1d994bb3bcfac35b437

closed 2024-12-17 by Alvin1Zhang