espressif/esp-idf #11242: GPIO42 1.7V  during deep sleep,

zbqxyz, 2023-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

GPIO42 1.7V  during deep sleep,

Alvin1Zhang, 2023-04-23:
Thanks for reporting, would you please help provide more details per [the issue template](https://github.com/espressif/esp-idf/issues/new/choose)? Thanks.

Alvin1Zhang, 2023-04-26:
Thanks for reporting, will close due to short of details, feel free to reopen with more updates.

closed 2023-04-26 by Alvin1Zhang