espressif/esp-idf #11244: Bluetooth: missing common definitions / dupplicated definitions

eiffel31, 2023-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In bluedroid, same codes are defined in 2 files.
Ex:
components/bt/host/bluedroid/api/include/api/esp_gatt_defs.h:
#define ESP_GATT_UUID_ALERT_NTF_SVC                 0x1811          /* Alert Notification Service*/
#define ESP_GATT_UUID_HID_SVC                       0x1812          /* HID Service*/
#define ESP_GATT_UUID_SCAN_PARAMETERS_SVC           0x1813          /* Scan Parameters Service*/
#define ESP_GATT_UUID_RUNNING_SPEED_CADENCE_SVC     0x1814          /* Running Speed and Cadence Service*/
...

components/bt/host/bluedroid/stack/include/stack/sdpdefs.h:
#define UUID_SERVCLASS_ALERT_NOTIFICATION       0x1811      /* alert notification service */
#define UUID_SERVCLASS_LE_HID                   0x1812     /*  HID over LE */
#define UUID_SERVCLASS_SCAN_PARAM               0x1813      /* Scan Parameter service */
#define UUID_SERVCLASS_RSC                      0x1814      /* RUNNERS SPEED AND CADENCE SERVICE      */
(note: I removed code for GLUCOSE, inserted out of order between 1813 and 1814!)

In nimble, there is no definition of standard values. They are defined many times with different names.
Ex:
components/bt/host/nimble/nimble/apps/blecent/src/blecent.h:
#define BLECENT_SVC_ALERT_UUID              0x1811
#define BLECENT_CHR_SUP_NEW_ALERT_CAT_UUID  0x2A47
#define BLECENT_CHR_NEW_ALERT               0x2A46
#define BLECENT_CHR_SUP_UNR_ALERT_CAT_UUID  0x2A48

components/bt/host/nimble/nimble/apps/bleprph/src/bleprph.h:
#define GATT_SVR_SVC_ALERT_UUID               0x1811
#define GATT_SVR_CHR_SUP_NEW_ALERT_CAT_UUID   0x2A47
#define GATT_SVR_CHR_NEW_ALERT                0x2A46
#define GATT_SVR_CHR_SUP_UNR_ALERT_CAT_UUID   0x2A48

components/bt/host/nimble/nimble/apps/btshell/src/btshell.h:
#define GATT_SVR_SVC_ALERT_UUID               0x1811
#define GATT_SVR_CHR_SUP_NEW_ALERT_CAT_UUID   0x2A47
#define GATT_SVR_CHR_NEW_ALERT                0x2A46
#define GATT_SVR_CHR_SUP_UNR_ALERT_CAT_UUID   0x2A48

components/bt/host/nimble/nimble/nimble/host/services/ans/include/services/ans/ble_svc_ans.h:
#define BLE_SVC_ANS_UUID16                                  0x1811
#define BLE_SVC_ANS_CHR_UUID16_SUP_NEW_ALERT_CAT            0x2a47
#define BLE_SVC_ANS_CHR_UUID16_NEW_ALERT                    0x2a46
#define BLE_SVC_ANS_CHR_UUID16_SUP_UNR_ALERT_CAT            0x2a48

examples/bluetooth/nimble/ble_spp/spp_client/main/ble_spp_client.h:
#define GATT_SVR_SVC_ALERT_UUID              0x1811
#define GATT_SVR_CHR_SUP_NEW_ALERT_CAT_UUID  0x2A47
#define GATT_SVR_CHR_NEW_ALERT               0x2A46
#define GATT_SVR_CHR_SUP_UNR_ALERT_CAT_UUID  0x2A48

examples/bluetooth/nimble/ble_spp/spp_server/main/ble_spp_server.h:
#define GATT_SVR_SVC_ALERT_UUID               0x1811
#define GATT_SVR_CHR_SUP_NEW_ALERT_CAT_UUID   0x2A47
#define GATT_SVR_CHR_NEW_ALERT                0x2A46
#define GATT_SVR_CHR_SUP_UNR_ALERT_CAT_UUID   0x2A48

examples/bluetooth/nimble/ble_spp/spp_server/main/gatt_svr.c:
#define BLE_SVC_ANS_UUID16                                  0x1811
#define BLE_SVC_ANS_CHR_UUID16_SUP_NEW_ALERT_CAT            0x2a47
#define BLE_SVC_ANS_CHR_UUID16_NEW_ALERT                    0x2a46
#define BLE_SVC_ANS_CHR_UUID16_SUP_UNR_ALERT_CAT            0x2a48

And so on for these files (we still have the same 4 example constant with different names):
examples/bluetooth/nimble/blecent/main/blecent.h
examples/bluetooth/nimble/bleprph/main/bleprph.h
examples/bluetooth/nimble/throughput_app/blecent_throughput/main/gattc.h
examples/system/ota/advanced_https_ota/main/ble_helper/include/nimble_gatts.h

This is just a small extract to show the problem on some constants.

Constants are duplicated everywhere, in components, in examples, sometimes in .h files, sometimes in .c files.
A common, centralized definition, should be built for all these values defined in the specifications.
The common definition is at bluetooth level, it is not specific to some stack. But building a definition common for the 2 stacks may be challenging, depending how development is organized.

rahult-github, 2023-04-30:
> In nimble, there is no definition of standard values. They are defined many times with different names. 

ESP-IDF nimble codebase is cloned from upstream mynewt-nimble. Generally, we try to keep the codebase in IDF as similar to one present in upstream for two reasons , quick upgration and easy migration for users who already are using upstream nimble and hence they need no / very less modification to their existing codebase to migrate. 

The newly developed examples though have carry forwarded the same nomenclature and hence this observation.

esp-cjh, 2023-11-20:
This issue will be closed and that they can reopen anytime.

closed 2023-11-21 by Indastri