espressif/esp-idf #11246: Not able to set the clock source as UART_SCLK_XTAL for uart 2 ,chip ESP32 S3

Ramakrishna247, 2023-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

const uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        //.rx_flow_ctrl_thresh = 122,
        //.source_clk = UART_SCLK_APB
        .source_clk = UART_SCLK_XTAL,
    };

It is giving an error identifier "UART_SCLK_XTAL" is undefined , Actually i wanted to use UART 2 in power save mode  using esp_pm_configure(&pm_config).   

The chip iam using is ESP32 S3 N16R8 module .  

Also I have a doubt  here "XTAL" means 40 MHZ crystal or external 32 kHZ crystal.  in uart_types.h  I see that UART_SCLK_XTAL is not enabled. is there any settings to enable this  XTAL in sdkconfig.h ?

suda-morris, 2023-04-23:
```c
    const int uart_num = UART_NUM_2;
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_XTAL,
    };
    uart_param_config(uart_num, &uart_config);
```
I didn't see compile or runtime issue with the above configuration on esp32s3. Please check your IDF version, and try again after the upgrade.

Ramakrishna247, 2023-04-23:
Hi Suda-morris,
Thanks for checking , Also can you also please confirm two things
1) Is XTAL means 40 MHZ crystal or 32khz external crystal?
2) can we use UART 2 with wifi power save example.?

songruo, 2023-04-24:
Hi @Ramakrishna247,

1. The "XTAL" is referring to the 40MHz crystal. 
2. I am not sure what are you trying to use UART 2 for in the power_save example. The UART modules are not able to function properly during light sleep, due to the lack of APB clock. If the UART port only needs to work in chip active mode, then I believe there should be no problem to use UART 2.

Ramakrishna247, 2023-04-24:
Hi @Songruo,

Thanks for the reply,
For UART , what we want to achieve is , we have another STM controller  , we want to have communication between STM and ESP32, we want to wake up ESP32 using GPIO or UART  of STM controller and  via server connected through wifi.

ESP32 is able to receive socket messages from server  by adding some code to wifi power save example. 

But for UART and GPIO the issue is ESP32 continuously switching between active and light sleep  and also it is using DFS. 
 so we are trying to understand how can we make use of UART or GPIO  of STM to move ESP32 to active mode and keep it in active mode until we want to put it back to sleep/automatic light sleep.

Is this possible ? if yes can you give us sample code or any document link that can help us.

Ramakrishna247, 2023-04-24:
Hi songruo,

I have added the gpio wake up code from light sleep example to wifi power save example and i see that "light_sleep_task" is being  called again and again.  
 Device is waking up with reason  timer again and again.  even though i haven't enabled timer wakeup. 
I have attached project folder to recreate the issue.  (For ESP32 S3 N16R8 )
[wifi_power_save_soc.zip](https://github.com/espressif/esp-idf/files/11310149/wifi_power_save_soc.zip)

songruo, 2023-04-24:
I believe this is due to the auto light sleep feature. FreeRTOS decides when to go into sleep, and sets a timer wakeup internally to wake up the system when it think it is the time to wakeup.

Also, it is probably not a proper way to have auto light sleep feature enabled and have a `while(1)` loop task which keeps fetching wakeup cause and printing to console. The chip can not really take the benefit of auto light sleep when there is such a `while(1)` loop. Such observation is indeed disturbing how the chip actually does.

If you want to know when the chip goes into sleep and when it wakes up, I think you can use a GPIO and observe its logic level with a logic analyzer or an oscilloscope. You can set an IO to HIGH initially. When the chip enters light sleep, the IO will switch to sleep configuration, so its level will be LOW; When the chip wakes up from sleep, the IO will get back to HIGH. In this way, you can get an idea of how auto light sleep actually works.

Then for the GPIO wakeup, you can also use the same method to verify whether it is working properly.

Ramakrishna247, 2023-04-24:
Got it , But my main concern is can we put device completely in to active mode if we put the device once in to automatic light sleep mode? 
 Can we run device to function at full cpu frequency and stop DFS ? , what i see is even UART1 is not working properly when DFS is enabled and i don't see any API to stop power management and DFS.

igrr, 2023-04-24:
You can acquire a [power management lock](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html#power-management-locks) for the duration of time when the ESP is communicating with the host.

For example, you could use the following design:
1. Host sets an IO to indicate that it wants to communicate
2. ESP wakes up from light sleep by GPIO
3. ESP handles the GPIO change by an interrupt handler
4. In the interrupt handler, set a semaphore to wake up the task which will do UART communication
5. The task should acquire the power management lock and then communicate over UART
6. When communication is over (e.g. the host sends a command to let the ESP go to sleep), release the power management lock and suspend the communication task on the semaphore again.

Ramakrishna247, 2023-04-24:
Ok Thanks for the detailed explanation , I will try to implement this.

Ramakrishna247, 2023-04-26:
Hi igrr,

Your suggestion was very helpful, once i acquire PM lock , UART is working fine , but there is one issue , that is for interrupt part internal pull down resistor is not working, even though i have enabled internal pull down resistor  interrupt is getting triggered again , i have made interrupt enabled for postive edge , until i connect the gpio to ground interrupt is getting triggered again and again ,  this issue is may be due to no clock source at the time of sleep ?

We have freezed our hardware i cannot use external pull down resistor  , is there a way to enable pull down resistor , i have tried rtc power domain ON  with esp_sleep_pd_config but no use.