espressif/esp-idf #11247: Frequent I2C transmit errors when I2S is activated

hyunel, 2023-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4726-gdf9310ada2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I2C transmits data without any errors

### What is the actual behavior?

When I2S channel is enabled, I2C transmission has frequent errors (and sometimes reads the wrong value)

### Steps to reproduce.

Here is my test code (If everything is correct, the value read by I2C will be 191):

```c
/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "driver/i2s_std.h"
#include "driver/i2c.h"
#include "esp_log.h"

const char *TAG = "main";

void app_main(void)
{
    ESP_LOGI(TAG, "Hello world!");
    
    /* Print chip information */
    esp_chip_info_t chip_info;
    uint32_t flash_size;
    esp_chip_info(&chip_info);
[... 335 lines cut ...]
E (4863) main: E: -1
E (4863) main: E: -1
E (4923) main: E: -1
W (4953) main: V: 127
E (4973) main: E: -1
E (4973) main: E: -1
E (5003) main: E: -1
E (5033) main: E: -1
E (5043) main: E: -1
W (5133) main: V: 127
E (5133) main: E: -1
E (5163) main: E: -1
W (5173) main: V: 127
W (5183) main: V: 127
E (5193) main: E: -1
E (5193) main: E: -1
W (5233) main: V: 127
E (5283) main: E: -1
```

### More Information.

 - When `i2s_channel_enable(tx_handle)` is commented out, then no error occurs.
 - Tried changing the I2S clock source, but it didn't work.
 - I found that turning down `mclk_multiple` (e.g. to 256) resulted in a significant reduction in error frequency.

closed 2023-04-21 by hyunel

toanadas1, 2023-12-27:
Hi @hyunel 
I'm encountering the same issue with I2S and I2C, could you please point out at which commit this bug has been fixed?

hyunel, 2023-12-27:
> Hi @hyunel I'm encountering the same issue with I2S and I2C, could you please point out at which commit this bug has been fixed? ESP-IDF: v5.1.1

This was supposed to be my pcb design problem, which I finally solved by adding I2C pull-up resistors.

toanadas1, 2023-12-28:
@hyunel Thanks for the information