espressif/esp-idf #11248: Can't use the std::vector - fatal error: vector: No such file or directory

RJSDevel, 2023-04-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tryed to use the vector but it isn't available
with enabled cpp via target_compile_options(${COMPONENT_LIB} PRIVATE -std=gnu++20), the same problem
How to get it?

Alvin1Zhang, 2023-04-22:
Thanks for reporting, would you please help provide more details as per [the issue template](https://github.com/espressif/esp-idf/issues/new/choose)? Thanks.

0xjakob, 2023-04-23:
@RJSDevel `std::vector` is indeed available, we are actually [using `std::vector`](https://github.com/espressif/esp-idf-cxx/blob/main/include/i2c_cxx.hpp#L140) in [ESP-IDF-C++](https://github.com/espressif/esp-idf-cxx). Do you have a minimal reproducer for the build error? Could you provide the full compiler call that is executed? It looks a bit as if the include path to the toolchain is missing.

RJSDevel, 2023-04-23:
@0xjakob I've got the project just from LEDC example, and if i add #include <vector> than appears the error fatal error: vector: No such file or directory

RJSDevel, 2023-04-23:
[image]

RJSDevel, 2023-04-23:
[image]

RJSDevel, 2023-04-23:
I tryed to add to any projects from examples and the result the same

erhankur, 2023-04-24:
@RJSDevel For any given input file, the file name suffix determines what kind of compilation is done. In your case, GCC recognizes C files and will use C compilation instead of C++. So, basically, first of all, you need to change your file extension to `.cpp`  And second add the `extern "C"` directive to the `app_main`  function to make it visible in the C namespace.

RJSDevel, 2023-04-24:
@erhankur Thank you a lot! I've finally got vector, but too got other errors
[image]

erhankur, 2023-04-24:
C++ has own its huge rule set that differs then C. Of course, you have to make the necessary changes when you switch to C++. I think error messages are pretty clear to understand and fix the errors.

RJSDevel, 2023-04-24:
@erhankur yes, I think I will manage to it, thank you 😊

closed 2023-04-24 by erhankur