espressif/esp-idf #11250: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Ybump14, 2023-04-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

In my esp32, when connecting to an existing AP, I found that if the WIFI SSID configured for ESP32 is correct and the WIFI PWD is incorrect, it will trigger a crash 
Guru Medication Error: Core 0 panic'ed (LoadProhibited) after a long run Exception was unhandled.

### What is the actual behavior?

trying to connect to WIFI and always fail (due to intentionally configuring the wrong WIFI password)

### Steps to reproduce.

1. Step
2. Step
3. Step
...
...

### Debug Logs.

```plain
[2023-04-22 10:39:43.556] I (1832792) wifi:new:<5,0>, old:<5,0>, ap:<255,255>, sta:<5,0>, prof:1
[2023-04-22 10:39:43.556] I (1832793) wifi:state: init -> auth (b0)
[2023-04-22 10:39:43.556] I (1832798) wifi:state: auth -> assoc (0)
[2023-04-22 10:39:43.615] I (1832825) wifi:state: assoc -> run (10)
[2023-04-22 10:39:43.615] I (1832839) wifi:<ba-add>idx:0 (ifx:0, 9e:2b:a6:a1:1a:c1), tid:5, ssn:0, winSize:64
[2023-04-22 10:39:45.733] I (1834971) wifi:<ba-add>idx:1 (ifx:0, 9e:2b:a6:a1:1a:c1), tid:0, ssn:3, winSize:64
[2023-04-22 10:39:47.978] I (1837231) wifi:state: run -> init (fc0)
[2023-04-22 10:39:47.981] I (1837231) wifi:<ba-del>idx
[2023-04-22 10:39:47.983] I (1837232) wifi:<ba-del>idx
[2023-04-22 10:39:47.986] I (1837232) wifi:new:<5,0>, old:<5,0>, ap:<255,255>, sta:<5,0>, prof:1
[2023-04-22 10:39:47.991] E (1837235) wifi:age_reorder: null rap
[2023-04-22 10:39:47.994] I (17:35:57.646) WIFI: free heap size: 4149567 bytes
[2023-04-22 10:39:48.000] I (17:35:57.650) WIFI: free dram size: 4149842 bytes
[2023-04-22 10:39:48.007] I (17:35:57.655) WIFI: retry 3 to connect to the ap
[2023-04-22 10:39:48.014] I (17:35:57.661) WIFI: wifi disconnect ap
[2023-04-22 10:39:51.356] I (1840610) wifi:new:<12,0>, old:<5,0>, ap:<255,255>, sta:<12,0>, prof:1
[2023-04-22 10:39:51.362] I (1840611) wifi:state: init -> auth (b0)
[2023-04-22 10:39:52.358] I (1841612) wifi:state: auth -> init (200)
[2023-04-22 10:39:52.361] I (1841612) wifi:new:<12,0>, old:<12,0>, ap:<255,255>, sta:<12,0>, prof:1
[2023-04-22 10:39:52.367] I (17:36:02.020) WIFI: free heap size: 4149851 bytes
[2023-04-22 10:39:52.373] I (17:36:02.022) WIFI: free dram size: 4149998 bytes
[2023-04-22 10:39:52.380] I (17:36:02.028) WIFI: retry 4 to connect to the ap
[2023-04-22 10:39:52.385] I (17:36:02.034) WIFI: wifi disconnect ap
[2023-04-22 10:39:55.739] I (17:36:05.392) WIFI: free heap size: 4149887 bytes
[2023-04-22 10:39:55.749] I (17:36:05.392) WIFI: free dram size: 4150034 bytes
[2023-04-22 10:39:55.749] I (17:36:05.392) WIFI: retry 5 to connect to the ap
[2023-04-22 10:39:55.754] I (17:36:05.398) WIFI: wifi disconnect ap
[2023-04-22 10:39:59.143] I (1848357) wifi:new:<5,0>, old:<12,0>, ap:<255,255>, sta:<5,0>, prof:1
[2023-04-22 10:39:59.143] I (1848358) wifi:state: init -> auth (b0)
[2023-04-22 10:39:59.143] I (1848365) wifi:state: auth -> assoc (0)
[2023-04-22 10:39:59.155] I (1848407) wifi:state: assoc -> run (10)
[2023-04-22 10:39:59.170] I (1848424) wifi:<ba-add>idx:0 (ifx:0, 9e:2b:a6:a1:1a:c1), tid:5, ssn:0, winSize:64
[2023-04-22 10:39:59.261] I (1848514) wifi:<ba-add>idx:1 (ifx:0, 9e:2b:a6:a1:1a:c1), tid:0, ssn:1, winSize:64
[2023-04-22 10:40:03.560] I (1852810) wifi:state: run -> init (fc0)
[2023-04-22 10:40:03.564] I (1852810) wifi:<ba-del>idx
[2023-04-22 10:40:03.564] I (1852810) wifi:<ba-del>idx
[2023-04-22 10:40:03.566] I (1852811) wifi:new:<5,0>, old:<5,0>, ap:<255,255>, sta:<5,0>, prof:1
[2023-04-22 10:40:03.570] I (1852814) wifi:new:<12,0>, old:<5,0>, ap:<255,255>, sta:<12,0>, prof:1
[2023-04-22 10:40:03.578] I (1852821) wifi:state: init -> auth (b0)
[2023-04-22 10:40:03.582] Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[2023-04-22 10:40:03.587] 
[2023-04-22 10:40:03.587] Core  0 register dump:
[2023-04-22 10:40:03.590] PC      : 0x401a9f80  PS      : 0x00060330  A0      : 0x8016d551  A1      : 0x3ffd59e0  
[2023-04-22 10:40:03.598] A2      : 0x3f80ee28  A3      : 0x3f825234  A4      : 0x00000005  A5      : 0x00000002  
[2023-04-22 10:40:03.606] A6      : 0x00000005  A7      : 0x00001005  A8      : 0x3f8261b8  A9      : 0x00000000  
[2023-04-22 10:40:03.615] A10     : 0x3f80ee28  A11     : 0x00000001  A12     : 0x400910a8  A13     : 0x3ffb50a4  
[2023-04-22 10:40:03.620] A14     : 0x400d65dc  A15     : 0x003fffff  SAR     : 0x00000003  EXCCAUSE: 0x0000001c  
[2023-04-22 10:40:03.629] EXCVADDR: 0x00000017  LBEG    : 0x40083d61  LEND    : 0x40083d69  LCOUNT  : 0x00000027  
[2023-04-22 10:40:03.636] 
[2023-04-22 10:40:03.636] 
[2023-04-22 10:40:03.636] Backtrace: 0x401a9f7d:0x3ffd59e0 0x4016d54e:0x3ffd5a00 0x4016d5eb:0x3ffd5a30 0x4017bf15:0x3ffd5a60 0x4017c301:0x3ffd5a80 0x401a93ad:0x3ffd5aa0
[2023-04-22 10:40:03.651] 
[2023-04-22 10:40:03.651] 
[2023-04-22 10:40:03.651] 
[2023-04-22 10:40:03.651] 
[2023-04-22 10:40:03.651] ELF file SHA256: f3425eb9e6309601
[2023-04-22 10:40:03.653]
```

Xiehanxin, 2023-04-23:
hi, @Ybump14. 
1. I have  tried the [station example ](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station) in IDF and can't reproduce your problem. could you tried this simple example and double check if the connecting step and code are similar to the example.
2. I have seen backterce in your log, if you use idf monitor these will be converted to code locations(function name, file name, and line number) and it seems that you used other serial terminal tool like minicom, you will see the ELF in your project and  you can use _xtensa-esp32-elf-addr2line -pfiaC -e <elf_file><backterce address>_ in your terminal to get the source. 
3. You can get more details in https://docs.espressif.com/projects/esp-idf/en/release-v3.2/get-started/idf-monitor.html and https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html

Ybump14, 2023-04-24:
@Xiehanxin Thank, I feel that you are a developer from China based on your Github name. May I ask if we can communicate in Chinese in the future. I will try the advice you provided, We will run a station example to observe if the same thing will eventually happen

---

In the above stack, we attempted to parse and obtained the following results:

PS D:\16.CompanyProject\09.ESP32_CAN\25.0301.9043-05-rc> xtensa-esp32-elf-addr2line -pfiaC -e build/IOT_904315_ESP32.elf Backtrace: 0x401a9f7d:0x3ffd59e0 0x4016d54e:0x3ffd5a00 0x4016d5eb:0x3ffd5a30 0x4017bf15:0x3ffd5a60 0x4017c301:0x3ffd5a80 0x401a93ad:0x3ffd5aa0
0x00000bac: ?? ??:0
0x401a9f7d: sta_input at ??:?
0x4016d54e: ieee80211_update_channel at ??:?
0x4016d5eb: ieee80211_update_channel at ??:?
0x4017bf15: chm_acquire_lock at ??:?
0x4017c301: chm_return_home_channel at ??:?
0x401a93ad: wdev_mac_wakeup at ??:?

---

Sherry616, 2024-05-15:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. 
Thanks for using our Espressif product.

closed 2024-05-15 by Sherry616