espressif/esp-idf #11252: esp_phy/Kconfig: "Reduce PHY TX power when brownout reset" is not visible in v4.4

AxelLin, 2023-04-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-433-g3cec3a0026

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

USB

### What is the expected behavior?

"Reduce PHY TX power when brownout reset" config option is not visible in v4.4

### What is the actual behavior?

I think the problem is "depends on ESP_BROWNOUT_DET" in b91d25e5756da298936938c6789c36c133145ab6, 
ESP_BROWNOUT_DET does not exist in v4.4 branch, should be ESP32_BROWNOUT_DET.

### Steps to reproduce.

Just execute "idf.py menuconfig" to check it.

AxelLin, 2023-05-27:
@mhdong Would you fix this? The fix is really trivial.

Alvin1Zhang, 2023-06-09:
Thanks for reporting and tracking. The fix has been merged internally and now being synced to Github. Thanks.