espressif/esp-idf #11254: Sharp AC IR Transmission

AhmedShaalan, 2023-04-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to modify the ir_nec_transceiver to send Sharp AC IR commands.

I have update the duty cycle
```c
rmt_carrier_config_t carrier_cfg = {
        .duty_cycle = 0.50, 
        .frequency_hz = 38000, // 38KHz
};
```

I have updated the scan_code to be as follows
```c
uint8_t scan_code[13] = {0xAA, 0x5A, 0xCF, 0x10, 0x05, 0x31, 0x32, 0x00, 0x08, 0x80, 0x04, 0xF0, 0x11};
ESP_ERROR_CHECK(rmt_transmit(tx_channel, nec_encoder, scan_code, sizeof(scan_code), &transmit_config));
```

I have update mark & space timings. However, I removed the ending symbol since Sharp AC doesn't have it

```c
nec_encoder->nec_leading_symbol = (rmt_symbol_word_t){
    .level0 = 1,
    .duration0 = 3800ULL * config->resolution / 1000000,
    .level1 = 0,
    .duration1 = 1900ULL * config->resolution / 1000000,
};

rmt_bytes_encoder_config_t bytes_encoder_config = {
    .bit0 = {
        .level0 = 1,
        .duration0 = 470 * config->resolution / 1000000,
        .level1 = 0,
        .duration1 = 500 * config->resolution / 1000000,
    },
    .bit1 = {
        .level0 = 1,
        .duration0 = 470 * config->resolution / 1000000,
        .level1 = 0,
        .duration1 = 1400 * config->resolution / 1000000,
    },
};
```

Then, in the rmt_encode_ir_nec function i have updated it as follows: 
```c
uint8_t *scan_code = (uint8_t *)primary_data;
```

and updated the switch statement as follows:
```c
switch (nec_encoder->state)
{
case 0: // send header
    encoded_symbols += copy_encoder->encode(copy_encoder, channel, &nec_encoder->nec_leading_symbol,
                                            sizeof(rmt_symbol_word_t), &session_state);
    if (session_state & RMT_ENCODING_COMPLETE)
    {
        nec_encoder->state = 1; // we can only switch to next state when current encoder finished
    }
    if (session_state & RMT_ENCODING_MEM_FULL)
    {
        state |= RMT_ENCODING_MEM_FULL;
        goto out; // yield if there's no free space to put other encoding artifacts
    }
// fall-through
case 1: // send command
    encoded_symbols += bytes_encoder->encode(bytes_encoder, channel, scan_code, data_size, &session_state);
    if (session_state & RMT_ENCODING_COMPLETE)
    {
        nec_encoder->state = RMT_ENCODING_RESET; // back to the initial encoding session
        state |= RMT_ENCODING_COMPLETE;
    }
    if (session_state & RMT_ENCODING_MEM_FULL)
    {
        state |= RMT_ENCODING_MEM_FULL;
        goto out; // yield if there's no free space to put other encoding artifacts
    }
}
```

## Results
I have used [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266) Arduino library to decode the signal of the above code. 

<details><summary>Results</summary>
<p>
Protocol  : UNKNOWN
Code      : 0x85B048FC (105 Bits)

uint16_t rawData[209] = {3752, 1940,  424, 544,  430, 1438,  428, 540,  434, 1436,  426, 542,  436, 1430,  460, 510,  438, 1430,  434, 536,  440, 1428,  436, 536,  438, 1430,  434, 1432,  460, 512,  464, 1406,  436, 538,  462, 1402,  442, 1426,  438, 1428,  438, 1430,  438, 532,  438, 538,  434, 1430,  438, 1430,  436, 536,  436, 536,  434, 538,  432, 542,  404, 1462,  408, 546,  424, 542,  436, 538,  460, 1408,  436, 532,  440, 1430,  460, 510,  438, 538,  460, 510,  436, 540,  408, 544,  426, 1462,  404, 550,  426, 540,  434, 536,  436, 1432,  434, 1434,  430, 538,  438, 534,  438, 536,  436, 1432,  434, 538,  438, 534,  432, 1436,  436, 1430,  438, 534,  434, 522,  426, 480,  492, 544,  432, 540,  434, 538,  436, 538,  436, 534,  436, 538,  432, 522,  426, 542,  456, 516,  436, 534,  438, 1430,  438, 536,  436, 534,  438, 536,  456, 518,  428, 544,  404, 546,  430, 538,  434, 538,  438, 536,  436, 536,  436, 538,  430, 1434,  434, 542,  404, 548,  426, 1460,  406, 546,  430, 538,  434, 538,  434, 540,  432, 540,  434, 540,  432, 540,  406, 566,  406, 544,  434, 1436,  430, 1438,  426, 1462,  428, 1438,  430, 1440,  430, 540,  406, 548,  424, 544,  432, 1456,  410, 562,  410, 538,  434};  // UNKNOWN 85B048FC
</p>
</details> 

## Expected Results
<details><summary>Details</summary>
<p>
Protocol  : SHARP_AC
Code      : 0xAA5ACF1005313200088004F011 (104 Bits)

Mesg Desc.: Model: 3 (A903), Power: On, Mode: 2 (Cool), Temp: 20C, Fan: 3 (Low), Swing(V): 0 (N/A), Turbo: Off, Ion: Off, Light: -, Clean: Off

uint16_t rawData[211] = {3832, 1856,  460, 484,  464, 1426,  488, 456,  462, 1428,  462, 480,  462, 1430,  462, 482,  460, 1432,  464, 472,  470, 1428,  462, 480,  488, 1404,  458, 1430,  462, 482,  462, 1430,  460, 490,  462, 1428,  462, 1428,  462, 1428,  462, 1428,  462, 476,  468, 482,  462, 1428,  482, 1414,  460, 474,  470, 482,  460, 476,  468, 482,  460, 1430,  460, 482,  462, 482,  462, 488,  462, 1428,  462, 482,  462, 1426,  462, 476,  468, 476,  470, 480,  460, 482,  464, 486,  464, 1428,  484, 458,  462, 482,  462, 484,  460, 1428,  462, 1428,  464, 482,  460, 486,  466, 476,  470, 1428,  460, 482,  462, 482,  462, 1428,  458, 1434,  458, 482,  462, 480,  470, 480,  462, 476,  468, 482,  460, 476,  470, 482,  460, 478,  464, 484,  460, 490,  460, 484,  460, 472,  472, 482,  458, 1430,  460, 484,  488, 452,  466, 480,  462, 486,  462, 480,  464, 482,  462, 482,  462, 480,  462, 472,  470, 476,  468, 482,  464, 1432,  462, 482,  462, 484,  462, 1428,  462, 472,  468, 486,  458, 482,  460, 482,  462, 488,  462, 482,  460, 484,  460, 484,  460, 476,  492, 1406,  460, 1430,  486, 1404,  462, 1434,  458, 1432,  458, 480,  466, 482,  458, 484,  458, 1432,  484, 458,  460, 484,  460, 480,  462};  // SHARP_AC

uint8_t state[13] = {0xAA, 0x5A, 0xCF, 0x10, 0x05, 0x31, 0x32, 0x00, 0x08, 0x80, 0x04, 0xF0, 0x11};
</p>
</details> 

## Conclusion 
As you can see the code is shorter and different.

zador-blood-stained, 2023-04-22:
Hello.

Your received code is different because your bits are reversed - you need to add `.flags.msb_first = 1,` to your `bytes_encoder_config` structure:

```
rmt_bytes_encoder_config_t bytes_encoder_config = {
    .bit0 = {
        .level0 = 1,
        .duration0 = 470 * config->resolution / 1000000,
        .level1 = 0,
        .duration1 = 500 * config->resolution / 1000000,
    },
    .bit1 = {
        .level0 = 1,
        .duration0 = 470 * config->resolution / 1000000,
        .level1 = 0,
        .duration1 = 1400 * config->resolution / 1000000,
    },
    .flags.msb_first = 1,
};
```

But I found this issue because of a similar problem: I need to send a signal that consists of a header (1 RMT symbol) and a data payload (I need 4 bytes, but so far I tried different payloads ranging from 1 to 6 bytes) using similar code based on the NEC example, and I'm receiving correct bits but the last bit seems to be missing half the time. Right now I'm still debugging because of inconsistent results with an ESP8266 board as a receiver, I plan to use a logic analyzer on the output pin on the sending board.

My (IR sending) device is ESP32-C3, ESP-IDF version is 5.0.1

zador-blood-stained, 2023-04-22:
Update: I think it's just the IRremoteESP8266 library decoder missing the last bit because there is no ending code in the transmission or it is not decoded properly.

zador-blood-stained, 2023-04-22:
There should be an ending symbol otherwise it is impossible to differentiate short mark+short space (bit 0) from short mark+long space (bit 1) in the last bit of this type of IR protocol.

I just looked at the signal on the transmitting LED from my AC remote that I'm trying to emulate and there are actually 33 marks (not including the start bit) for 32 bits of data that are sent. There is an extra pulse at the end of the transmission.

So there needs to be an ending symbol similar to the NEC protocol.

AhmedShaalan, 2023-04-22:
Yeah I think you're right, there should be an ending symbol.

Also, I don't think `.flags.msb_first = 1,` made a difference.
I have tried sending 0xFF except the first byte.
`uint8_t scan_code[13] = {0x00, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};`

<details><summary>Results With MSB flag</summary>
<p>
Protocol  : UNKNOWN
Code      : 0xD5D93CDF (105 Bits)
uint16_t rawData[209] = {3806, 1898,  474, 496,  478, 498,  440, 516,  426, 544,  434, 534,  472, 498,  472, 498,  474, 498,  472, 1396,  474, 1392,  474, 1398,  470, 1396,  474, 1396,  468, 1400,  470, 1400,  438, 1430,  434, 1436,  404, 1466,  450, 1434,  436, 1382,  488, 1426,  470, 1398,  476, 1394,  470, 1398,  446, 1420,  472, 1396,  478, 1390,  476, 1392,  466, 1404,  438, 1430,  434, 1436,  404, 1464,  404, 1480,  436, 1432,  438, 1428,  466, 1400,  474, 1396,  466, 1400,  472, 1394,  470, 1396,  472, 1396,  470, 1400,  470, 1398,  466, 1402,  436, 1434,  432, 1436,  404, 1464,  424, 1458,  436, 1432,  438, 1428,  440, 1428,  470, 1396,  468, 1398,  470, 1398,  472, 1394,  470, 1398,  472, 1396,  464, 1404,  438, 1430,  434, 1434,  404, 1464,  426, 1460,  432, 1434,  436, 1430,  464, 1404,  468, 1398,  468, 1400,  466, 1400,  468, 1400,  470, 1396,  470, 1400,  466, 1402,  468, 1402,  460, 1410,  430, 1440,  426, 1444,  448, 1434,  434, 1432,  460, 1406,  466, 1402,  468, 1400,  468, 1400,  466, 1402,  468, 1398,  468, 1400,  466, 1404,  464, 1406,  434, 1434,  432, 1438,  428, 1440,  426, 1460,  432, 1434,  460, 1406,  438, 1430,  466, 1400,  464, 1404,  464, 1404,  462, 1406,  464, 1402,  466, 1404,  462, 1408,  460, 1410,  432, 1436,  430};  // UNKNOWN D5D93CDF
</p>
</details> 

<details><summary>Results Without MSB flag</summary>
<p>
Protocol  : UNKNOWN
Code      : 0xD5D93CDF (105 Bits)
uint16_t rawData[209] = {3786, 1896,  474, 496,  472, 504,  472, 500,  476, 492,  442, 520,  402, 564,  438, 532,  472, 496,  476, 1394,  472, 1396,  476, 1392,  468, 1404,  436, 1432,  406, 1464,  424, 1462,  434, 1432,  438, 1428,  470, 1398,  470, 1396,  502, 1366,  470, 1398,  446, 1420,  472, 1394,  474, 1394,  472, 1396,  474, 1398,  436, 1432,  408, 1464,  402, 1480,  434, 1434,  460, 1408,  462, 1404,  472, 1394,  476, 1392,  472, 1396,  478, 1388,  476, 1392,  474, 1394,  472, 1396,  474, 1394,  466, 1404,  436, 1434,  406, 1462,  404, 1464,  448, 1436,  438, 1430,  464, 1400,  470, 1396,  476, 1392,  468, 1400,  472, 1394,  472, 1396,  474, 1396,  470, 1400,  464, 1404,  436, 1434,  430, 1438,  404, 1464,  436, 1448,  436, 1432,  440, 1428,  466, 1402,  474, 1392,  472, 1396,  470, 1398,  470, 1398,  470, 1398,  468, 1400,  468, 1402,  462, 1408,  432, 1436,  404, 1464,  424, 1460,  434, 1432,  436, 1430,  466, 1402,  470, 1396,  470, 1398,  470, 1400,  468, 1398,  470, 1398,  468, 1400,  468, 1402,  464, 1406,  434, 1436,  428, 1440,  424, 1444,  448, 1434,  434, 1432,  462, 1404,  466, 1402,  466, 1402,  466, 1400,  468, 1400,  466, 1400,  464, 1406,  466, 1402,  464, 1404,  458, 1410,  432, 1438,  428, 1458,  406, 1460,  432, 1434,  436};  // UNKNOWN D5D93CDF
</p>
</details>

zador-blood-stained, 2023-04-22:
`msb_first` flips the bit order within each byte, byte order stays the same

Actually comparing dumps from your first message and looking at the comment in the IRRemoteESP8266 library for the [Sharp AC code](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Sharp.cpp) it looks like `msb_first` is not needed here because the library flips the bit order before decoding and printing the hex result
```
///   Historically sendSharp() sends address & command in
///   MSB first order. This is actually incorrect. It should be sent in LSB
///   order. The behaviour of sendSharp() hasn't been changed to maintain
///   backward compatibility.
```

Decoding dumps from your first message results in:
Received:
```
0101 0101 0101 1010 1111 0011 0000 1000 1010 0000 1000 1100 0100 1100 0000 0000 0001 0000 0000 0001 0010 0000 0000 1111 1000 100
```
Expected:
```
0101 0101 0101 1010 1111 0011 0000 1000 1010 0000 1000 1100 0100 1100 0000 0000 0001 0000 0000 0001 0010 0000 0000 1111 1000 1000
```
`Protocol : SHARP_AC Code : 0xAA5ACF1005313200088004F011 (104 Bits)`

Flipping `0101 0101` results in `1010 1010` or AA hex, flipping `0101 1010` results in `0101 1010` or 5A hex, flipping `1111 0011` results in `1100 1111` or CF hex and so on

AhmedShaalan, 2023-04-22:
Good to know, thanks!

Now, I think the issue is that IRRemoteESP8266 library is not telling the whole story on the console output.
I will go through the raw code and try to find out what I'm missing on the protocol itself. 
Also, I will go through the code of IRRemoteESP8266 again maybe I can find something.

And, I will keep you posted if you are interested.

AhmedShaalan, 2023-04-23:
I figured out the ending signal, it is 470us.
Also, I managed to get it working with a code i wrote to generate mark/space with duty cycle of 50% and frequency of 38,000Hz on a Pin.

However, I can't get it to work with the RMT encoder.

```
Protocol  : UNKNOWN
Code      : 0xA3DBDD67 (106 Bits)
uint16_t rawData[211] = {3798, 1888,  472, 498,  478, 1390,  476, 492,  484, 1386,  476, 498,  482, 1386,  480, 494,  480, 1388,  476, 492,  484, 1386,  478, 490,  482, 1390,  482, 1386,  478, 494,  480, 1386,  480, 494,  474, 1392,  482, 1386,  482, 1386,  480, 1388,  476, 496,  478, 492,  476, 1392,  480, 1386,  482, 486,  454, 510,  452, 518,  406, 564,  466, 1402,  458, 510,  474, 494,  484, 486,  482, 492,  484, 494,  470, 488,  450, 1430,  440, 518,  404, 564,  462, 506,  480, 490,  482, 1388,  480, 488,  510, 466,  478, 496,  474, 496,  438, 1430,  466, 490,  452, 518,  408, 562,  470, 1400,  438, 530,  480, 490,  478, 1390,  480, 1388,  478, 494,  480, 490,  482, 492,  482, 490,  444, 514,  452, 518,  434, 536,  470, 498,  482, 486,  510, 468,  480, 492,  478, 500,  436, 516,  454, 1416,  452, 518,  434, 534,  470, 498,  478, 494,  478, 494,  478, 494,  474, 500,  436, 516,  406, 564,  442, 526,  472, 500,  478, 1386,  482, 488,  482, 494,  480, 490,  474, 502,  436, 516,  406, 564,  466, 502,  476, 494,  476, 496,  502, 472,  478, 494,  464, 490,  406, 1464,  450, 1430,  440, 1428,  468, 1400,  504, 1364,  478, 494,  470, 488,  450, 520,  406, 1464,  404, 566,  436, 532,  474, 1396,  468};  // UNKNOWN A3DBDD67
```