espressif/esp-idf #11256: BLE MESH OTA

gomez9656, 2023-04-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello there Espressif team!

I'm currently working with the ESP32-C3 in a BLE MESH app. I'm mostly using the generic level server to transmit the values that I need, and it works as expected.

I would like to add an OTA feature. As far as I understand, I can't send the wifi credentials trough the generic-level server because the payload size is too small. 

Is there a proper way to do this with different model in BLE MESH?

Thanks a lot in advance

JiaLWang, 2023-10-25:
Hi @gomez9656 ,

Sorry for the late reply. Based on your description there may be three options for you.
* It is better to use the vendor model to send vendor data.  
* Big data transfer using the Mesh BLOB model from the latest release of Mesh v1.1. (refer to https://www.bluetooth.com/specifications/specs/mesh-binary-large-object-transfer-model/)
* Mesh-based OTA is directly implemented using the newly released DFU model of Mesh v1.1. (refer to https://www.bluetooth.com/specifications/specs/mesh-device-firmware-update-model/)

We are sorry to say that the latter two options are not yet supported by our current ESP-IDF and will take some time to be supported.

smartcharge, 2024-01-03:
@JiaLWang 

Do you have an approximate date for when the OTA will be ready?

JiaLWang, 2024-01-04:
Hi @smartcharge ,

There is currently no definite end time, but we will release it as soon as it is completed.

matt-wood-ct, 2024-02-16:
Hi @JiaLWang now that ESP-IDF 5.2 has come out with Mesh V1.1 support is mesh DFU from the bluetooth SIG model being worked on interally, I am working on a project which ideally needs mesh DFU

venkateshgr-ayla, 2024-03-19:
@JiaLWang Even our organization is waiting for the BLE Mesh OTA support. This would be blocker for us as well.

Davide-Pedroncelli, 2024-03-24:
@JiaLWang The same for me, I need BLE Mesh OTA support. Is there a date now?

JiaLWang, 2024-03-28:
Hi All,

sorry for the late reply, at the moment we have completed the development of most of the DFU features, and are currently testing the relevant content and there is no specific timeline yet, we estimate that it will be officially released in the second half of the year. 

Thanks.

ShahinHaque97-CT, 2024-05-13:
@JiaLWang 

It seems ESP IDF v5.3 will be released early second half of the year, Would it be possible for this to be included in this release?

venkateshgr-ayla, 2024-06-18:
@JiaLWang do you have any beta version for us to start looking at?

sathishd-ayla, 2024-06-25:
Application Setup:
I'm currently working with the ESP32-S3 in a BLE MESH app where other nodes in mesh is ESP32H2. I'm mostly using the vendor model to transmit the values that I need for application, and it works as expected.
PSRAM is enabled in ESP32S3.

I would like to add an OTA feature for ESP32H2. since CONFIG_BLE_MESH_RX_SDU_MAX is 384, OTA binary size of 1MB is sent in chunks over BLE-Mesh. When the chunks are sent over mesh after few chunks I see an message "18:54:02.213 E v mod: (535026) BLE_MESH: Out of relay buffers
18:54:02.214 E v mod: (535026) BLE_MESH: Out of segment buffers
18:54:02.214 E v mod: (535027) BLE_MESH: Failed to send client message 0x00c002e5"

I have checked about the issue and tried to adjust the CONFIG_BLE_MESH_ADV_BUF_COUNT, but there is no improvement on the chunks transfer over BLE-Mesh.
Also I came across Mesh BLOB model, which can be used for Big Data transfer.
I have attached the sdkconfig for BLE-Mesh as follows,
[sdkconfig_BLE-Mesh.txt](https://github.com/user-attachments/files/15951853/sdkconfig_BLE-Mesh.txt)

Is there a way to do 1MB binary transfer in BLE MESH?

Note: We are blocked on this OTA Transfer.

Thanks a lot in advance
[Uploading sdkconfig_BLE-Mesh.txt…]()

JiaLWang, 2024-06-28:
Hi @sathishd-ayla,

Guessing from the error log, the message may be sent too fast, causing out of buffer. You can wait until you receive the ACK of the previous message before sending the next message. 

The Mesh BLOB model can be used to transfer large files.

After Mesh OTA is released, you can directly use Mesh OTA to transfer large files(In fact, Mesh OTA is essentially large file transfer through Mesh BLOB model).

venkateshgr-ayla, 2024-06-28:
@JiaLWang, can you point us to some examples related to "Mesh BLOB model"?

Also, can we get some beta version of Mesh OTA?

venkateshgr-ayla, 2024-07-03:
@JiaLWang with your Mesh OTA, how much time it takes for a 1MB OTA file transfer over BLE mesh?
What is the binary size to just have BLE mesh and OTA support?

ShahinHaque97-CT, 2024-07-03:
@venkateshgr-ayla 
For context on the nRF52832 Mesh DFU on Zephyr RTOS takes about 40min on an update of 370KB for me, Not saying the ESP on a completely different RTOS will be the same, but I imagine it to be quite similar as they are all following the same standards.

venkateshgr-ayla, 2024-07-03:
That's good information. Thanks a lot.
My concern is, with no WiFi enabled or used the BLE mesh enabled binary itself is too bulky and hence the OTA is huge time.

matt-wood-ct, 2024-07-03:
> That's good information. Thanks a lot. My concern is, with no WiFi enabled or used the BLE mesh enabled binary itself is too bulky and hence the OTA is huge time.

Mesh OTA is not fast, really it should be done in the background of your application so its non disruptive and completes "eventually™"

venkateshgr-ayla, 2024-07-12:
Thanks @matt-wood-ct .
@JiaLWang any idea when we will have Mesh OTA official release?

venkateshgr-ayla, 2024-07-12:
any data sent at 1sec interval over BLE mesh give below error after sending 2-3kb data or even before that.
Used IDF 5.2.2 as well. The mesh communication is not stable in these IDF so far

forx157, 2024-07-15:
Hi, @venkateshgr-ayla 
Both `Out of relay buffers` and `Out of segment buffers` are caused by the same thing: ADV Buffer exhaustion. Therefore, the macro `BLE_MESH_RELAY_ADV_BUF` can be enabled to avoid relay packets from taking up the resources of ADV packets. It is also possible to allocate more ADV packet resources by increasing the macro `BLE_MESH_ADV_BUF_COUNT`.

venkateshgr-ayla, 2024-08-05:
@JiaLWang have you released BLE Mesh OTA? If so which version of IDF is it?

ShahinHaque97-CT, 2024-08-05:
Doesn't look like it's released yet.

venkateshgr-ayla, 2024-08-05:
https://github.com/espressif/esp-idf/blob/v5.3/components/bt/esp_ble_mesh/v1.1/api/models/include/esp_ble_mesh_mbt_model_api.h
This file shows blob method which has to be used for BLE Mesh OTA.

Are there any beta releases which has BLE Mesh OTA?

hypnotik-krishna, 2025-08-07:
> Hi @gomez9656 ,
>
> Sorry for the late reply. Based on your description there may be three options for you.
>
> * It is better to use the vendor model to send vendor data.
> * Big data transfer using the [...]

You have commented this on 25 Oct, 2023,
Have you completed it now ?

juanpgg-git, 2025-08-09:
@hypnotik-krishna I was able to use the vendor model to share the wifi credentials. 
I think there's a lot of interest for people to use BLE MESH OTA