espressif/esp-idf #11258: Unable to install in virtual studio

zjf1165, 2023-04-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Unable to install after upgrading to MacOs 13.3.1(Intel Core i7)

### What is the actual behavior?

Unable to install after upgrading to MacOs 13.3.1(Intel Core i7)

### Steps to reproduce.

just download virtual studio code and install Esp-idf

### Build or installation Logs.

```plain
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.10 / 0.10) KB
Downloading ESP-IDF v5.0.1 (release version)...
esp-idf-v5.0.1.zip has (925738) KB
esp-idf-v5.0.1.zip progress: 0.00% (1.35 / 925737.58) KB
...
esp-idf-v5.0.1.zip progress: 100.00% (925737.58 / 925737.58) KB
Downloaded v5.0.1 (release version). Extracting...

Extracted /Users/zhoujunfeng/esp/esp-idf-v5.0.1.zip in /Users/zhoujunfeng/esp.

/bin/sh: xtensa-esp32-elf-gcc: command not found
Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: xtensa-esp32-elf-gcc: command not found
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: xtensa-esp32-elf-gcc: command not found

/bin/sh: openocd: command not found
Command failed: openocd --version
/bin/sh: openocd: command not found
Error checking openocd-esp32 version
Error: Command failed: openocd --version
/bin/sh: openocd: command not found

/bin/sh: xtensa-esp32s2-elf-gcc: command not found
Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: xtensa-esp32s2-elf-gcc: command not found
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: xtensa-esp32s2-elf-gcc: command not found

/bin/sh: esp32ulp-elf-as: command not found
Command failed: esp32ulp-elf-as --version
/bin/sh: esp32ulp-elf-as: command not found
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
/bin/sh: esp32ulp-elf-as: command not found

Downloading GDB for Xtensa
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz has (37488) KB
Failed download. Retrying...
Error: Error: Error creating dist directory
Waiting 8 seconds...
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.00% (1.35 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.01% (2.69 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.01% (4.04 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.01% (5.38 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.02% (6.73 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.02% (8.07 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.03% (9.42 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.03% (10.77 / 37487.09) KB
ENOENT: no such file or directory, open '/Users/zjf/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz'
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz has (37488) KB
Failed download. Retrying...
Error: Error: Error creating dist directory
Waiting 16 seconds...
ENOENT: no such file or directory, open '/Users/zjf/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz'
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz has (37488) KB
Failed download. Retrying...
Error: Error: Error creating dist directory
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.00% (1.35 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.01% (2.69 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.01% (4.04 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.01% (5.38 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.02% (6.73 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.02% (8.07 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.03% (9.42 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.03% (10.77 / 37487.09) KB
xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz progress: 0.03% (12.11 / 37487.09) KB
Error during ESP-IDF setup
ENOENT: no such file or directory, open '/Users/zjf/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz'
```

brianignacio5, 2023-04-23:
Seems to be a connection issue since downloads are failing. Maybe try to use the select download server Espressif in the `ESP-IDF: Configure ESP-IDF extension` setup wizard.

zjf1165, 2023-04-25:
> Seems to be a connection issue since downloads are failing. Maybe try to use the select download server Espressif in the `ESP-IDF: Configure ESP-IDF extension` setup wizard.

The problem still exists。

I can use install.sh to download all the files, but Virtual Studio Code still prompts me to download them.

brianignacio5, 2023-04-25:
Based on what you are sharing it seems downloads get broken due to connection issue. Did you tried using the Espressif download server in the setup wizard ? What is the output in that case ? You could use an existing setup as described in the setup docs. Just choose existing ESP-IDF directory and IDF_TOOLS_PATH (.espressif directory)

zjf1165, 2023-04-25:
> Based on what you are sharing it seems downloads get broken due to connection issue. Did you tried using the Espressif download server in the setup wizard ? What is the output in that case ? You [...]

Yes，I tried using github and Espressif，but the error was the same。

When I choose to install Workspace or Workspace Folder, the following error occurs：
[image]

brianignacio5, 2023-05-17:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/696142616.zip) ?

brianignacio5, 2023-05-17:
@espressif-bot move this to https://github.com/espressif/vscode-esp-idf-extension ?

brianignacio5, 2023-07-03:
Should have some fixes for this in the new v1.6.4! 

Please give it a try

lorenz9314, 2023-10-12:
Just a heads up as this has happened to me. I accidentally installed the plugin on a remote machine, I was still connected to. Realizing my mistake, I disconnected and attempted to install the plugin on my local machine, failing with the same result. It took me a day until I realized that the install paths set during the welcome screen, where still those from the remote machine somehow. The downloads failed because these paths didn't even exist on my local machine. A hint might be this message at the very bottom:

`ENOENT: no such file or directory, open '/Users/zjf/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-apple-darwin14.tar.gz'`

Does this path or the user 'zjf' exist on your machine?