espressif/esp-idf #11259: Directory with (nearly) duplicated files

eiffel31, 2023-04-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

All the files in the directory components/bt/porting/nimble/include/os are also in the directory components/bt/host/nimble/nimble/porting/nimble/include/os

Some files are only in components/bt/host/nimble/nimble/porting/nimble/include/os

The common files are identical or nearly identical.
The differences are most likely problems than solutions. Files were identical at the origin (copy time) and changes not reported to the copy made them out-of-sync.

Same problem for components/bt/host/nimble/nimble/porting/nimble/include/nimble/nimble_port.h and components/bt/porting/nimble/include/nimble/nimble_port.h

Duplicates files has to be removed from the project.

rahult-github, 2023-05-09:
The reason for such architecture is explained [here](https://github.com/espressif/esp-idf/issues/11236#issuecomment-1519893298).

eiffel31, 2023-05-09:
Hello,
The two issues are unrelated.
Here, the files in components/bt/porting/nimble/include/os are reported as duplicated files but out-of-sync files.

These files are real source code, not examples files.

Either you remove those files, or you add some copy mechanism or verification system in order to ensure that the files are always identical.

rahult-github, 2023-08-22:
The header files re-arrangement is needed to ensure different IPs are supported. However, the request to have files in sync is valid and a [commit](https://github.com/espressif/esp-idf/commit/6ace246ba2f7452a3f8d78006327870fe7d90600) for same was recently made. We will try to keep the header files in sync going ahead.