espressif/esp-idf #11260: TWAI: Adjusting filters

mickeyl, 2023-04-23:
### Is your feature request related to a problem?

It looks a bit over-the-top to reinstall the whole TWAI driver when all you want to do is program another set of filters, which – in normal operation – is pretty often the case. I'm using TWAI to reprogram ECUs and typically you leave everything on accept and fiddle with bitrates until you have got one valid receive.

Afterwards, you keep the bitrate intact and only adjust the filters regularly to talk to multiple devices.

Can you suggest an alternative to reinstalling the whole driver?

### Describe the solution you'd like.

A `twai_update_filters` would be perfect.

mickeyl, 2024-04-30:
Not even a single comment? Is that such an esoteric request?

wanckl, 2025-07-04:
@mickeyl  Tank you for your inclusion.

And esp32c5 is now also support FD feature, and we are recently going on certification. 

Looking forward to your new design.