espressif/esp-idf #11261: ESP32-S3FN8 - BLE service problem

marbalon, 2023-04-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board.

### Power Supply used.

USB

### What is the expected behavior?

The BLE GATT service should be discoverable on every software restart.

### What is the actual behavior?

I'm developing the app with GATT server and everything is working fine: the device is discoverable, and I can connect and communicate. But when I do the software restart looks like the device is not broadcasting its own name. Just for a test, I added an option to scan devices nearby to verify if BT is working. The device works fine - see devices but still not broadcasting his own name.

What is strange it is always after the first software restart. On the second, and third, the fourth software restart fixes the issue.

### Steps to reproduce.

Nothing special is needed, just a first software restart (using esp_restart() function ,in our case after OTA upgrade or UART command). The problem does not appear on the ESP32-S3 with external flash - only on the ESP32-S3FN8.

zhp0406, 2023-11-17:
@marbalon 
Is this issue still present in the latest version?Can you provide me with the complete log?

marbalon, 2023-11-17:
Currently, I'm busy with other projects but I will try to post your logs when I switch to this project.
- What IDF branch should I use for the test?
- What logs do you want? Anything specific I should select in menuconfig?

zhp0406, 2023-11-21:
Please test on the latest master branch, and also test on the branch where the issue occurred earlier.

According to your description, the issue only occurs on the ESP32-S3FN8. If you can provide an example that reproduces the problem, I would greatly appreciate it.

zhp0406, 2023-11-21:
I need all the logs after powering on and starting up

closed 2024-12-17 by Alvin1Zhang