espressif/esp-idf #11262: RMT RX Filter "signal_range_min_ns" doesn't work

hamdiolgun, 2023-04-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-dev-6416-gef4b1b7704

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM32E

### Power Supply used.

Battery

### What is the expected behavior?

I am using RMT peripheral to communicate with other devices over HOMEBUS. There are a lot of noises on the bus, as natural.
I am trying to eliminate the noises with RMT RX filter module but i can't achieve it.
The communication baudrate 9600, and 1 bit 104us. 
For Homebus, bit 0 means "52us 0-52us 1", bit 1 means "104us 1"

I set the rx config struct as;
static const rmt_receive_config_t rmt_receive_config = {
        .signal_range_min_ns = (uint32_t)18 * 1000, //18us
        .signal_range_max_ns = (uint32_t)104 * 12 * 1000 //~1200us
};
_[rmt_receive_config_t::signal_range_min_ns](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/api-reference/peripherals/rmt.html#_CPPv4N20rmt_receive_config_t19signal_range_min_nsE) specifies the minimal valid pulse duration (either high or low level). A pulse whose width is smaller than this value will be treated as glitch and ignored by the hardware._
I expect that i should not get signals less than 18us by rmt_rx_done_event_data_t.
[image]

### What is the actual behavior?

I got all signals higher than 1us by rmt_rx_done_event_data_t rx data.

### Steps to reproduce.

rmt_channel_handle_t rmt_rx_channel;
const rmt_receive_config_t rmt_receive_config = {
        .signal_range_min_ns = (uint32_t)1000 * 1000,
        .signal_range_max_ns = (uint32_t)vrf_bit_us * VRF_BYTE_TOTAL_BIT_COUNT * 1000
};
QueueHandle_t rmt_receive_queue;

 rmt_rx_channel_config_t rx_channel_cfg = {
        .gpio_num = CONFIG_IO_VRF_RX_PIN,
        .clk_src = RMT_CLK_SRC_DEFAULT, //80MHz
        .resolution_hz = 1000000, // 1MHz resolution, 1 tick = 1us
        .mem_block_symbols = 5 * SOC_RMT_MEM_WORDS_PER_CHANNEL,
        .flags.invert_in = VRF_GPIO_RX_INVERT,
        .flags.with_dma = 0,
        .flags.io_loop_back = 0,
    };
    ESP_ERROR_CHECK(rmt_new_rx_channel(&rx_channel_cfg, &rmt_rx_channel));
    configASSERT(rmt_rx_channel);
    rmt_receive_queue = xQueueCreate(VRF_DATA_BUF_SIZE, sizeof(rmt_rx_done_event_data_t));
    configASSERT(rmt_receive_queue);
    rmt_rx_event_callbacks_t rcbs = {
        .on_recv_done = rmt_rx_done_callback,
    };
    ESP_ERROR_CHECK(rmt_rx_register_event_callbacks(rmt_rx_channel, &rcbs, rmt_receive_queue));
    ESP_LOGI("RMT", "Init done.");

   rmt_enable(rmt_rx_channel);
   rmt_receive(rmt_rx_channel, rmt_raw_symbols, sizeof(rmt_raw_symbols), &rmt_receive_config);//ready

### Debug Logs.

```plain
[0]{50, 0, 470, 1}{1, 9}
[1]{50, 0, 3, 1}{1, 0}
[2]{10, 0, 562, 1}{0, 11}
[3]{48, 0, 54, 1}{1, 1}
[4]{50, 0, 3, 1}{1, 0}
[5]{1, 0, 987, 1}{0, 19}
[6]{48, 0, 262, 1}{1, 5}
[7]{50, 0, 3, 1}{1, 0}
[8]{7, 0, 772, 1}{0, 15}
[9]{48, 0, 574, 1}{1, 11}
[10]{50, 0, 3, 1}{1, 0}
[11]{11, 0, 456, 1}{0, 9}
[12]{49, 0, 262, 1}{1, 5}
[13]{50, 0, 2, 1}{1, 0}
[14]{6, 0, 774, 1}{0, 15}
```

suda-morris, 2023-04-26:
@hamdiolgun 

Thanks for reporting! In fact, `.signal_range_min_ns = (uint32_t)18 * 1000, //18us` will result in a very large register value **1440** (which was calculated in the https://github.com/espressif/esp-idf/blob/master/components/driver/rmt/rmt_rx.c#L368). However, the underlying register is only 8bit-width, so actually what being set in the register is only `1440/256` which is only **5**. 

We will add a check in the driver in case the user sets a value that will lead to out-of-register-range.

Meanwhile, you can reduce the RMT clock resolution by choosing another clock source, e..g  `.clk_src = RMT_CLK_SRC_REF_TICK, //1MHz`, then we can get a valid register value after calculation.

hamdiolgun, 2023-04-26:
@suda-morris 
Thanks for your reply.
I had been realized the calculation of filter threshold.
`rmt_ll_rx_set_filter_thres(hal->regs, channel_id, ((uint64_t)group->resolution_hz * config->signal_range_min_ns) / 1000000000UL);`
And i've set the signal_range_min_ns according that,
```
rmt_rx_channel_config_t rx_channel_cfg = {
    .gpio_num = CONFIG_IO_VRF_RX_PIN,
    .clk_src = RMT_CLK_SRC_DEFAULT, //80MHz
    .resolution_hz = 1000000, // 1MHz resolution, 1 tick = 1us
    .mem_block_symbols = 5 * SOC_RMT_MEM_WORDS_PER_CHANNEL,
    .flags.invert_in = VRF_GPIO_RX_INVERT,
    .flags.with_dma = 0,
    .flags.io_loop_back = 0,
};
```
So,
group->resolution_hz = 1000000
config->signal_range_min_ns = 18 * 1000
filter thres = (1000000 * 18 * 1000) / 1000000000UL = 18
How did you calculated 1440 ?

Best regards

suda-morris, 2023-04-26:
> So, group->resolution_hz = 1000000

This is a trap of RMT. When setting the filter threshold, we're using the RMT's "group resolution" (one group has several channels). Which by default, equals the source clock's frequency (80MHz e.g.).  See https://github.com/espressif/esp-idf/blob/master/components/driver/rmt/rmt_common.c#L166

`.resolution_hz = 1000000` this configuration is set the "channel resolution".

closed 2023-05-08 by espressif-bot