espressif/esp-idf #11269: How to allocate dma buffer in external PSRAM (esp32s3)

ch-duan, 2023-04-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[esp-idf]: version v5.0.1
[chip]: esp32s3
[image]
I have used bluetooth blufi, bluetooth adv、WIFI、SPI、USB in my project. These use too much internal DMA memory.
How to allocate dma buffer in external PSRAM？

suda-morris, 2023-04-25:
This depends on whether the DMA used by the peripheral is able to access the PSRAM. For example, the SPI uses GDMA, and GDMA itself does support fetching data from PSRAM, so that means, on esp32s3, the SPI is able to transfer a PSRAM buffer. The USB is using a dedicated DMA engine, which is not possible to access the PSRAM.

ch-duan, 2023-04-25:
> This depends on whether the DMA used by the peripheral is able to access the PSRAM. For example, the SPI uses GDMA, and GDMA itself does support fetching data from PSRAM, so that means, on esp32s3, [...]

Can I allocate internal memory used by bluetooth and wifi to psram?
How does spi use the dma buffer allocated by external psram?
[image]
I tried to allocate the dma buffer received by spi to MALLOC_CAP_32BIT, but the received data will be wrong.
Can you give me some examples or documentation?

suda-morris, 2023-04-26:
> How does spi use the dma buffer allocated by external psram?

SPI **driver** currently doesn't support this feature, but it's WIP. In the esp-idf, the LCD driver is using the PSRAM DMA buffer heavily. e.g. https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/src/esp_lcd_panel_io_i80.c#L475

ch-duan, 2023-04-26:
I will try to use GDMA. Is there any plan for ESP32 to support all the memory used by Bluetooth, WIFI, and USB in external PSRAM?

gm-jiang, 2023-05-12:
@ch-duan The ESP32 hardware does not support WiFI Bluetooth data to directly access PSRAM through DMA, also the ESP32 has no USB peripheral. For ESP32-s3 I think you can first receive the data to internal RAM through DMA, and then do the data transfer from internal memory to PSRAM through EDMA.

Alvin1Zhang, 2023-06-05:
Thanks for reporting, will close due to short feedback, feel free to reopen.

closed 2023-06-05 by Alvin1Zhang