espressif/esp-idf #11270: Example of how to do on-chip key generation with the DS Peripheral?

torntrousers, 2023-04-25:
I'm looking at how to do key generation for the Digital Signature Peripheral for use with Mutual TLS connections.

The [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/ds.html#key-generation) says: 

_Both the HMAC key and the RSA private key have to be created and stored before the DS peripheral can be used. This needs to be done in software on the ESP32-S2 or alternatively on a host. ... You can find instructions on how to calculate and assemble the private key parameters in ESP32-S2 Technical Reference Manual_

which points to here: https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf#digsig

Its quite complicated and hard to understand!

Is there any code sample available that shows how to do this on-chip key generation in code?

chipweinberger, 2023-04-25:
see my code here

https://github.com/espressif/esp_secure_cert_mgr/issues/8

chipweinberger, 2023-04-25:
More code for reference.

You'll need to adapt this but it should still be useful.

```
include "esp_ds.h"
#include "esp_efuse.h"

#include "mbedtls/rsa.h"
#include "mbedtls/rsa_internal.h"
#include "mbedtls/platform_util.h"
#include "mbedtls/error.h"
#include "mbedtls/sha256.h"
#include "mbedtls/bignum.h"
#include "mbedtls/entropy.h"
#include "mbedtls/ctr_drbg.h"

#include "jcommon/system/pd_freertos_utils.h"
#include "jcommon/utils/pd_utils.h"
#include "jcommon/utils/pd_error.h"
#include "jcommon/logs/pd_log.h"
#include "jcommon/logs/pd_verbose.h"

#include "jcommon/hardware_info/pd_efuse.h"
#include "jcommon/hardware_info/pd_efuse_utils.h"
#include "jcommon/hardware_info/pd_hardware_info.h"
#include "jcommon/logs/pd_verbose.h"

#include "pd_espds.h"

static uint16_t TAGn = 0;
static const char* TAG = "pd espds";

// Forward Define 
void pd_espds_virtual_keygen_if_needed(pd_error_t* error);

static void print_mpi(const mbedtls_mpi* mpi, const char* name)
{
    size_t olen = 0;
    uint32_t len = MBEDTLS_MPI_MAX_SIZE * 4 + 1;
    char* pp = (char*)calloc(1, len);
    mbedtls_mpi_write_string(mpi, 16, pp, len, &olen);
    printf("%s: %s\n", name, pp);
    free(pp);
}

void pd_mbedtls_rsa_keygen(mbedtls_rsa_context* rsa, pd_error_t* error)
{
    pd_err_unset(error);

    bool twd_dis = false;

    // allocations
    mbedtls_entropy_context entropy;
    mbedtls_ctr_drbg_context ctrDrbg;
    mbedtls_entropy_init( &entropy );
    mbedtls_ctr_drbg_init( &ctrDrbg );
    {
        ESP_LOGI(TAG, "rsa keygen");

        // the personalize string just adds a litte
        // bit of "guarenteed" entropy, as a backup.
        const char *personalize = "jamcorder13849781";

        // Disable Idle Task WDT
        // Formatting takes a long time.
        eTry(,pd_idle_task_wdt_disable,(error));
        twd_dis = true;

        int ret = 0;
[... 235 lines cut ...]
            ESP_LOGW(TAG, "skip burning fuses. is developer device & fuses found.");

        } else {

            // print all keys
            pd_efuse_print_all_keys();

            ESP_LOGI(TAG, "burning fuseBlock: %u", fuseBlock);

            // burn hmac key into fuses
            sTry(err =,esp_efuse_write_key,(fuseBlock,
                ESP_EFUSE_KEY_PURPOSE_HMAC_DOWN_DIGITAL_SIGNATURE, hmac, 32));
        } 

        pd_err_success(error);
    }
end:
    // free
    free_safe((void**)&params);
    free_safe((void**)&encrypted);
    mbedtls_mpi_free(&rinv);
    mbedtls_rsa_free(&rsa);
    return;
}
```

torntrousers, 2023-04-25:
Thanks so much Chip, those look very interesting, will go away and study them. The next step would be to use that key to create either a self signed certificate or a certificate signing request - you don;t happen to have code for that too?

chipweinberger, 2023-04-25:
no, but use chatgpt. it should be familiar with mbedtls.

chipweinberger, 2023-04-25:
after my code, you'd need to call `esp_ds_start_sign` to sign the certificate data. And then somehow add that signature to the certificate itself. This would be using mbedtls.

chipweinberger, 2023-04-26:
A certificate is just a ASN.1 encoded text data. The signature is just appended to the end of the certificate data, along with which algo was used to generate the signature

[image]

chipweinberger, 2023-04-26:
> With those, the keys could just be stored in an encrypted partition and no one could get at them, could they?

Indeed. But storing private key ESP-DS is more secure because the keys are entirely inaccessible to software. 'C' code usually has tons of software exploits.

torntrousers, 2023-06-12:
> 

Hi @chipweinberger , can you give me a pointer where those jcommon includes come from?

chipweinberger, 2023-06-12:
You shouldn't need them! :)

The mbTry, sTry, eTry macros just check the error and goto end.

AdityaHPatwardhan, 2023-07-18:
@torntrousers 
We already have a a pip tool to do the chip generation on host machine https://pypi.org/project/esp-secure-cert-tool/

As for the on-chip key generation I think the [sample code](https://github.com/espressif/esp-idf/issues/11270#issuecomment-1521371130) mention by @chipweinberger looks like it can do the job for you.

Basically the steps are as follows:
1) Generate the RSA key on chip. This requires some cryptographic library for key generation. mbedTLS is most popular and easy to use.
2) Generate an HMAC key randomly.
3) Feed to required data to [esp_ds_encrypt_params](https://github.com/espressif/esp-idf/blob/4fc2e5cb95abb1e7339f11929bc9e916a9c3d15a/components/esp_hw_support/include/esp_ds.h#L227)
and generate the cipher text.
4) Store the cipher text and feed it to [esp_ds_sign](https://github.com/espressif/esp-idf/blob/4fc2e5cb95abb1e7339f11929bc9e916a9c3d15a/components/esp_hw_support/include/esp_ds.h#L227) for performing sign operation using the key.

If you are looking for doing the on-chip key generation for products then Espressif offers [Secure Pre-provisioning service](https://www.espressif.com/en/news/Espressif_preprovisioning) where modules are shipped with pre-programmed keys and certificates with all the relevant security features enabled.

AdityaHPatwardhan, 2023-07-21:
Here is one more [reference](https://github.com/espressif/esp-idf/blob/9a1cc59338aa151bc6333f85c8169d36a478fe2a/components/esp_hw_support/test_apps/esp_hw_support_unity_tests/main/test_ds.c#L106) for the parameter encryption.

I am closing the issue since relevant information is already available in the above discussion. Please feel free to re-open the issue if you are still having any problem.

closed 2023-07-21 by AdityaHPatwardhan

liliumjsn, 2024-04-03:
Hi @chipweinberger ,how did you manage to generate and write the device certificate?
There is no documentation at all for this. Is Espressif just promoting the pre-provision service?
Because it doesn't make any sense to not have an on-device provisioning API for the esp_secure_cert utility.

AdityaHPatwardhan, 2024-04-03:
@liliumjsn Please note, the ESP-IDF does not provide any mechanism for generating the private keys or certificates. It is provided by the crypto library. Hence you would find the documentation regarding generating the private keys or certificates in [mbedTLS](https://github.com/Mbed-TLS/mbedtls/tree/development) repository or any other relevant crypto library.
`esp_secure_cert` utility is strictly used for managing the available private keys/certs or other private data and storing them in convenient manner with simple APIs for storage and retrieval. It does not offer any APIs related to the parameters generation.

liliumjsn, 2024-04-03:
@AdityaHPatwardhan `esp_secure_cert` can only be used on the device to read ds storage data and not write them. In order to write data to ds partition you will either have to use a host and `esp_secure_cert` or generate manually tlv format data and write them to flash ds partition? Am I wrong?

AdityaHPatwardhan, 2024-04-03:
@liliumjsn You are right, At the moment the write option is only supported through external python interface. The firmware at the moment is only able to read the data. We currently have a MR in progress that also enables the write operations for the esp_secure_cert_mgr. Once that is merged the firmware shall write to the esp_secure_cert_partition.

liliumjsn, 2024-04-03:
@AdityaHPatwardhan Is [this](https://github.com/espressif/esp_secure_cert_mgr/files/11504441/support_writing.patch.zip) the latest we've come some far?