espressif/esp-idf #11272: ESP32-S2 Kaluga dev board Display test

krupis, 2023-04-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I have recently purchased ESP32-S2 Kaluga development board.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-esp32-s2-kaluga-1-kit.html

Currently, I develop on esp-idf v4.4.3 hence I havent been able to use the Kaluga dependacy via the component registry as it requires esp-idf v5.0 but that is not an issue.

I have been able to build and flash the example code that I found:
```
esp\esp-idf\examples\peripherals\lcd\tjpgd
```
I just changed the display pins as shown below:
```
#define EXAMPLE_PIN_NUM_DATA0          9  /*!< for 1-line SPI, this also refered as MOSI */
#define EXAMPLE_PIN_NUM_PCLK           15
#define EXAMPLE_PIN_NUM_CS             11
#define EXAMPLE_PIN_NUM_DC             13
#define EXAMPLE_PIN_NUM_RST            16

#define EXAMPLE_PIN_NUM_BK_LIGHT       6
```

The example flashed and surprisingly it seemed to work (I can see the image being displayed on the Kaluga dev board)

Questions:

1. I do not fully understand why the image is messed up a little bit. See the image below:
 https://imgur.com/a/KmljNgR

Do I need to adjust the parameters in the code to ensure the image in the correct format is displayed? Can you please clarify how to do that?

2. Is there any simple way to control the SPI display?
I just simply want to display some text or numbers, Is there a function that would allow to do that?

suda-morris, 2023-04-26:
@krupis 

> I do not fully understand why the image is messed up a little bit.

Please disable the color inverting function by calling  ` ESP_ERROR_CHECK(esp_lcd_panel_invert_color(panel_handle, false));`.  This is not a bug in the software but something depends on your LCD module.

> Is there any simple way to control the SPI display?

`esp_lcd` component is a driver component, which doesn't include GUI rendering stuffs. You may want to try some GUI library like [lvgl](https://components.espressif.com/components/lvgl/lvgl)

tore-espressif, 2023-04-26:
@krupis I'm sorry that the BSP does not work for you. For you initial tests, you can use Kaluga BSP v1.2.0 that supports IDF v4.4 https://components.espressif.com/components/espressif/esp32_s2_kaluga_kit/versions/1.2.0

and a basic example from tag `examples_v4.4` https://github.com/espressif/esp-bsp/tree/examples_v4.4/examples/display_audio

These older version might miss some of the new features introduced in newer version of the BSP. We plan to re-introduce support for IDFv4.4 releases to BSP soon

krupis, 2023-04-26:
> @krupis I'm sorry that the BSP does not work for you. For you initial tests, you can use Kaluga BSP v1.2.0 that supports IDF v4.4 [...]

Would you be able to explain how to run esp32_s2_kaluga_kit example project?

https://components.espressif.com/components/espressif/esp32_s2_kaluga_kit/versions/1.2.0
or
https://github.com/espressif/esp-bsp/tree/examples_v4.4/esp32_s2_kaluga_kit

Does not actually explain how can I create and build the project..

krupis, 2023-04-26:
> @krupis
>
>
> Please disable the color inverting function by calling ` ESP_ERROR_CHECK(esp_lcd_panel_invert_color(panel_handle, false));`. This is not a bug in the software but something depends on your [...]

Sure I can add:
```
ESP_ERROR_CHECK(esp_lcd_panel_invert_color(panel_handle, false));
```
Only thing that does is invert colors. Now the background is orange color and the letters are blue.
[image]

But the whole image is still messed up (warped and mirrored). 
Is this example supposed to generate this kind of image effect or it should be displayed as image.jpg
Actual image displayed:
https://ibb.co/KXt4M8Y

igrr, 2023-04-26:
I think this is warping effect is referred to as "pretty effect" implemented in the example source code, please check https://github.com/espressif/esp-idf/blob/3cec3a0026098d1b027f2103ec154a15baf97318/examples/peripherals/lcd/tjpgd/main/pretty_effect.c.

krupis, 2023-04-26:
> I think this is warping effect is referred to as "pretty effect" implemented in the example source code, please check https://github.com/espressif/esp-idf/blob/3cec3a0026098d1b027f2103ec154a15baf97318/examples/peripherals/lcd/tjpgd/main/pretty_effect.c.

Thanks for pointing that out! I should have checked it before. Now it makes sense why the displayed image is warped but I dont believe it should be mirrorred right? By looking at the displayed image, you can hardly tell it says "ESP32".

Do you think this is also a part of pretty effect or there is some issue with the display initialization?

payday1991, 2023-04-26:
> Thanks for pointing that out! I should have checked it before. Now it makes sense why the displayed image is warped but I dont believe it should be mirrorred right? By looking at the displayed [...]

Hi.
I've had this problem as well. I don't remember exactly what and where I've done it, but there should be the orientation settings in the display driver ST7789.c or so. You can change the rotation, but somewhere closer to the end of this file there is the code which overrides your settings. You have to write your screen rotation settings in there. Check that.

igrr, 2023-04-27:
Indeed settings such as rotation and mirroring may differ from display to display, and this basic IDF example does not attempt to get this right for every type of a development board. In esp-bsp components we do make sure that for each board the display is initialized with the correct settings. However if you are starting from a basic example, some DIY will probably be required, as pointed out by @payday1991.

krupis, 2023-04-27:
> Indeed settings such as rotation and mirroring may differ from display to display, and this basic IDF example does not attempt to get this right for every type of a development board. In esp-bsp [...]

Thanks for clarifying. Yeah that makes sense.

I just installed v5.0.1 and able to build the example project for kaluga by creating a new project using command
```
idf.py create-project-from-example "espressif/esp32_s2_kaluga_kit^2.2.3:display_audio"
```
as described in component registry:
[image]

I will let you know if I am able to get the display working properly with this example

krupis, 2023-04-28:
@igrr 

I have flashed the official kaluga kit example to my kaluga devboard and I can say that everything works fine! I am impressed!

Just having one issue with touch pad.
I refer to the documentation here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-esp32-s2-kaluga-1-kit.html

In order to get touch pad display to work, I must put switch positions to OFF
[image]

And thats exactly what I have done:
[image]

I have connected the touch pad to dedicated connect, (the clear plastic that came with the touch pad is on the bottom side of the touch pad).

[image]

As soon as I turn ON the devboard, it is constantly rebooting (I believe the issue is because touch pad clicked callback is constantly being called but I am  not touching the buttons)
```
H¬\[Z±¹ation scheme version is Line Fitting[1B][0m
[1B][0;32mI (753) adc button: Calibration Success[1B][0m
[1B][0;32mI (753) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (763) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (763) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (773) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (773) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (783) LVGL: Starting LVGL task[1B][0m
[1B][0;32mI (783) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
btn handler 
[1B][0;31mE (793) ES8311: es8311_voice_volume_get(380): I2C read/write error[1B][0m

assert failed: disp_set_volume disp_example.c:48 (volume_arc)

Backtrace: 0x40025bda:0x3ffcdde0 0x4002d3a5:0x3ffcde00 0x4003318d:0x3ffcde20 0x40088149:0x3ffcdf40 0x40087e38:0x3ffcdf60 0x4002f78d:0x3ffce000

ELF file SHA256: d23d4b603e268197

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025665
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1760
load:0x4004c000,len:0xabc
load:0x40050000,len:0x31b4
entry 0x4004c1c0
[1B][0;32mI (24) boot: ESP-IDF v5.0.1 2nd stage bootloader[1B][0m
[1B][0;32mI (24) boot: compile time 08:06:50[1B][0m
[1B][0;32mI (24) boot: chip revision: v0.0[1B][0m
[1B][0;32mI (26) boot.esp32s2: SPI Speed      : 80MHz[1B][0m
[1B][0;32mI (31) boot.esp32s2: SPI Mode       : DIO[1B][0m
[1B][0;32mI (36) boot.esp32s2: SPI Flash Size : 4MB[1B][0m
[1B][0;32mI (41) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (46) boot: Partition Table:[1B][0m
[1B][0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (72) boot:  2 factory          factory app      00 00 00010000 00100000[1B][0m
[1B][0;32mI (79) boot:  3 storage          Unknown data     01 82 00110000 002f0000[1B][0m
[1B][0;32mI (87) boot: End of partition table[1B][0m
[1B][0;32mI (91) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1502ch ( 86060) map[1B][0m
[1B][0;32mI (117) esp_image: segment 1: paddr=00025054 vaddr=3ffc47e0 size=01cdch (  7388) load[1B][0m
[1B][0;32mI (119) esp_image: segment 2: paddr=00026d38 vaddr=40024000 size=092e0h ( 37600) load[1B][0m
[1B][0;32mI (132) esp_image: segment 3: paddr=00030020 vaddr=40080020 size=56238h (352824) map[1B][0m
[1B][0;32mI (202) esp_image: segment 4: paddr=00086260 vaddr=4002d2e0 size=074f4h ( 29940) load[1B][0m
[1B][0;32mI (218) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (218) boot: Disabling RNG early entropy source...[1B][0m
[... 19 lines cut ...]
[1B][0;32mI (531) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM[1B][0m
[1B][0;32mI (537) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM[1B][0m
[1B][0;32mI (544) spi_flash: detected chip: generic[1B][0m
[1B][0;32mI (548) spi_flash: flash io: dio[1B][0m
[1B][0;32mI (582) cpu_start: Starting scheduler on PRO CPU.[1B][0m
[1B][0;32mI (583) example: Initializing SPIFFS[1B][0m
[1B][0;32mI (723) example: Partition size: total: 2293769, used: 303662[1B][0m
[1B][0;32mI (723) gpio: GPIO[45]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;31mE (723) ES8311: es8311_init(329): I2C read/write error[1B][0m
[1B][0;32mI (733) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (723) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (743) adc button: calibration scheme version is Line Fitting[1B][0m
[1B][0;32mI (753) adc button: Calibration Success[1B][0m
[1B][0;32mI (753) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (763) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (763) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (773) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (773) button: IoT Button Version: 2.5.1[1B][0m
[1B][0;32mI (783) LVGL: Starting LVGL task[1B][0m
[1B][0;32mI (783) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
btn handler 
[1B][0;31mE (793) ES8311: es8311_voice_volume_get(380): I2C read/write error[1B][0m

as
```

igrr, 2023-05-02:
@krupis The issue you are seeing seems to be related to I2C:
```
[1B][0;31mE (793) ES8311: es8311_voice_volume_get(380): I2C read/write error[1B][0m
```
— which is probably because the ESP8311 codec's I2C interface is connected to GPIOs 7 and 8. If you flip the "touch switch" for the corresponding GPIOs to "OFF", I2C interface will no longer be connected and I2C communication will fail.

Please try setting these 2 switches back on; as far as i can tell from the [touch extension board schematic diagram](https://dl.espressif.com/dl/schematics/ESP-LyraP-TouchA_V1.1_SCH_20200325A.pdf), GPIOs 7 and 8 aren't used to connect any touch pads, anyway. Only GPIOs 

[image]

Given this part of [Kaluga board schematic diagram](https://dl.espressif.com/dl/schematics/ESP32-S2-Kaluga-1_V1_3_SCH_20200526A.pdf), GPIOs 7 and 8 correspond to the 7th switch on SW3 and 1st on SW4.

krupis, 2023-05-03:
Thanks for clarifying

closed 2023-05-03 by igrr