espressif/esp-idf #11273: MODEM: Failed to connect to MQTTS broker

stdenits, 2023-04-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to connect to the local MQTTS server using GSM modem SIMCom A7672X (USB connection used) and ESP32S3, but it is failed.

Connection URL looks like: "mqtts://IP:PORT", where: IP - fixed (white) IP, PORT - opened portforwarded port at the internet provider terminal.

I have tested connection via modem to insecure MQTT broker - Mosquito. Works fine.

And test connection from PC (which connected to the Internet through Smartphone and do not have access to the LAN) to the local MQTTS server via MQTT Explorer and SSL certificates. Works fine.

Connection to secured MQTTS broker - Mosquito, failed with the same error.

Please, reply to: kaija78@gmail.com to get full log or entire project.

```
W (126253) esp-tls: Failed to open new connection in specified timeout
E (126253) transport_base: Failed to open a new connection
E (126273) mqtt_client: Error transport connect
I (126273) MQTT: MQTT_EVENT_ERROR
E (126273) MQTT: Last error reported from esp-tls: 0x8006
I (126273) MQTT: Last errno string 0 (Success)
I (126283) MQTT: MQTT_EVENT_DISCONNECTED
I (141283) MQTT: Other event id:7
```

stdenits, 2023-04-29:
@david-cermak,
Hello.

>Have you tried to increase the connection timeout? 
Did set it to 30000 ms. Did not help.

Attaching a slightly edited log of the mbed-tls session.
[mqtt_ssl.txt](https://github.com/espressif/esp-idf/files/11359478/mqtt_ssl.txt)

Please, tell me if you need more info.

david-cermak, 2023-05-02:
Thanks for sharing. It looks like the client has sent the handshake message successfully, but hasn't received any response (trying to `recv()` 5 bytes) and, after 30 seconds gave up with an error. Now we need to find out where did those 5 bytes get lost.

Could you also please share the (most verbose) logs from your local mosquito ?

> And test connection from PC (which connected to the Internet through Smartphone and do not have access to the LAN) to the local MQTTS server via MQTT Explorer and SSL certificates. 

Could you please also check the connection via WiFi station connecting to your Smartphone? This is just to rule out the PPPoS layers.

stdenits, 2023-05-03:
@david-cermak,

>Could you also please share the (most verbose) logs from your local mosquito?

Mosquito addon for Home Assistant tells something like that:

>New connection from <IP:PORT> on port 8883.
>Client <unknown> has exceeded timeout, disconnecting.

Not sure if it possible to get more detailed error message.

Attaching two logs file for scenario, when ESP32 connects to the smartphone access point (to get Internet connection) as station and trying for:
a) [mqtt_smartphone_ap.txt](https://github.com/espressif/esp-idf/files/11380686/mqtt_smartphone_ap.txt) - connect to insecure local MQTT server through Internet;
b) [mqtts_smartphone_ap.txt](https://github.com/espressif/esp-idf/files/11380687/mqtts_smartphone_ap.txt) - connect to secured local MQTTS server through Internet with SSL certificates.

Both cases are successful.

david-cermak, 2023-05-03:
Thanks again for the updates!
So, it works over WiFi with the same TLS authentication, but fails over PPPoS while waiting for a handshake reply from the server; and the server says:
```
Client has exceeded timeout, disconnecting.
```
(but the disconnection is not propagated to the client).

This is strange, it looks like some server side timeout (or some kind of interruption on the way to the client?). Could you please try to increase the baudrate and CPU frequency to see if it succeeds? 
Also would it be possible to share a packet capture (maybe just filtered to see only the TCP packets on 8883 port) on your local network?

stdenits, 2023-05-03:
>Could you please try to increase the baudrate and CPU frequency to see if it succeeds?

>I (398) cpu_start: cpu freq: 240000000 Hz

ESP32-S3 works at 240 MHz.

>Also would it be possible to share a packet capture (maybe just filtered to see only the TCP packets on 8883 port) on your local network?

Sure, but it will take some time - I am not familiar with network sniffers.
Will post LAN packet log as soon as possible.

stdenits, 2023-05-04:
@david-cermak,
Hi.

Got LAN log file for TCP protocol and port 8883 for scenario when ESP32 trying to connect to the MQTTS via GSM modem using SSL certificates.

If possible, could you write to the email address given in the OP-post. 
I would not like to expose certificate data to the public.

Or, if that's enough, I can provide some screenshots.

david-cermak, 2023-05-15:
@stdenits Could you please try to:

* Increase the baud rate to `230400` bauds?
* Increase the RTO to at least 3 seconds (`CONFIG_LWIP_TCP_RTO_TIME`=3000)
* Increase the priority of the UART task?

stdenits, 2023-05-16:
@david-cermak 
Debian OS reported that the initial TCP retransmit timeout setting is:
```
net.ipv4.tcp_retries1 = 3
```
(seconds)

I'm a bit confused with UART. GSM modem SimCOM A7672X connected to the ESP32 via USB interface, not UART.

david-cermak, 2023-05-16:
> net.ipv4.tcp_retries1 = 3

It seems like this is the number of re-transmits, not the initial RTO, also default to 3 (times, not seconds)

Based on your packet capture, the initial RTO is about 300ms (=the time after which the system retries if not ACK received).

> I'm a bit confused with UART.

Oh, apologies, forgot that this issue is about USB, not UART.
I'm sorry but, could you please rewire the modem to use UART interface and try again? Just to see if the connection succeeds and on which baudrate.

stdenits, 2023-05-18:
@david-cermak ,
> could you please rewire the modem to use UART interface and try again?
Sure.

GSM modem wired with UART did connect to the local MQTTS server without error.
UART speed: 115200 bps
CONFIG_LWIP_TCP_RTO_TIME = 1500 #default

Attaching redacted LOG file.
[uart_modem_to_mqtss.txt](https://github.com/espressif/esp-idf/files/11507814/uart_modem_to_mqtss.txt)

>Increase the RTO to at least 3 seconds (CONFIG_LWIP_TCP_RTO_TIME=3000)
Did no effect for USB connection: can not connect to MQTTS server.

closed 2023-10-05 by stdenits