espressif/esp-idf #11274: dhcpc wifi-sta+eth results in memory leak and constant crashes (endless dhcp_discover)

0xFEEDC0DE64, 2023-04-25:
We are using the ESP32 with WiFi (STA) and ethernet, and when both are connected and the dhcp client is started, I see some endless loop of dhcpc running (a lot of dhcp_discover messages). The free memory goes down very quickly from around 100k free RAM to 0 and a crash afterwards.

Why are there so many calls to dhcp_discover()? The problem goes away if disable dhcpc on the ethernet interface (static ip) or I don't connect the LAN cable to ethernet.

I have enabled some lwip log output to better show the strange endless loop.

The esp-idf version we are using is this: https://github.com/0xFEEDC0DE64/esp-idf/tree/old_fixes

The log output:

```
[feedc0de@kuischronk homecontroller-firmware]$ idf.py -p /dev/ttyACM0 -b 921600 monitor -b 115200 -E
Executing action: monitor
Running idf_monitor in directory <firmware-repo>
Executing "<my-home>/.espressif/python_env/idf5.1_py3.10_env/bin/python <firmware-repo>/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 3 <firmware-repo>/build_hw1/goecontroller_hw1.elf --encrypted -m '<my-home>/.espressif/python_env/idf5.1_py3.10_env/bin/python' '<firmware-repo>/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0' '-b' '921600'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
u up.
I (1394) cpu_start startup.c:425 start_cpu0_desh_write_prepare(): Erase flash 12288 bytes @ 0x60000
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:2, clock div:2
secure boot v2 enabled
secure boot verification succeeded
load:0x3fff00b8 len:0x3b78
load:0x40078000 len:0x6324
load:0x40080400 len:0x4
0x40080400: _init at ??:?

load:0x40080404 len:0xe68
entry 0x4008061c
I (1371) cpu_start cpu_start.c:424 call_start_cpu0(): Pro cpu up.
I (1371) cpu_start cpu_start.c:214 start_other_core(): Starting app cpu, entry point is 0x40081634
0x40081634: call_start_cpu1 at <firmware-repo>/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start cpu_start.c:176 call_start_cpu1(): App cpu up.
I (1395) cpu_start startup.c:425 start_cpu0_default(): Pro cpu start user code
I (1395) cpu_start startup.c:427 start_cpu0_default(): cpu freq: 240000000 Hz
I (1401) cpu_start startup.c:433 start_cpu0_default(): Application information:
I (1409) cpu_start startup.c:435 start_cpu0_default(): Project name:     goecontroller_hw1
I (1418) cpu_start startup.c:438 start_cpu0_default(): App version:      1.0.4-20-g422d8d8-dirty
I (1427) cpu_start startup.c:444 start_cpu0_default(): Compile time:     Apr 24 2023 16:44:36
I (1436) cpu_start startup.c:448 start_cpu0_default(): ELF file SHA256:  8dfe6f5f230b23c0...
I (1446) cpu_start startup.c:449 start_cpu0_default(): ESP-IDF:          v5.1-dev-3808-gc1729c0207-dirty
I (1456) cpu_start startup.c:451 start_cpu0_default(): Min chip rev:     v3.0
I (1464) cpu_start startup.c:452 start_cpu0_default(): Max chip rev:     v3.99 
I (1472) cpu_start startup.c:455 start_cpu0_default(): Chip rev:         v3.0
I (1480) heap_init heap_caps_init.c:90 heap_caps_init(): Initializing. RAM available for dynamic allocation:
I (1490) heap_init heap_caps_init.c:113 heap_caps_init(): At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1500) heap_init heap_caps_init.c:113 heap_caps_init(): At 3FFC11F8 len 0001EE08 (123 KiB): DRAM
I (1509) heap_init heap_caps_init.c:113 heap_caps_init(): At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1519) heap_init heap_caps_init.c:113 heap_caps_init(): At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1529) heap_init heap_caps_init.c:113 heap_caps_init(): At 4009780C len 000087F4 (33 KiB): IRAM
I (1539) spi_flash esp_flash_api.c:459 detect_spi_flash_chip(): detected chip: generic
I (1547) spi_flash esp_flash_api.c:321 esp_flash_init_main(): flash io: dio
W (1555) flash_encrypt flash_encrypt.c:67 esp_flash_encryption_init_checks(): Flash encryption mode is DEVELOPMENT (not secure)
I (1571) COREDUMP core_dump_flash.c:75 esp_core_dump_flash_init(): Init core dump to flash
I (1576) COREDUMP core_dump_flash.c:81 esp_core_dump_flash_init(): Found partition 'test'
I (1584) COREDUMP core_dump_flash.c:82 esp_core_dump_flash_init(): Found partition '0x3ffc2dd0'
I (1594) COREDUMP core_dump_flash.c:83 esp_core_dump_flash_init(): Found partition '0x3ffc2de8'
I (1603) COREDUMP core_dump_flash.c:84 esp_core_dump_flash_init(): Found partition 'coredump' @ 60000 65536 bytes
E (1614) COREDUMP core_dump_flash.c:504 esp_core_dump_partition_and_size_get(): Incorrect size of core dump image: -944226484
I (1627) app_start app_startup.c:99 esp_startup_start_app(): Starting scheduler on CPU0
I (1635)[0pp_mI rt63pp_mairtuask:app esarttarc:1_stmai_apasoth: Scorte( :nSCaU0i[gmscheduler on CPU1
I (1645) main_task app_startup.c:206 main_task(): Calling app_main()
E (1645) MAIN main.cpp:84 app_main(): sizeof(api)=4828 sizeof(configs)=11376
I (1655) CONFIG configmanager_priv.h:60 init(): nvs_flash_read_security_cfg() returned: ESP_OK
[... 971 lines cut ...]
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "<my-home>/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/__main__.py", line 4, in <module>
    main()
  File "<my-home>/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/idf_monitor.py", line 378, in main
    monitor.main_loop()
  File "<my-home>/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/idf_monitor.py", line 172, in main_loop
    self._main_loop()
  File "<my-home>/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/idf_monitor.py", line 278, in _main_loop
    super()._main_loop()
  File "<my-home>/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/idf_monitor.py", line 212, in _main_loop
    self.serial_handler.handle_commands(data, self.target, self.run_make, self.console_reader,
  File "<my-home>/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/base/serial_handler.py", line 202, in handle_commands
    console_reader.stop()
  File "<my-home>/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/base/stoppable_thread.py", line 56, in stop
    self._cancel()
  File "<my-home>/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/base/console_reader.py", line 86, in _cancel
    fcntl.ioctl(self.console.fd, termios.TIOCSTI, b'\0')
OSError: [Errno 5] Input/output error
idf_monitor failed with exit code 1
[feedc0de@kuischronk homecontroller-firmware]$ 
```

0xFEEDC0DE64, 2023-04-26:
I had some success mitigating this issue by rate limiting the sending of DHCP_DISCOVER messages, but it is a hacky code and changes the lwip implementation.

Are we allowed to change code from 2003 in lwip and pull-request it to espressif?

binary1230, 2023-04-26:
I ran smack into this exact issue the other day. It's been fixed in upstream LWIP
https://github.com/espressif/esp-lwip/issues/53

This is a serious issue that's present in IDF v5.0.1, is there any estimate on the next release or a hotfix for this? I've run into this with production boards

0xFEEDC0DE64, 2023-05-12:
Fixed since our latest esp-idf rebase onto master

closed 2023-05-12 by 0xFEEDC0DE64

AxelLin, 2023-05-15:
> This is a serious issue that's present in IDF v5.0.1, is there any estimate on the next release or a hotfix for this? I've run into this with production boards

v5.0.2 inclues this fix: 62983624aec0a1eb45712170fa317255bf5219ec

AxelLin, 2023-05-20:
@david-cermak @freakyxue 
v4.4 and v4.3 also need fix.

Alvin1Zhang, 2023-06-20:
Thanks for reporting and sorry for the slow turnaround, fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/61a0c12c29c1920738d05c0db3f6c174eef04542.

AxelLin, 2023-06-20:
@Alvin1Zhang 
But the fix for v4.3 is not yet available. (Needs to update esp-lwip submodule in v4.3 branch)