espressif/esp-idf #11277: Support for LAN9303 ethernet module

krupis, 2023-04-26:
### Is your feature request related to a problem?

Im always frustrated when dealing with ETH in esp32, I have to connect each ESP32 device via separate LAN cable. For large scale project, this can be real pain.

### Describe the solution you'd like.

Currently, the supported Ethernet IC's  are:
IP101
RTL8201
LAN87XX
DP83848

[image]

All these Ethernet IC's must be connected in star connection fashion (each device must be separately connected to a router or a switch)

From what I know, there is an ethernet IC's LAN9303 that allows user to connect other devices in daisy chain fashion thus simplifying overall system design (shorter cables needed to connect devices to the internet if devices are nearby).

[image]

### Describe alternatives you've considered.

IP101
RTL8201
LAN87XX
DP83848

suda-morris, 2023-04-26:
FYI, we have support of another Switch driver: https://github.com/espressif/esp-eth-drivers/blob/master/ksz8863/README.md

I would recommend you to open the feature request in that repo

kostaond, 2023-04-26:
You can also consider realize daisy-chaining connection using SW bridge and SPI Ethernet modules if your application does not require high bandwidth network, see https://github.com/espressif/esp-idf/tree/master/examples/network/bridge

krupis, 2023-04-26:
> FYI, we have support of another Switch driver: https://github.com/espressif/esp-eth-drivers/blob/master/ksz8863/README.md
> 
> I would recommend you to open the feature request in that repo

Thanks for quick reply.

I never understood how to properly use component registry. Can you clarify this to me once and for all?

**What I tried so far**

1. Open ESP-IDF 5.0 CMD (esp-idf command line tool)
2. Execute command:
```
idf.py create-project-from-example "espressif/ksz8863^0.2.2:simple_switch"
```
The command output:
```
C:\Espressif\frameworks\esp-idf-v5.0.1>idf.py create-project-from-example "espressif/ksz8863^0.2.2:simple_switch"
Executing action: create-project-from-example
Example "simple_switch" successfully downloaded to c:\espressif\frameworks\esp-idf-v5.0.1\simple_switch
Done
```

3. I have opened this project and try to build it:
```
-- Found Git: C:/Users/petrikas.lu/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/petrikas.lu/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/petrikas.lu/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/petrikas.lu/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/build.cmake:519 (message):
  ERROR: The override_path you're using is pointing to directory
  "C:\Espressif\frameworks" that is not a component.

Call Stack (most recent call first):
  C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v5.0.1/simple_switch/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -S c:\Espressif\frameworks\esp-idf-v5.0.1\simple_switch -B c:\Espressif\frameworks\esp-idf-v5.0.1\simple_switch\build" terminated with exit code: 1. 
 * ```

Not too sure what is this error about

kostaond, 2023-04-26:
Try to remove `override_path` from `idf_component.yml` file.

igrr, 2023-04-26:
@kostaond perhaps we can tag a new version of the component so that it gets re-uploaded with `override_path` removed? Maybe the issue is that the last version was uploaded by a slightly too old idf-component-manager which didn't strip `override_path` from examples...

kostaond, 2023-04-26:
> @kostaond perhaps we can tag a new version of the component so that it gets re-uploaded with override_path removed? Maybe the issue is that the last version was uploaded by a slightly too old idf-component-manager which didn't strip override_path from examples...

Interesting, I'll make the update. Thanks for help.

kostaond, 2023-04-28:
@krupis please consider using KSZ8863 or multiple SPI Ethernet modules in bridge mode.

Since solution already exists, I am closing this issue. Please feel free to reopen if you don't find it sufficient. However, please provide strong justification why to add support of LAN9303.

closed 2023-04-28 by kostaond