espressif/esp-idf #11278: Flash Encryption in development Mode

AvinashSinghPES, 2023-04-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

we are using flash encryption in development mode. but one problem occur during runtime application ,when initialized the spiffs that time got a error "****spiffs mount failed : -10025 formatting****" after that the application starts crashing or controller restart again and again with same problem.

this is the partition table:

# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     0x11000,  0x6000,
nvs_key,    data, nvs_keys, , 0x1000, encrypted,
phy_init, data, phy,     ,  0x1000,
factory,  app,  factory, , 1M,
otadata, data, ota,   ,  0x2000,
ota_0, app, ota_0,   , 0x180000,
ota_1, app, ota_1,   , 0x180000,
storage,  data, spiffs,  ,3M,

Please provide me some solution regarding this issue.

Thanks,
Avinash Singh

igrr, 2023-04-26:
@AvinashSinghPES a couple of questions:
1. Are you uploading a SPIFFS partition image from host, or are you just starting with an empty partition?
2. If you are uploading SPIFFS partition image from host, can you describe how you are doing this? 
3. Can you post the code you are using to initialize SPIFFS in your application?
4. Which IDF version are you using (check with `idf.py --version`) and which chip are you using?

igrr, 2023-04-26:
@AvinashSinghPES I get the following output with the code you have shared:

On first run, after flashing:
```
I (273) example: Initializing SPIFFS
W (273) SPIFFS: mount failed, -10025. formatting...
I (8503) example: Performing SPIFFS_check().
I (12933) example: SPIFFS_check() successful
I (12933) example: Partition size: total: 896321, used: 0
I (12933) example: Opening file
I (13033) example: File written
I (13073) example: Renaming file
I (13133) example: Reading file
I (13133) example: Read from file: 'Hello World!'
I (13133) example: SPIFFS unmounted
```

On second run
```
I (273) example: Initializing SPIFFS
I (383) example: Performing SPIFFS_check().
I (4813) example: SPIFFS_check() successful
I (4813) example: Partition size: total: 896321, used: 502
I (4813) example: Opening file
I (4913) example: File written
I (4913) example: Renaming file
I (4973) example: Reading file
I (4973) example: Read from file: 'Hello World!'
I (4973) example: SPIFFS unmounted
```

Do you see something different? If yes, please post the complete log output you get.

AvinashSinghPES, 2023-04-26:
@igrr  yes, there is some difference . check my log  output :
[ESP_logs.txt](https://github.com/espressif/esp-idf/files/11332749/ESP_logs.txt)

igrr, 2023-04-26:
I see, from your log it looks like a reset happens, but there is no information in the log about the reset cause. Here are some more troubleshooting suggestions for you:
1. Try to get the output from ROM bootloader, it will show the reset reason. Currently there is no ROM output visible. This could be because ROM output is disabled by pulling GPIO 15 low, or because your application is configured to use a different UART baud rate than the default one (115200). See if you can change that, so that the ROM bootloader output is visible, then check the reset reason printed by the ROM.
   - If you still can't get the reset reason from ROM output, call `esp_reset_reason()` function at application startup and see which value it returns.
2. Try the example project bundled with esp-idf, examples/storage/spiffs, enabling flash encryption and flashing it to the same hardware? Is the same issue observed?
3. Is your program initializing anything before initializing SPIFFS? You can try to gradually simplify the program until it starts working, which can help isolate the part of the code or a configuration which leads to this issue.

AvinashSinghPES, 2023-04-26:
@igrr  we reprogrammed the controller and just happens this :

1.
 rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
invalid header: 0x621e56bd
ets Jul 29 2019 12:21:46

2. and not getting anything over serial UART printing. previously it was printing logs which we already shared with you. now it is completely blank.

AvinashSinghPES, 2023-04-27:
Hi @mahavirj  please check this issue, we are using flash encryption in development mode but when we try to reprogram show some error which already mention above.

mahavirj, 2023-04-27:
Please try the suggestion I had provided in my earlier [comment](https://github.com/espressif/esp-idf/issues/11212#issuecomment-1511147322). If you are using Eclipse IDE then you can update the command as shown [here](https://github.com/espressif/idf-eclipse-plugin#customizeLaunchConfig).

mahavirj, 2023-05-05:
@AvinashSinghPES Did this suggestion help you? If yes, please close the issue.

closed 2023-05-05 by AvinashSinghPES