espressif/esp-idf #11279: After reset, HTTP server takes more than one second between sending blocks of data

CleberPeter, 2023-04-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## Environment

* Development Kit: Custom Board
* Module or chip used: ESP32-WROOM-32UE
* IDF version (run git describe --tags to find it): v5.0
* Build System: idf.py
* Compiler version (run xtensa-esp32-elf-gcc --version to find it): 12.2.0
* Operating System: Ubuntu 22.04
* Using an IDE?: Yes (Visual StudioCode)
* Power Supply: External Source

### Custom Board

* Network Interface: A LAN8720 phy whose RMII bus clock is provided by IO17 (CLK180) and phy reset is connected to IO16.
* Filesystem: A 4-wire bus sdcard with a FAT file system with a single folder containing the web application files.

## Problem Description

Initially, the server acts adequately, even when sending large files. Although after a few reset cycles, the HTTP server takes more than one second between sending data blocks.

### Expected Behavior

That the http server maintains performance no matter how many times the device has been restarted.

## Steps to reproduce

1. Build and run **restful_server** **example** with this configs:

```
#define CONFIG_EXAMPLE_WEB_DEPLOY_SD 1
#define CONFIG_EXAMPLE_CONNECT_ETHERNET 1
#define CONFIG_EXAMPLE_ETHERNET_EMAC_TASK_STACK_SIZE 2048
#define CONFIG_EXAMPLE_USE_INTERNAL_ETHERNET 1
#define CONFIG_EXAMPLE_ETH_PHY_LAN87XX 1
#define CONFIG_EXAMPLE_ETH_MDC_GPIO 23
#define CONFIG_EXAMPLE_ETH_MDIO_GPIO 18
#define CONFIG_EXAMPLE_ETH_PHY_RST_GPIO 16
#define CONFIG_EXAMPLE_ETH_PHY_ADDR 0
#define CONFIG_EXAMPLE_CONNECT_IPV4 1
```
And this ethernet configs:
```
#define CONFIG_ETH_RMII_CLK_OUTPUT 1
#define CONFIG_ETH_RMII_CLK_OUT_GPIO 17
```
2. After, placing at least one file on the SDCARD with a size greater than 1.7MB and perform an http-request with the GET method for the file name present on the card.
3. Monitor the time taken to receive all the file. 
4. If the file download takes less than 4s restart the esp32 and redo step 3, sometimes multiple resets are needed to show the problem.

## Debug Logs

In the wireshark capture below, it is possible to verify two delays 9.95 ~ 11.26 and 11.27 ~ 12.79, that is, during this period no data was sent to the client. This delay is repeated from time to time for each request requested by the client until a new reset is performed.

It is important to emphasize that the problem is intermittent, that is, sometimes multiple resets are necessary for it to manifest itself.

[image]

* Full wireshark log:

[log_wireshark.zip](https://github.com/espressif/esp-idf/files/11335754/log_wireshark.zip)

The following log tech shows a time greater than 2s (16356 ~ 18616) between sending two consecutive blocks of 10240 bytes:

```
D (16356) httpd_txrx: httpd_send_all: sent = 10240
D (16356) httpd_txrx: httpd_send_all: sent = 2
D (16366) esp.emac: receive len= 60
D (16376) httpd_txrx: httpd_send_all: sent = 6
D (16376) esp.emac: receive len= 60
D (16416) esp.emac: receive len= 99
D (16466) esp.emac: receive len= 92
D (16606) esp.emac: receive len= 60
D (16606) esp.emac: receive len= 92
D (16706) esp.emac: receive len= 75
D (16706) esp.emac: receive len= 95
D (16706) esp.emac: receive len= 75
D (16706) esp.emac: receive len= 95
D (16706) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbc068
D (16716) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbc068
D (16716) esp.emac: receive len= 698
D (16796) esp.emac: receive len= 216
D (16796) esp.emac: receive len= 217
D (16826) esp.emac: receive len= 60
D (16866) esp.emac: receive len= 75
D (16866) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbc068
D (16866) esp.emac: receive len= 95
D (16866) esp.emac: receive len= 75
D (16866) esp.emac: receive len= 95
D (16876) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbc068
D (16876) esp.emac: receive len= 92
D (16906) esp.emac: receive len= 211
D (16946) esp.emac: receive len= 60
D (16956) esp.emac: receive len= 60
D (17026) esp.emac: receive len= 60
D (17116) esp.emac: receive len= 64
D (17366) esp.emac: receive len= 92
D (17386) esp.emac: receive len= 92
D (17506) esp.emac: receive len= 60
D (17516) esp.emac: receive len= 60
D (17576) esp.emac: receive len= 216
D (17616) esp.emac: receive len= 217
D (17636) esp.emac: receive len= 92
D (17766) esp.emac: receive len= 718
D (17826) esp.emac: receive len= 60
D (17956) esp.emac: receive len= 60
D (18126) esp.emac: receive len= 92
D (18136) esp.emac: receive len= 75
D (18136) esp.emac: receive len= 95
D (18136) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbc068
D (18136) esp.emac: receive len= 75
D (18136) esp.emac: receive len= 95
D (18146) esp.emac: receive len= 89
D (18146) esp.emac: receive len= 69
D (18156) esp.emac: receive len= 89
D (18156) esp.emac: receive len= 69
D (18156) esp.emac: receive len= 92
D (18166) esp.emac: receive len= 60
D (18166) esp.emac: receive len= 0
D (18166) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbc068
D (18296) esp.emac: receive len= 60
D (18316) esp.emac: receive len= 171
D (18366) esp.emac: receive len= 60
D (18396) esp.emac: receive len= 64
D (18426) esp.emac: receive len= 75
D (18426) esp.emac: receive len= 95
D (18426) esp.emac: receive len= 75
D (18426) esp.emac: receive len= 95
D (18426) esp.emac: receive len= 89
D (18436) esp.emac: receive len= 69
D (18436) esp.emac: receive len= 89
D (18436) esp.emac: receive len= 69
D (18446) esp.emac: receive len= 0
D (18446) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbc068
D (18456) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffbc068
D (18546) esp.emac: receive len= 89
D (18546) esp.emac: receive len= 89
D (18546) esp.emac: receive len= 69
D (18546) esp.emac: receive len= 69
D (18556) esp.emac: receive len= 0
D (18576) esp.emac: receive len= 216
D (18606) esp.emac: receive len= 60
D (18616) esp.emac: receive len= 60
D (18616) esp.emac: receive len= 60
D (18616) esp.emac: receive len= 60
D (18616) httpd_txrx: httpd_send_all: sent = 10240
D (18616) httpd_txrx: httpd_send_all: sent = 2
D (18626) esp.emac: receive len= 217
D (18626) httpd_txrx: httpd_send_all: sent = 6
D (18646) esp.emac: receive len= 60
D (18646) esp.emac: receive len= 60
D (18646) esp.emac: receive len= 60
D (18646) esp.emac: receive len= 60
D (18646) esp.emac: receive len= 60
```
* Full debug log:

[full_debug_log.txt](https://github.com/espressif/esp-idf/files/11335887/debug_log.txt)

## Additional Informations

* I tested this same situation in the 4.4.1 version of espidf and found the same problem;
* I put a GPIO manipulation before and after the fread executed in the sdcard reading to verify if the delay was due to the reading of the files. However, I confirmed that the reading is taking place correctly without delays for all tests.
* I tested the **iperf example** several times and got an average rate of 75Mbps without any kind of delay.

I also made a dedicated application with the following behavior:

* receives the http request and reads the file in blocks of 10240, but sends only six bytes to the client.

```
uint8_t chunk[WEBSERVER_CHUNK_SIZE];
FILE* fp = fopen(file_path, "r");
if (fp)
{
    do
    {
        chunk_size = fread(chunk, sizeof(char), WEBSERVER_CHUNK_SIZE, fp);
        
        if (chunk_size)
        {
            gpio_set_level(5, 0);
            send(socket, chunk, 6);
            gpio_set_level(5, 1);
        }
    } while (chunk_size > 0);

    fclose(fp);
}
```
To my surprise I could check the same problem no data send to client in 4.7 ~ 5.75, wireshark log below:

[image]

I could also verify that the pin is kept at a low level during the delay, that is, the application blocks the send method call.

I also tried to close the file (fclose) before sending the data (send), but it had no effect.

Having said that, the problem seems to be related to the reading of files from the card and the consequent sending of any data via tcp/ip.

Is there any other log that I can generate to try to identify the problem?

Thanks in advance for any help.

ESP-YJM, 2023-05-04:
@CleberPeter We need check the delay is in ethernet protocol stack or lwip(or higher HTTP) stack. You can add this pacth and check the log. The patch will print TCP flow and you can compare the captured packets and clearly see the time spent from the upper layer to the lower layer.
[print_tcp_flow.patch](https://github.com/espressif/esp-idf/files/11392678/print_tcp_flow.patch)

CleberPeter, 2023-05-05:
Hello @ESP-YJM, thanks for the feedback.

I isolated the HTTP stack problem, since this is verified even over an ordinary socket connection. The code block below reads a 1.7 MB file in 1024B blocks and sends, for each block read, only 1 byte to a TCP client.

```
int chunk_size;
FILE* fp = fopen(file_path, "r");
if (fp)
{
    do
    {
        chunk_size = fread(chunk, sizeof(char), 1024, fp);
        if (chunk_size)
        {
            gpio_set_level(5, 0);
            tcp_server_send(socket, (uint8_t *) chunk, 1);
            gpio_set_level(5, 1); 
        }
    } while (chunk_size > 0);
    fclose(fp);
}
```
Result:

[image]

[log_wireshark_1B.zip](https://github.com/espressif/esp-idf/files/11408410/log_wireshark_1B.zip)

During the 1.4s delay without sending data to the phy, the stack stores the data and after unlocking it sends it in a single packet of 816B (packet 329). Same case on packet 352.

I performed the same tests by disabling the nagle's algorithm (TCP_NODELAY = 1):

[image]

[log_wireshark_nodelay.zip](https://github.com/espressif/esp-idf/files/11408455/log_wireshark_nodelay.zip)

In this case we can see the same delay, but without buffering.

Result of applying the patch:

```
I (10619) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (10629) TEST: @@ WiFi Tx TCP - L:41, S:3503347889, A:1012056700, SP:80, DP:37640, F:18
I (10629) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503347890, SP:37640, DP:80, F:10
I (10639) TEST: @@ WiFi Tx TCP - L:45, S:3503347890, A:1012056700, SP:80, DP:37640, F:18
I (10649) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503347895, SP:37640, DP:80, F:10
I (10659) TEST: @@ WiFi Tx TCP - L:41, S:3503347895, A:1012056700, SP:80, DP:37640, F:18
I (10669) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503347896, SP:37640, DP:80, F:10
I (10679) TEST: @@ WiFi Tx TCP - L:44, S:3503347896, A:1012056700, SP:80, DP:37640, F:18
I (10689) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503347900, SP:37640, DP:80, F:10
I (10699) TEST: @@ WiFi Tx TCP - L:45, S:3503347900, A:1012056700, SP:80, DP:37640, F:18
I (11839) TEST: @@ WiFi Tx TCP - L:45, S:3503347900, A:1012056700, SP:80, DP:37640, F:18
I (11839) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503347905, SP:37640, DP:80, F:10
I (11839) TEST: @@ WiFi Tx TCP - L:792, S:3503347905, A:1012056700, SP:80, DP:37640, F:18
I (11849) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503348657, SP:37640, DP:80, F:10
I (11859) TEST: @@ WiFi Tx TCP - L:45, S:3503348657, A:1012056700, SP:80, DP:37640, F:18
I (11869) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503348662, SP:37640, DP:80, F:10
I (11879) TEST: @@ WiFi Tx TCP - L:44, S:3503348662, A:1012056700, SP:80, DP:37640, F:18
I (11889) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503348666, SP:37640, DP:80, F:10
I (11899) TEST: @@ WiFi Tx TCP - L:41, S:3503348666, A:1012056700, SP:80, DP:37640, F:18
I (11899) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503348667, SP:37640, DP:80, F:10
I (11909) TEST: @@ WiFi Tx TCP - L:44, S:3503348667, A:1012056700, SP:80, DP:37640, F:18
I (11919) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503348671, SP:37640, DP:80, F:10
I (11929) TEST: @@ WiFi Tx TCP - L:45, S:3503348671, A:1012056700, SP:80, DP:37640, F:18
I (13339) TEST: @@ WiFi Tx TCP - L:45, S:3503348671, A:1012056700, SP:80, DP:37640, F:18
I (13339) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503348676, SP:37640, DP:80, F:10
I (13339) TEST: @@ WiFi Tx TCP - L:931, S:3503348676, A:1012056700, SP:80, DP:37640, F:18
I (13349) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503349567, SP:37640, DP:80, F:10
I (18839) TEST: @@ WiFi Tx TCP - L:40, S:3503349566, A:1012056700, SP:80, DP:37640, F:10
I (18839) TEST: @@ WiFi Rx TCP - L:40, S:1012056700, A:3503349567, SP:37640, DP:80, F:10
```
[image]

[log_wireshark_with_patch.zip](https://github.com/espressif/esp-idf/files/11408616/log_wireshark_with_patch.zip)

The impression I have is that the blocking is taking place at a higher layer, since when the stack is passed on so that the information can be transmitted, the data has already been buffered (10699 ~ 11839) and (11929 ~ 13339).

ESP-YJM, 2023-05-08:
[image]
From the patch log, it seems that the 1.2s is due to TCP retransmission. So maybe some reason make the ethernet to loss the TCP packet and LWIP will retransmite it after about 1.2s.

CleberPeter, 2023-05-08:
@ESP-YJM thank you very much for your support.

I understand, however as reported here: https://github.com/espressif/esp-idf/issues/6607#issuecomment-788651636 the packet retransmission time is calculated dynamically according to the network speed.

Having said that, and considering that before the delay, the client's response time is in the order of microseconds, is the retransmission time of 1.2 seconds reasonable?

Furthermore, considering that the problem is only verified while there are sdcard readings is it plausible to assume that this is causing esp32 to lose packets received by the LAN8720?

ESP-YJM, 2023-05-09:
@CleberPeter I think the retransmission depends on tcp_slowtmr. It execute every 2 * 250 ms. So the retransmission interval will cost lots time. So 1.2s is reasonable for LWIP but not for retransmission. 
If verified the packet loss is due to sdcard readings, you can re-rasie a issue to seek for help. I'm not very familiar with ethernet protocol stack.

ESP-YJM, 2023-06-13:
@CleberPeter Any update for this issue? I will close it, and feel free to re-open it if you have some update.

closed 2023-06-13 by ESP-YJM