espressif/esp-idf #11280: BLE connection fails when WiFi coexistence is enabled for ESP32-C3

ynormend, 2023-04-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-DevKitM-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

Any BLE central can connect to the board configured as BLE peripheral then interact with the GATT server.

### What is the actual behavior?

The connection fails right after having been established.

### Steps to reproduce.

1. Build and flash the `bluetooth/nimble/bleprph_wifi_coex` example
2. Wait for the board to start BLE advertisement after it has connected to the WiFi AP.
3. Use a BLE utility tool to discover the board and connect to it. I'm using the "BLE Scanner 4.0" app on macOS 13 (Apple M1 chip). The connection is established then fails immediately after (it repeats until the utility tool gives up).

### Debug Logs.

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16b8
load:0x403cc710,len:0x944
load:0x403ce710,len:0x2dfc
entry 0x403cc710
I (30) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (30) boot: compile time 10:03:18
I (30) boot: chip revision: v0.3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=181a0h ( 98720) map
[... 271 lines cut ...]
I (16708) wifi_prph_coex: our_id_addr_type=0 our_id_addr=7c:df:a1:76:53:86
I (16708) wifi_prph_coex: peer_ota_addr_type=0 peer_ota_addr=a0:78:17:99:84:db
I (16718) wifi_prph_coex: peer_id_addr_type=0 peer_id_addr=a0:78:17:99:84:db
I (16728) wifi_prph_coex: conn_itvl=12 conn_latency=0 supervision_timeout=36 encrypted=0 authenticated=0 bonded=0
64 bytes from 93.184.216.34 icmp_seq=16 ttl=49 time=247 ms
I (17828) wifi_prph_coex: disconnect; reason=520
I (17838) wifi_prph_coex: handle=1 our_ota_addr_type=0 our_ota_addr=7c:df:a1:76:53:86
I (17838) wifi_prph_coex: our_id_addr_type=0 our_id_addr=7c:df:a1:76:53:86
I (17848) wifi_prph_coex: peer_ota_addr_type=0 peer_ota_addr=a0:78:17:99:84:db
I (17848) wifi_prph_coex: peer_id_addr_type=0 peer_id_addr=a0:78:17:99:84:db
I (17858) wifi_prph_coex: conn_itvl=12 conn_latency=0 supervision_timeout=36 encrypted=0 authenticated=0 bonded=0
I (17868) NimBLE: GAP procedure initiated: advertise;
I (17878) NimBLE: disc_mode=2
I (17878) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (17888) NimBLE:

64 bytes from 93.184.216.34 icmp_seq=17 ttl=49 time=279 ms
64 bytes from 93.184.216.34 icmp_seq=18 ttl=49 time=305 ms
64 bytes from 93.184.216.34 icmp_seq=19 ttl=49 time=320 ms
64 bytes from 93.184.216.34 icmp_seq=20 ttl=49 time=253 ms
```

### More Information.

The issue cannot be reproduced with an ESP32-S3 board, or without WiFi coexistence (`bleprph` example), or with ESP-IDF v4.4.3.

wmy-espressif, 2023-05-04:
Hi @ynormend  thanks for your bug report. We have reproduced this issue locally and our engineers will investigate in this issue. 
From our analysis, this issue only occurs on new version iphones that support 2Mbps PHY, BLE disconnects after the PHY rate changes from 1Mbps to 2Mbps.

You can try another old version iphone or some other Android smartphones and test whether the issue does not occur, if yes, then it is likely that we are facing the same issue.

ynormend, 2023-05-05:
Thanks for your response @wmy-espressif ! I tried with an Android phone and it didn't occur indeed, so we're likely facing the same issue.

kobbok, 2023-05-05:
I think I'm also experiencing the same issue but with the ESP32-C3-DevKitC-02 using esp-idf v5.0.1 and the same bluetooth/nimble/bleprph_wifi_coex example. Neither an iPhone 12 (iOS 16.3) nor an iPad Air (4th generation, iOS 16.2) can connect to the peripheral. In my case the iOS device never connects and stays in the "connecting" state.
Android works without any problems.
If I switch to using the Bluedroid host the device connects then the connection fails immediately.

What seems to fix the problem is adding a ~2s delay after a successful connection to the Wifi and before starting advertising.

AxelLin, 2023-05-18:
> Hi @AxelLin thanks for the suggestion. In fact that we have found that it is the change in libphy.a that causes this issue,

@wmy-espressif Any update? The stable branches have this bug now, it needs fix ASAP.

wmy-espressif, 2023-05-18:
Hi @ynormend  @kobbok @AxelLin , we have founded the cause of this issue, and our bugfix is merged into master branch already and it can take some days to be synchronized to esp-idf github repo, also we are preparing the backported MRs on previous release branches. Before the MRs are merged and synchronized to github, you can use the following patch in your project:

[phy_lib_patch.tar.gz](https://github.com/espressif/esp-idf/files/11503919/phy_lib_patch.tar.gz)

To apply the patch, please use the library files in the attachment to replace the ones in `esp-idf/components/esp_phy/lib`.

AxelLin, 2023-05-18:
> Hi @ynormend @kobbok @AxelLin , we have founded the cause of this issue, and our bugfix is merged into master branch 

@wmy-espressif 
Be aware that most people don't use master for real projects.
It's important to get the fix for stable branches, which has impact on real projects.

wmy-espressif, 2023-05-18:
@AxelLin thanks for the reminder, the backports are in progress.

ynormend, 2023-05-19:
Thanks for the patch @wmy-espressif ! I noticed it contains libraries for both ESP32-C3 and ESP32-S3. Does it mean both platforms have the issue? I didn't observe it with ESP32-S3.

wmy-espressif, 2023-05-19:
Hi @ynormend yes the libraries are for ESP32-C3 and ESP32-S3, the two chips have similar implementations on RF/modem. I don't know why ESP32-S3 doesn't reproduce the issue. According to our RF engineers, the cause is that the filter bandwidth configured is too narrow for BLE 2M PHY, therefore I guess this patch is good for the performance of 2M PHY.

I would like to inform you that our fixes are already merged in the following branches: master, release/v4.4, release/v5.1, but currently only release/v5.1 is updated to github with this fix, the git commit SHA is: e8dba711ff93e4d93febcb7155e3b09f96ede1cb.  It can take several days for the other branches to be synchronized.

wmy-espressif, 2023-06-13:
Hi @ynormend , @kobbok , @AxelLin , for your information, the fixes for this issue have been merged into our release branches:

<html>
<body>

ESP-IDF Branch | Fixed Commit ID | Fixed ESP-IDF Version
-- | -- | --
master | 5838c2ec | expected in v5.2
release/v5.1 | e8dba711 | expected in v5.1
release/v5.0 | 5a68c496 | expected in v5.0.3
release/v4.4 | 1071b13c | expected in v4.4.5
release/v4.3 | 3a10cc9e | expected in v4.3.6

Please try out the specific commits you need.

Alvin1Zhang, 2023-06-14:
Thanks for reporting.

closed 2023-06-14 by Alvin1Zhang