espressif/esp-idf #11282: How do I Configure the Dual antenna of ESP 32 WROOM DA module in Auto mode

VaibhavSingh9511, 2023-04-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have been trying to test the signal strength of an ESP 32 Wroom DA module using its dual antenna, but I am unable to set them in auto mode. As in auto mode, the antenna itself will be configured as per the signal strength unavailable. I have tried to do that ( const wifi_ant_config_t ) but I haven't found much of a difference in the RSSI strength between a regular module and a DA module.

```
void wifi_init_sta(void)
{
    s_wifi_event_group = xEventGroupCreate();

    ESP_ERROR_CHECK(esp_netif_init());

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    esp_event_handler_instance_t instance_any_id;
    esp_event_handler_instance_t instance_got_ip;
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_any_id));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
                                                        IP_EVENT_STA_GOT_IP,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_got_ip));
    // esp_wifi_set_ant_gpio(5, 18);
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS,
            /* Setting a password implies station will connect to all security modes including WEP/WPA.
             * However these modes are deprecated and not advisable to be used. Incase your Access point
             * doesn't support WPA2, these mode can be enabled by commenting below line */
            .threshold.authmode = WIFI_AUTH_WPA2_PSK,
        },
    };

    const wifi_ant_config_t wifi_ant_config = {
        .rx_ant_mode = WIFI_ANT_MODE_ANT0,
        .rx_ant_default = WIFI_ANT_ANT0,
        .tx_ant_mode = WIFI_ANT_MODE_ANT1,
        .enabled_ant0 = 1,
        .enabled_ant1 = 0

    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_set_ant(&wifi_ant_config));

    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(TAG, "wifi_init_sta finished.");

    /* Waiting until either the connection is established (WIFI_CONNECTED_BIT) or connection failed for the maximum
     * number of re-tries (WIFI_FAIL_BIT). The bits are set by event_handler() (see above) */
    EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
                                           WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
                                           pdFALSE,
                                           pdFALSE,
                                           portMAX_DELAY);

    /* xEventGroupWaitBits() returns the bits before the call returned, hence we can test which event actually
     * happened. */
    if (bits & WIFI_CONNECTED_BIT)
    {
        ESP_LOGI(TAG, "connected to ap SSID:%s password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    }
    else if (bits & WIFI_FAIL_BIT)
    {
        ESP_LOGI(TAG, "Failed to connect to SSID:%s, password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    }
    else
    {
        ESP_LOGE(TAG, "UNEXPECTED EVENT");
    }

    /* The event will not be processed after unregister */
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip));
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
    vEventGroupDelete(s_wifi_event_group);
}
```

Can someone please suggest me how can I configure the antenna correctly? As of now I not really seeing much of a difference between the two, say DA module gives me a better RSSI value by a margin of 5 to 10 Dbi points, which I don't think is much. I have tried to configure the antennas on the DA separately as well as together but the result is almost the same.

VaibhavSingh9511, 2023-04-28:
@nachiketkukade Can you please help me with this ??

nachiketkukade, 2023-05-08:
@VaibhavSingh9511 , pls check [Antenna Switch Example](https://github.com/espressif/esp-idf/blob/master/examples/wifi/antenna/main/antenna_switch_example_main.c) for how to use multiple antennas

VaibhavSingh9511, 2023-05-10:
Hello @nachiketkukade Sir.  Thank you for the support provided earlier. I will surely try to test that code as well.  Last week, I conducted a comparison between the signal strength of a single PCB antenna connected to ESP 32 and a dual antenna connected ESP32 WROOM DA module. The objective of this test was to identify the module with better signal strength for our product.

To conduct this test, we configured the two modules to scan nearby wifi using an ardArduinouino example code. We configured the antennas in the DA module in AUTO mode as per the Arduino ide example code and placed the two modules next to each other.

After flashing the code onto the DA module, we observed that the DA module was able to sense a few extra numbers of wifi networks in comparison to the normal ESP32 with a single PCB-connected antenna. However, there wasn't a significant difference in the RSSI value between both modules (a difference of 10 to 12 RSSI points). We expected a better response from the DA module.

I have attached the images of our serial monitor and a portion of the code for your reference. I would be grateful if you could verify if these are the expected results from the DA module.

Thank you for your guidance and opinion on this matter.

Best regards,

Vaibhav Singh

```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

/* These are the GPIOs connected to the antenna switch on the ESP32-WROOM-DA.
 * Both GPIOs are not exposed to the module pins and cannot be used except to
 * control the antnnas switch.
 * 
 * For more details, see the datashhet at:
 * https://www.espressif.com/sites/default/files/documentation/esp32-wroom-da_datasheet_en.pdf
 */

#define GPIO_ANT1   2   // GPIO for antenna 1
#define GPIO_ANT2   25  // GPIO for antenna 2 (default)

void setup()
{
    bool err = false;
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);

    /* Attention: This is the manual prodecure for the dual antenna configuration.
     * If you choose the ESP32-WROOM-DA module from the Tools -> Board, this configuration
     * is not necessary!
     * 
     * Set WiFi dual antenna configuration by passing the GPIO and antenna mode for RX ant TX
     */ 
    err = WiFi.setDualAntennaConfig(GPIO_ANT1, GPIO_ANT2, WIFI_RX_ANT_AUTO, WIFI_TX_ANT_AUTO);

    /* For more details on how to use this feature, see our docs:
     * https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html
     */

    if(err == false) {
        Serial.println("Dual Antenna configuration failed!");
    } else {
        Serial.println("Dual Antenna configuration successfuly done!");
    }

    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
[... 93 lines cut ...]
            case WIFI_AUTH_WPA3_PSK:
                Serial.print("WPA3");
                break;
            case WIFI_AUTH_WPA2_WPA3_PSK:
                Serial.print("WPA2+WPA3");
                break;
            case WIFI_AUTH_WAPI_PSK:
                Serial.print("WAPI");
                break;
            default:
                Serial.print("unknown");
            }
            Serial.println();
            delay(10);
        }
    }
    Serial.println("");

    // Delete the scan result to free memory for code below.
    WiFi.scanDelete();

    // Wait a bit before scanning again.
    delay(5000);
}
```

nachiketkukade, 2025-01-19:
Closing due to no further response

closed 2025-01-19 by nachiketkukade