espressif/esp-idf #11288: RMII from ESP at 2MHz, RMII to ESP at 50MHz. KSZ8794 Ethernet Switch

Technosasquach, 2023-04-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, currently trying to implement a simple ethernet switch with the KSZ8794. For some reason, this has been one of the more difficult things I have done.

The Ethernet chip appears to be running fine (detecting / connecting to ethernet cables, speed and activity leds blinking on connected ports, passing ethernet data to other ports and over RMII to the ESP, and responding over SMI).
The ESP is able to establish a link to the switch following the basic ethernet example. However, while the switch is transmitting following the RMII spec, but the ESP is transmitting 20x~ish slower? See attached logic analyzer captures.
The console is indicating that the link is up, and waiting to be leased an IP from a router / DHCP server. But when I have a cabled from a router, or DHCP server on my computer no further action happens according to control.

I do not know enough about the specifics of RMII or the depths of the ESP-IDF to understand if this is a specification intention, configuration, clocking or hardware issue. I am surprised this is happening as I have not been able to find anything like this occurring on the ESP or otherwise.

IDF: v5.1-dev-4557 (idf.py --version)
Chip: ESP32-Devkit V4, no modifications so far (also tried with a WROOM)
Eth board: custom, but just a breakout to header and eth jack and magenetics on board.
Connections: flying 10cm dupont type wires. Connections are as specified in the Ethernet Examples README

What I have tried so far: (unless otherwise mentioned, I had the same outcome as above)

- Clocking for the ethernet chip.
    - 25MHz oscillator into XI, configured to generate a 50MHz clock. (Was only mode where the eth switch did not output 50MHz?)
    - 50MHz oscillator supplied to switch and ESP.
    - 50MHz clock from ESP into switch.
    - The above two modes with 25MHz supplemental, appropriately config-ed with LED straps.
- Eth LED straps
    - REFCLK-OUT enabled/disabled
    - RGMII*/RMII mode for Port 4 to ESP (LED1-0)
    - As per above: 25MHz supplemental when supplied 50MHz
    - As per above: Clock source input or output (only works when input)
    - RMII mode selection, LED3-1 = PD, LED3-0 = PU
- ESP configuration
    - With and without automatic phy addr finding (eth chip is at 0)
    - With and without reseting
    - Clock output on regular GPIO16/17 and GPIO 0. Notably it did not work on 16? No apparent clock. Worked on 17 and 0.
- ESP ETH Drivers
    -  Have used the KSZ80XX driver, only one out of supplied that works without modification. Model number reads as 8021/8031
    - Generic driver based on KSZ80XX without any device specific operations.
    - The same generic driver but with link speed, duplex, flow control all hard set to be what they should be. (I was not sure if the IDF used those to effect the RMII communication)

A note on the drivers that I observed. I am aware that since its an ethernet switch it does not have the same behavior. The reported link by the esp.emac when verbose logging is turned on is 'half duplex', '10M' and 'no flow control'. I assume this a byproduct of it looking into registers that have not been transfered from the 80XX to the 8794. I would write a phy driver using the SPI layer as that seems to be standard practice for this type of chip. Though given this RMII, I don't want to put in the work unless there is evidence it would help. (The design this is going into has one SPI split over 3 different devices already, so this would go from an isolated bring up to full integration before knowing its working). I have also been looking into the KSZ8863 work, but none of those differences looks like it should / would impact this?

Things I am unable to confirm fully:

- I don't have access to an suitable oscilloscope that can confirm the integrity of the 50MHz clock. The logic analyzer indicates a consistent clock over all samples so far, both at the ESP pin, and Ethernet pin

I'm happy for the outcome to be that I'm just missing something small somewhere, or to go look in a particular direction. Any and all help is appreciated :)

[image]
[image]
[image]
[image]
[image]
[image]
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/11348625/sdkconfig.txt)

kostaond, 2023-04-28:
@Technosasquach your question does not address any real issue related to ESP-IDF. It is more or less a question regarding your specific project. Therefore, GH issue is not correct platform to ask. You should rather ask the question on [ESP Forum](https://www.esp32.com/). Please do so in the future.

Is there any reason why you don't use KS8863? The driver is ready to be used. There is also a hardware reference design shared. I would recommend you to start there since there are a lot of problems in your question. I mean, you tried a lot of things but I am afraid you are missing some basic concepts.

- You need to make decision which CLK configuration use, please see [KSZ8794 HW Design Checklist](https://ww1.microchip.com/downloads/en/DeviceDoc/KSZ8794CNX-HW-Design-Checklist-00003877.pdf) Section 7.3.
- Using 10 cm wires to connect high speed RMII might be tricky. I've used similar approach too, it usually works but it is highly depended on robustest of PHY chip. I would recommend you to use proper design or at least solder the wires, match length and use wires as short as possible. 
- You just cannot use KSZ80XX driver as is. If you do so, you maximally might get access to one of the PHY Ports (1-3) registers in via MIIM interface, see Section 4.11 in KSZ8794 datasheet. To get full control of the chip, you need to access it via SPI.
- You need to properly configure interface between ESP32 and Port 4 (e.g. configure the speed, duplexity), see "Register 6 (0x06): Global Control 4".

To summarize, either fully switch to KSZ8863 or base your driver on [KSZ8863 driver](https://github.com/espressif/esp-eth-drivers/tree/master/ksz8863). Read the KSZ8863 README carefully to understand basic concepts. I quickly went over KSZ8794 datasheet and it seems the control is pretty much similar to KSZ8863.

Technosasquach, 2023-04-29:
@kostaond I really appreciate the time you took to reply, expecially given that this is not an ESP-IDF issue. I posted here as I felt relatively out of ideas and at least at the time it felt like and ESP/ESP-IDF configuration / programming issue. As per what you mentioned, Ill go execute on those few other areas.

But to anwser specificly for closure sakes if anyone else stumbles across this:

- The use of the KSZ8794 is needed because of the 3 PHY ports, and its the only one that has been in stock for 2ish years at this point. This is a retrofit on top of a retrofit and things are interestingTM. Otherwise, I would quite happily use the KSZ8863, and probably will do in the preceding project after this one.
- I initially wanted to use the 25MHz clock input, and output the 50MHz to the ESP. Though this did not work, still unsure about this. So I'm back at an external 50MHz oscillator to both ends, which was the basis of the testing I listed. Sorry I did not communicate that well.
- I thought the 10cm it would be pretty close as well (inside whatever signal tolerence). I initially had a board with everything on it appropiately, but as per the clock not working initiially, and general spiraling problems since, I put together a more prototypable platform. Ill respin that shortly.
- I'll freely admit the lower level eth drivers have been doing my head in. In hindsight its obvious... Ill go ahead and write a driver for it. My hesedency previously came from the SPI having multiple other peripherals on it, and I dident want to add more issues without knowing the current one/ones.
- Thanks for the advice, when I get to it I will do!

closed 2023-04-29 by Technosasquach

kostaond, 2023-05-02:
> I initially wanted to use the 25MHz clock input, and output the 50MHz to the ESP. Though this did not work, still unsure about this. So I'm back at an external 50MHz oscillator to both ends, which was the basis of the testing I listed. Sorry I did not communicate that well.
I thought the 10cm it would be pretty close as well (inside whatever signal tolerence). I initially had a board with everything on it appropiately, but as per the clock not working initiially, and general spiraling problems since, I put together a more prototypable platform. Ill respin that shortly.

It should work in any configuration. You just need to take care of the signal integrity. It's 50 MHz so the tracing, route impedance and termination needs to be in [specs.](https://resources.pcb.cadence.com/blog/2019-mii-and-rmii-routing-guidelines-for-ethernet). Also keep in mind that GPIO0 is boot strap pin for ESP32. Therefore you need to make sure there is no CLK signal during bootup. There are multiple techniques how to achieve it (keep source of CLK disabled, delay CLK start by RC circuit or control it via GPIO; if KSZ is source of the CLK, keep it in reset, etc.).
If you decided to use ESP32 as CLK source, you can use GPIO16/17. However, just keep in mind that these pins are not available when you use WROVER since they are occupied by PSRAM. You have to use WROOM in that case. 

> I'll freely admit the lower level eth drivers have been doing my head in. In hindsight its obvious... Ill go ahead and write a driver for it. My hesedency previously came from the SPI having multiple other peripherals on it, and I dident want to add more issues without knowing the current one/ones.
Thanks for the advice, when I get to it I will do!

That should not be much an issue since the control of the KSZ chip over SPI does not consume much traffic. The driver only periodically check the status after the startup configuration.