espressif/esp-idf #11289: about esp_tls_conn_write function

Inverter-Jo, 2023-04-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Naturally, the value returned by the above function depends on the length of the data.
However, in my case, if the data length exceeds 128, the return value becomes 0, and the data transfer fails.

What's the problem?

laukik-hase, 2023-05-08:
Hello, @Inverter-Jo!

Thank you for sending the issue report. I was investigating this issue but unfortunately could not reproduce it with the `https_request` example on the ESP-IDF https://github.com/espressif/esp-idf/commit/56123c52aaa08f1b53350c7af30c91320b352ef4 commit.

I was able to send more than 128 bytes with `esp_tls_conn_write`. Please find the attached debug logs below from my test setup.

<details>
  <summary><b>https_request: Debug logs</b></summary>

```log
I (9211) example: Start https_request example
I (9221) example: https_request using crt bundle
D (9221) esp-tls: host:www.howsmyssl.com: strlen 17
I (9231) main_task: Returned from app_main()
D (9251) esp-tls: [sock=54] Resolved IPv4 address: 34.71.45.200
D (9251) esp-tls: [sock=54] Connecting to server. HOST: www.howsmyssl.com/a/check, Port: 443
D (9551) esp-tls-mbedtls: Use certificate bundle
D (9551) esp-tls: handshake in progress...
D (9881) intr_alloc: Connected src 76 to int 4 (cpu 1)
D (9971) esp-x509-crt-bundle: 138 certificates in bundle
I (9991) esp-x509-crt-bundle: Certificate validated
I (11081) example: Connection established...
I (11091) example: 198 bytes written
D (11091) example: Request - 
GET https://www.howsmyssl.com/a/check HTTP/1.1
Host: www.howsmyssl.com
User-Agent: esp-idf/5.2-dev (esp32s3)
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8

I (11101) example: Reading HTTP response...
D (11341) example: 511 bytes read
HTTP/1.1 200 OK
Content-Length: 2545
Access-Control-Allow-Origin: *
Connection: close
Content-Type: application/json
Date: Mon, 08 May 2023 06:58:14 GMT
Strict-Transport-Security: max-age=631138519; includeSubdomains; preload

{"given_cipher_suites":["TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_ECDSA_WITH_AES_256_CCM","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_RSA
D (11371) example: 511 bytes read
_WITH_AES_256_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_256_CCM_8","TLS_ECDHE_ECDSA_WITH_ARIA_256_GCM_SHA384","TLS_ECDHE_RSA_WITH_ARIA_256_GCM_SHA384","TLS_ECDHE_ECDSA_WITH_ARIA_256_CBC_SHA384","TLS_ECDHE_RSA_WITH_ARIA_256_CBC_SHA384","TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CCM","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA","TLS_ECD
D (11421) example: 157 bytes read
HE_ECDSA_WITH_AES_128_CCM_8","TLS_ECDHE_ECDSA_WITH_ARIA_128_GCM_SHA256","TLS_ECDHE_RSA_WITH_ARIA_128_GCM_SHA256","TLS_ECDHE_ECDSA_WITH_ARIA_128_CBC_SHA256","
D (11441) example: 511 bytes read
TLS_ECDHE_RSA_WITH_ARIA_128_CBC_SHA256","TLS_RSA_WITH_AES_256_GCM_SHA384","TLS_RSA_WITH_AES_256_CCM","TLS_RSA_WITH_AES_256_CBC_SHA256","TLS_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA","TLS_RSA_WITH_AES_256_CCM_8","TLS_ECDH_ECDSA_WITH_ARIA_256_GCM_SHA384","TLS_ECDH_RSA_WITH_ARIA_256_GCM_SHA384",
D (11491) example: 511 bytes read
"TLS_RSA_WITH_ARIA_256_GCM_SHA384","TLS_ECDH_ECDSA_WITH_ARIA_256_CBC_SHA384","TLS_ECDH_RSA_WITH_ARIA_256_CBC_SHA384","TLS_RSA_WITH_ARIA_256_CBC_SHA384","TLS_RSA_WITH_AES_128_GCM_SHA256","TLS_RSA_WITH_AES_128_CCM","TLS_RSA_WITH_AES_128_CBC_SHA256","TLS_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA"
D (11541) example: 511 bytes read
,"TLS_RSA_WITH_AES_128_CCM_8","TLS_ECDH_ECDSA_WITH_ARIA_128_GCM_SHA256","TLS_ECDH_RSA_WITH_ARIA_128_GCM_SHA256","TLS_RSA_WITH_ARIA_128_GCM_SHA256","TLS_ECDH_ECDSA_WITH_ARIA_128_CBC_SHA256","TLS_ECDH_RSA_WITH_ARIA_128_CBC_SHA256","TLS_RSA_WITH_ARIA_128_CBC_SHA256","TLS_EMPTY_RENEGOTIATION_INFO_SCSV"],"ephemeral_keys_supported":true,"session_ticket_supported":true,"tls_compression_supported":false,"unknown_cipher_suite_supported":false,"beast_vuln":false,"able_to_detect_n_minus_one_splitting":false,"insecure
D (11581) example: 68 bytes read
_cipher_suites":{},"tls_version":"TLS 1.2","rating":"Probably Okay"}
I (11591) example: connection closed

```
</details>

Could you please attach the following -

- Target chip or module (ESP32, ESP32-S2, ESP32-S3, etc.)
- ESP-IDF version (run `git describe --tags`) / commit ID
- A minimal example that reproduces the issue
- The `sdkconfig` file for your example
- Debug logs (with `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y`)

laukik-hase, 2023-05-24:
Hello, @Inverter-Jo!

Closing this issue due to lack of feedback, feel free to reopen with more updates. Thanks.

closed 2023-05-24 by laukik-hase