espressif/esp-idf #11291: Only privite declaration adc_oneshot_read_isr

kossnikita, 2023-04-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

There's no declaration of `adc_oneshot_read_isr` in [esp_adc/adc_oneshot.h](https://github.com/espressif/esp-idf/blob/56123c52aaa08f1b53350c7af30c91320b352ef4/components/esp_adc/include/esp_adc/adc_oneshot.h). There's only in private [adc_private.h](https://github.com/espressif/esp-idf/blob/56123c52aaa08f1b53350c7af30c91320b352ef4/components/esp_adc/include/esp_private/adc_private.h)

This prevents the function from being used without additional declaration.

### What is the actual behavior?

The declaration must be in a public header.

### Steps to reproduce.

1. IDF 5.0.1
2. Kconfig: ADC_ONESHOT_CTRL_FUNC_IN_IRAM = y
3.  #include <esp_adc/adc_oneshot.h>
4.  Call adc_oneshot_read_isr()

### Build or installation Logs.

```plain
error: implicit declaration of function 'adc_oneshot_read_isr'; did you mean 'adc_oneshot_read'? [-Werror=implicit-function-declaration]
   32 |   adc_oneshot_read_isr(handle, PH_ADC_CHANNEL, &raw);
      |   ^~~~~~~~~~~~~~~~~~~~
      |   adc_oneshot_read
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
```
```

peturdainn, 2023-10-30:
To be fair, the function is also not documented, but works.

I also think the _isr version should be using portENTER_CRITICAL_ISR  so maybe this code isn't used or debugged properly

peturdainn, 2023-11-29:
I have to come back to this, I have noticed the adc_oneshot_read_isr() function occasionally returns incorrect values, so maybe this function is not in the IDF API because it has issues?