espressif/esp-idf #11292: esp_crc8_le does not match the reference output?

wuyuanyi135, 2023-04-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1-N16R2

### Power Supply used.

USB

### What is the expected behavior?

By referencing to the [link](https://crccalc.com/?crc=Hi&method=crc8&datatype=ascii&outtype=0), I expect the `esp_crc8_le` output some matching result when processing a string "Hi". It should tell `CRC=0xEB`
```cpp
uint8_t result = esp_crc8_le(0, (uint8_t*)"Hi", 2);
ESP_LOGI(TAG, "CRC=0x%02x", result);
```

### What is the actual behavior?

```cpp

  ESP_LOGI(TAG, "CRCLE0=0x%02x",  esp_crc8_le(0, (uint8_t*)"Hi", 2));
  ESP_LOGI(TAG, "CRCLE1=0x%02x", ~esp_crc8_le(0, (uint8_t*)"Hi", 2));
  ESP_LOGI(TAG, "CRCLE2=0x%02x",  esp_crc8_le(0xff, (uint8_t*)"Hi", 2));
  ESP_LOGI(TAG, "CRCLE3=0x%02x", ~esp_crc8_le(0xff, (uint8_t*)"Hi", 2));

  ESP_LOGI(TAG, "CRCBE0=0x%02x",  esp_crc8_be(0, (uint8_t*)"Hi", 2));
  ESP_LOGI(TAG, "CRCBE1=0x%02x", ~esp_crc8_be(0, (uint8_t*)"Hi", 2));
  ESP_LOGI(TAG, "CRCBE2=0x%02x",  esp_crc8_be(0xff, (uint8_t*)"Hi", 2));
  ESP_LOGI(TAG, "CRCBE3=0x%02x", ~esp_crc8_be(0xff, (uint8_t*)"Hi", 2));
```

prints 
```
I (825) MAIN: CRCLE0=0x7d
I (825) MAIN: CRCLE1=0xffffff82
I (825) MAIN: CRCLE2=0x96
I (825) MAIN: CRCLE3=0xffffff69
I (835) MAIN: CRCBE0=0xc3
I (835) MAIN: CRCBE1=0xffffff3c
I (835) MAIN: CRCBE2=0x14
I (835) MAIN: CRCBE3=0xffffffeb
```

None of them matches the output `0xEB`. Am I using the rom crc8 incorrectly?

### Steps to reproduce.

see above

igrr, 2023-04-28:
This might be helpful: https://github.com/espressif/esp-idf/blob/56123c52aaa08f1b53350c7af30c91320b352ef4/components/esp_rom/include/esp_rom_crc.h#L23-L45

So you can reproduce 0x7d using http://www.sunshine2k.de/coding/javascript/crc/crc_js.html, set "input reflected", "result reflected", initial value 0xFF (which corresponds to crc=0 argument after inversion) and final XOR value 0xff (which corresponds to the inversion at the end), polynomial 0x07.

esp-wzh, 2023-04-28:
You can also invert the result of `esp_crc8_le`  to be consistent with `CRC-8/ROHC`

wuyuanyi135, 2023-04-28:
@igrr @esp-wzh Thank you all for the prompt and helpful response!

closed 2023-04-28 by wuyuanyi135