espressif/esp-idf #11294: esp32s3  ble作为中心设备，连接多个外围设备的时候，有时候会出现assert failed错误

zhu296, 2023-04-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

EPS32-S3-DevkitC-1

### Power Supply used.

USB

### What is the expected behavior?

BLE连接多个设备能正常工作，最少三个

### What is the actual behavior?

esp32s3  ble作为中心设备，连接多个外围设备的时候，有时候会出现assert failed

### Steps to reproduce.

esp32s3  ble作为中心设备，连接多个外围设备的时候，有时候会出现assert failed: tlsf_free tlsf.c:1049 (!block_is_free(block) && "block already marked as free")失败的错误(连接的设备越多，出现的概率越大)

1，基于example   gattc_multi_connect 工程，修改了remote_device_name和 外围设备的UUID
2，4个外围设备ble连接成功后，会每隔10S往中心设备发送300字节的数据
3，反复重启esp32s3 ，概率性出现assert failed: tlsf_free tlsf.c，
4，通过coredump，我定位到是在函数gatt_cl_send_next_cmd_inq中的osi_free出现的assert

### Debug Logs.

```plain
ssert failed: tlsf_free tlsf.c:1049 (!block_is_free(block) && "block already marked as free")

Backtrace: 0x40379fca:0x3fcab330 0x4038386d:0x3fcab350 0x40389b8d:0x3fcab370 0x4038908f:0x3fcab490 0x403889e4:0x3fcab4b0 0x4037a2de:0x3fcab4d0 0x40389bbd:0x3fcab4f0 0x42024ae0:0x3fcab510 0x42024dbe:0x3fcab540 0x4202666a:0x3fcab580 0x420268f6:0x3fcab5a0 0x42030649:0x3fcab5c0 0x420211ca:0x3fcab600 0x4203867a:0x3fcab620 0x40386f1d:0x3fcab650

ELF file SHA256: 46778c7b4c1eacf0

CPU halted.

下面是coredump数据：

(gdb)
(gdb) backtrace
#0  0x40379fd5 in panic_abort (
    details=0x3fcac87b "assert failed: tlsf_free tlsf.c:1049 (!block_is_free(block) && \"block already marked as free\")") at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/esp_system/panic.c:412
#1  0x40383878 in esp_system_abort (
    details=0x3fcac87b "assert failed: tlsf_free tlsf.c:1049 (!block_is_free(block) && \"block already marked as free\")") at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/esp_system/esp_system.c:135
#2  0x40389c70 in __assert_func (file=0x3fc9f683 "tlsf.c", line=<optimized out>, func=<optimized out>,
    expr=0x3fc9fa20 "!block_is_free(block) && \"block already marked as free\"")
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/newlib/assert.c:78
#3  0x40389172 in tlsf_free (tlsf=0x3fca566c, ptr=0x3fcb2ef0)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/heap/tlsf/tlsf.c:1049
#4  0x40388ac7 in multi_heap_free_impl (heap=0x3fca5658, p=0x3fcb2ef0)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/heap/multi_heap.c:241
#5  0x4037a2e9 in heap_caps_free (ptr=0x3fcb2ef0)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/heap/heap_caps.c:353
#6  0x40389ca0 in free (ptr=0x3fcb2ef0) at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/newlib/heap.c:39
#7  0x420264cd in gatt_cl_send_next_cmd_inq (p_tcb=0x3fcb0838)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/bt/host/bluedroid/stack/gatt/gatt_cl.c:1043
#8  0x420267a5 in gatt_client_handle_server_rsp (p_tcb=0x3fcb0838, op_code=<optimized out>, len=<optimized out>,
    p_data=0x3fcb3b06 <error: Cannot access memory at address 0x3fcb3b06>)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/bt/host/bluedroid/stack/gatt/gatt_cl.c:1167
#9  0x42028051 in gatt_data_process (p_tcb=0x3fcb0838, p_buf=0x3fcb3af4)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/bt/host/bluedroid/stack/gatt/gatt_main.c:1004
#10 0x420282dd in gatt_le_data_ind (chan=<optimized out>,
    bd_addr=0x3fcb0c1c <error: Cannot access memory at address 0x3fcb0c1c>, p_buf=0x3fcb3af4)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/bt/host/bluedroid/stack/gatt/gatt_main.c:587
#11 0x42032030 in l2c_rcv_acl_data (p_msg=0x3fcb3af4)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/bt/host/bluedroid/stack/l2cap/l2c_main.c:275
#12 0x42022bbd in btu_hci_msg_process (param=0x3fcb3af4)
--Type <RET> for more, q to quit, c to continue without paging--
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/bt/host/bluedroid/stack/btu/btu_task.c:144
#13 0x4203a061 in osi_thread_run (arg=<optimized out>)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/bt/common/osi/thread.c:165
#14 0x40387000 in vPortTaskWrapper (pxCode=0x4203a02c <osi_thread_run>, pvParameters=0x3fcf7c90)
    at D:/esp/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
(gdb)
(gdb)
(gdb)
```

zhu296, 2023-04-28:
现象好像跟 这个issuue [TW#26691] bluetooth heap corruption #2541 类似

esp-cjh, 2023-05-22:
@zhu296 是4个设备都连接时直接重启 ESP32S3 吗？重启后再次连接？

zhu296, 2023-05-23:
@zhu296 是4个设备都连接时直接重启 ESP32S3 吗？重启后再次连接？外围设备是哪些？
1，是在4个设备都连接时直接重启ESP32S3
2，重启后再次连接时概率性出现上面的错误
3，外围设备是4个一样的ble设备，外围设备ble连接成功后，会每隔10S往中心设备发送300字节的数据

esp-cjh, 2023-05-23:
@zhu296 
可以发一下完整的 log 文件吗？有抓包文件的话也发一下。

zhu296, 2023-05-23:
@esp-cjh 

[reboot.txt](https://github.com/espressif/esp-idf/files/11540039/reboot.txt)

esp-cjh, 2023-05-24:
@zhu296 你修改了 example 吗？能提供一下代码吗？我这边需要加 log 复现才能找到问题原因。

zhu296, 2023-05-25:
我已经定位到具体出问题的地方了，我修改了gattc_multi_connect示例里面的函数gattc_profile_a_event_handler profile处理函数。

在gattc_profile_a_event_handler函数中，加入了以下case。
		  
            case ESP_GATTC_CONNECT_EVT:
			{
               ESP_LOGI(DEBUG_TAG, "ESP_GATTC_CONNECT_EVT conn_id %d, if %d", p_data->connect.conn_id, gattc_if);
               gl_profile_tab[PROFILE_D_APP_ID].conn_id = p_data->connect.conn_id;
               memcpy(gl_profile_tab[PROFILE_D_APP_ID].remote_bda, p_data->connect.remote_bda, sizeof(esp_bd_addr_t));

               break;
            }
            加入这个case，并且加上memcpy这一行后，就出现两个问题：
			    1，基本每次发送数据都会出现tlfs  free assert
				2，connid 混乱
			
			怀疑是connid赋值冲突导致，应该connid在case ESP_GATTC_OPEN_EVT 已经赋值了。

esp-cjh, 2023-05-25:
@zhu296 
应该是你的代码写得有问题，你应该为 `PROFILE_D_APP_ID` 注册 `gattc_profile_d_event_handler` 函数，然后在这个函数里执行赋值或者发送数据等操作。
[image]

检查 `PROFILE_NUM` 的值是否修改了？是否存在数组越界的问题。

zhu296, 2023-05-25:
PROFILE_NUM  确定修改了，问题应该是gattc_profile_a_event_handler profile函数里面，ble协议事件处理有两个case ESP_GATTC_CONNECT_EVT和ESP_GATTC_OPEN_EVT ，只需要在其中一个case里面给connid赋值就额可以了。
导致问题的原因 是我在这个两个case里面都给connid赋值了

closed 2023-05-29 by zhu296