espressif/esp-idf #11299: Build fails on Apple M2 silicon

kegtron, 2023-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build of hello_world using 'idf.py build' should complete without error

### What is the actual behavior?

Build fails with  "-- Looking for sys/types.h - not found"

### Steps to reproduce.

1) Obtain Macbook Air with Ventura 13.1
2) Install Python 3.11.3
3) Install esp-idf v5.0.1 per install guide
4) Add "export IDF_PATH=~/esp/esp-idf-v5.0.1" to .zprofile
5) source .zprofile
6) ./install.sh esp32
7) ./export.sh
8) Copy examples/get-started/hello_world to new dir
9) idf.py build

### Build or installation Logs.

```plain
xxx@MacBook-Air hello_world % idf.py build
Executing action: all (aliases: build)
Running cmake in directory /Users/keck/esp/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /Users/xxx/esp/hello_world"...
-- Found Git: /usr/bin/git (found version "2.39.2 (Apple Git-143)") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/xxx/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/xxx/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/xxx/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /Users/xxx/esp/hello_world/sdkconfig
-- Looking for sys/types.h
-- Looking for sys/types.h - not found
-- Looking for stdint.h
-- Looking for stdint.h - not found
-- Looking for stddef.h
-- Looking for stddef.h - not found
-- Check size of time_t
-- Check size of time_t - failed
CMake Error at /Users/xxx/esp/esp-idf-v5.0.1/CMakeLists.txt:219 (message):
  Failed to determine sizeof(time_t)
```

### More Information.

- Same methods work fine on Intel-based mac with same OS version
- Using Python v3.10.0 has same results

igrr, 2023-05-02:
@kegtron Could you please attach `/Users/keck/esp/hello_world/build/CMakeFiles/CMakeOutput.log` file? It should contain more information which could help troubleshoot this issue.

Based on this part of the log,
``` 
-- Looking for sys/types.h
-- Looking for sys/types.h - not found
-- Looking for stdint.h
-- Looking for stdint.h - not found
-- Looking for stddef.h
-- Looking for stddef.h - not found
-- Check size of time_t
-- Check size of time_t - failed
```
it seems that the toolchain was not fully or correctly extracted. Or it could be because the compiler fails to run. `CMakeOutput.log` might contain more info about this.

kegtron, 2023-05-02:
@igrr Thank you for the consult

To your questions:
- CMakeOutput.log file is not present in hello_world/build/CMakeFiles directory
- xtensa-esp32-elf-gcc -v runs successfully
- arch = arm64
- python -c 'import platform; print(platform.machine())' = arm64
- All commands issued after running source export.sh

igrr, 2023-05-02:
Could you also check if the compiler finds the header files which are reported as missing? For example,
```
xtensa-esp32-elf-gcc -c -x c -o /dev/null - <<< "#include <sys/types.h>"
```
should not print any errors.

By the way, what is the output of `cmake --version`?

kegtron, 2023-05-02:
Compiler throws a ton of errors, but most are "Unsupported architecture"  like this:

```
hello_world % xtensa-esp32-elf-gcc -c -x c -o /dev/null - <<< "#include <sys/types.h>"
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/types.h:75,
                 from <stdin>:1:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:913:2: error: #error Unsupported architecture
  913 | #error Unsupported architecture
      |  ^~~~~
```
cmake version 3.26.2

igrr, 2023-05-03:
Okay, seems like we are getting closer to finding the cause of this issue! For some reason the Xtensa compiler is finding "sys/types.h" header among your system header files, instead of the header files bundled along with Xtensa toolchain. This may indicate that something is influencing the list compiler search paths.

Could you try running the same command with an additional `-v` flag? This may produce a lot of output, please capture all the output and attach it here. You can save the log to a file like this:
```
xtensa-esp32-elf-gcc -v -c -x c -o /dev/null - <<< "#include <sys/types.h>" 2> /tmp/gcc_log.txt
```

Could you also check if you have any of the following environment variables set? (You can run `env` to see the list):
* `CPATH`
* `C_INCLUDE_PATH`

If you see that such an environment variable is set, please try unsetting it and then run `xtensa-esp32-elf-gcc -v -c -x c -o /dev/null - <<< "#include <sys/types.h>"` again to confirm if that helps.

kegtron, 2023-05-03:
OK, log attached

CPATH and C_INCLUDE_PATH are not defining in my environment

[gcc_log.txt](https://github.com/espressif/esp-idf/files/11384393/gcc_log.txt)

igrr, 2023-05-03:
So the difference seems to be that Xcode headers path gets added ahead of the toolchain paths. In your log:
```
#include "..." search starts here:
#include <...> search starts here:
 /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include
 /Users/keck/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/include
 /Users/keck/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/include-fixed
 /Users/keck/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/sys-include
 /Users/keck/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/include
End of search list.
```

and with the same toolchain over here I get:
```
#include "..." search starts here:
#include <...> search starts here:
 /Users/ivan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/include
 /Users/ivan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/include-fixed
 /Users/ivan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/sys-include
 /Users/ivan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/include
End of search list.
```

The only way I found to get the output similar to yours was by setting `CPATH`. For example, after `export CPATH=/tmp/test1` I get
```
#include "..." search starts here:
#include <...> search starts here:
 /tmp/test1
 /Users/ivan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/include
 /Users/ivan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/include-fixed
 /Users/ivan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/sys-include
 /Users/ivan/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/include
```

I haven't found any other way to influence the compiler search path, other than via environment variables. (Since we are running gcc directly, there are obviously no incorrect command line arguments involved.)

I also found [this Stackoverflow answer](https://apple.stackexchange.com/a/372600) which suggests adding
```
export CPATH=`xcrun --show-sdk-path`/usr/include
```
to `.zshrc`. After doing this, I get the exact same error as you are getting.

Could you please confirm once more there isn't a CPATH environment variable involved, somehow? If that's not the case, I'm afraid I'm out of ideas.

kegtron, 2023-05-03:
My apologies - CPATH was, in fact, defined in my environment:

CPATH=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include

I've now commented it out and xtensa-esp32-elf-gcc -v -c -x c -o /dev/null - <<< "#include <sys/types.h>" runs without error. Progress!

Now when I run idf.py build on hello_world it is successful!

CPATH was the guilty part here - removing it and all is well. Thanks for all the help!!

closed 2023-05-03 by kegtron