espressif/esp-idf #11300: build fails with nimBLE and verbose logging

listout, 2023-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4883-g56123c52aa

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build completes successfully.

### What is the actual behavior?

Build fails with `/home/listout/esp/esp-idf-v5.0/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:1809:14: error: 'service' may be used uninitialized [-Werror=maybe-uninitialized]`

### Steps to reproduce.

1. Enable nimble stack (my sdkconfig can be found here: http://ix.io/4uwx)
2. Enable verbose logging
3. Build the project with IDF from the master branch

### Build or installation Logs.

```plain
The full build log can be found here: http://ix.io/4uwv
```

### More Information.

The fix (applies in the nimble submodule):

```diff
diff --git a/nimble/host/src/ble_gattc.c b/nimble/host/src/ble_gattc.c
index 4763f14e..63d967b4 100644
--- a/nimble/host/src/ble_gattc.c
+++ b/nimble/host/src/ble_gattc.c
@@ -1956,7 +1956,7 @@ static int
 ble_gattc_find_inc_svcs_rx_adata(struct ble_gattc_proc *proc,
                                  struct ble_att_read_type_adata *adata)
 {
-    struct ble_gatt_svc service;
+    struct ble_gatt_svc service = {0};
     int call_cb;
     int cbrc;
     int rc;
```

AxelLin, 2023-05-05:
Hit the same build error if CONFIG_COMPILER_OPTIMIZATION_PERF=y.

rahult-github, 2023-05-05:
We already have raised a fix internally. Once merged, it should be able to be seen on both master and release/v5.1 branch

listout, 2023-05-05:
@rahult-github Thanks, can I close this one then?

listout, 2023-05-05:
> CONFIG_COMPILER_OPTIMIZATION_PERF=y

I too have that option in my sdkconfig @AxelLin

rahult-github, 2023-05-05:
> @rahult-github Thanks, can I close this one then?

Hi @listout  , i suggest to wait till the commit gets pushed and merged. This issue will get updated automatically.

mattytrentini, 2023-05-30:
> We already have raised a fix internally. Once merged, it should be able to be seen on both master and release/v5.1 branch

@rahult-github is there any way to track the status of this fix? Building MicroPython for the ESP32-C6 is blocked by this issue and I'd like to test the fix when it's available. Thanks!

rahult-github, 2023-05-30:
> @rahult-github is there any way to track the status of this fix? Building MicroPython for the ESP32-C6 is blocked by this issue and I'd like to test the fix when it's available. Thanks!

Changes are available on master branch with commit 732c921c650c37c91c164fb0deaa5cecc5ce54ca . 

You can pull the latest repo OR cherry-pick the commit.

closed 2023-05-30 by rahult-github