espressif/esp-idf #11301: Assertion fails when ethernet disconnects and a tcp socket is listening

pylorak, 2023-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

No assert failure/panic/crash happens, and blocking socket call returns with an error code.

### What is the actual behavior?

ESP-IDF internal assertion fails and the software therefor crashes.

### Steps to reproduce.

Using ESP32-S3 connected to ethernet over an W5500. Ethernet communication works as expected, but if a TCP listener socket is open and the ethernet cable is suddenly unplugged, ESP-IDF panics.

### Debug Logs.

```plain
assert failed: tcp_abandon /IDF/components/lwip/lwip/src/core/tcp.c:582 (don't call tcp_abort/tcp_abandon for listen-pcbs)

Backtrace: 0x403766b7:0x3fca6390 0x4037f3ad:0x3fca63b0 0x4038682d:0x3fca63d0 0x4202c782:0x3fca64f0 0x4202c845:0x3fca6530 0x420362ac:0x3fca6550 0x420369a9:0x3fca6580 0x4202a8dd:0x3fca65a0 0x4202a980:0x3fca65c0 0x40383429:0x3fca65f0
0x403766b7: panic_abort at D:/esp-idf-5.0.1/esp-idf/components/esp_system/panic.c:423

0x4037f3ad: esp_system_abort at D:/esp-idf-5.0.1/esp-idf/components/esp_system/esp_system.c:153

0x4038682d: __assert_func at D:/esp-idf-5.0.1/esp-idf/components/newlib/assert.c:78

0x4202c782: tcp_abandon at D:/esp-idf-5.0.1/esp-idf/components/lwip/lwip/src/core/tcp.c:582 (discriminator 1)

0x4202c845: tcp_abort at D:/esp-idf-5.0.1/esp-idf/components/lwip/lwip/src/core/tcp.c:646

0x420362ac: lwip_netconn_do_close_internal at D:/esp-idf-5.0.1/esp-idf/components/lwip/lwip/src/api/api_msg.c:1012

0x420369a9: lwip_netconn_do_delconn at D:/esp-idf-5.0.1/esp-idf/components/lwip/lwip/src/api/api_msg.c:1225 (discriminator 2)

0x4202a8dd: tcpip_thread_handle_msg at D:/esp-idf-5.0.1/esp-idf/components/lwip/lwip/src/api/tcpip.c:162

0x4202a980: tcpip_thread at D:/esp-idf-5.0.1/esp-idf/components/lwip/lwip/src/api/tcpip.c:148

0x40383429: vPortTaskWrapper at D:/esp-idf-5.0.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

kostaond, 2023-09-21:
@pylorak thank you for reporting potential issue. Could you please provide more information of how to reproduce the issue? A code sniped would be much appreciated. I tried to run [tcp_server](https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/tcp_server) example and tried to unplug the cable when server was listening. However, I was not able to reproduce the issue.

kostaond, 2023-11-06:
@pylorak, could you please provide more information of how to reproduce the issue?

kostaond, 2024-01-18:
Closing due to lack of feedback.

closed 2024-01-18 by kostaond