espressif/esp-idf #11304: ESP32-C6 HA_on_off_light example not connecting to zigbee network.

spark404, 2023-05-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-350-g56123c52aa

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1-N

### Power Supply used.

USB

### What is the expected behavior?

The devkit installed with the HA_on_off_switch firmware should be able to control a second devkit running the HA_on_off_light firmware.

### What is the actual behavior?

The devkit running the HA_on_off_light repeatedly mentions the error `Network steering was not successful (status: -1)`.

### Steps to reproduce.

1. Erase flash on DevKit A and B (Both C6 devkits)
2. Clear NVRAM on DevKit A and B
3. Install HA_on_off_light on DevKit A
4. Install HA_on_off_switch on DevKit B
5. Monitor serial output on DevKit A

### More Information.

This issue seems similar to https://github.com/espressif/esp-idf/issues/10662

I verified the behaviour with release 5.1, this did work immediately.

closed 2023-05-06 by espressif-bot

chshu, 2023-05-06:
@spark404 The issue was caused by some change in 802.15.4 driver, and it has been fixed in the latest Zigbee lib. Please rebuild the example, it could download the latest Zigbee SDK component.

spark404, 2023-05-08:
@chshu Thanks for fixing this! I can confirm that it works now.