espressif/esp-idf #11305: Watchdog immediately after going to sleep

diplfranzhoepfinger, 2023-05-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Release v5.0

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

esp_deep_sleep_start();

--> Watchdog

### What is the actual behavior?

-

### Steps to reproduce.

-

### Debug Logs.

```plain
I (15:01:21.055) EcuSleepWakeup: Enable DEEP SLEEP wakeup via pin with pin mask 0x82.

I (15:01:21.060) EcuSleepWakeup: Enable DEEP SLEEP wakeup via timer for 20000000 microsec.

I (15:01:21.069) EcuSleepWakeup: Device goes into DEEP SLEEP

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

 

Core  0 register dump:

PC      : 0x40048c2b  PS      : 0x00060734  A0      : 0x80048c8e  A1      : 0x3fcabe60

A2      : 0x03ff0000  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3fceaa3c

A6      : 0x0000000e  A7      : 0xff000000  A8      : 0x6001001c  A9      : 0xa011c000

A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3fceaa90  A13     : 0x00000000

A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000c  EXCCAUSE: 0x00000005

EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc

 

 

Backtrace: 0x40048c28:0x3fcabe60 |<-CORRUPTED

 

 

 

 

ELF file SHA256: 00b53cab2ef92625

 

I (12962) esp_core_dump_flash: Save core dump to flash...

E (12969) esp_core_dump_flash: Not enough space to save core dump!

E (12975) esp_core_dump_elf: Failed to prepare core dump storage (257)!

E (12982) esp_core_dump_common: Core dump write binary failed with error=257

I (12990) esp_core_dump_flash: Core dump has been saved to flash.

Rebooting...

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

 

Core  0 register dump:

PC      : 0x4038681e  PS      : 0x00060434  A0      : 0x803771a4  A1      : 0x3fcabbe0

0x4038681e: xt_utils_get_core_id at D:/Espressif/frameworks/esp-idf-v5.0/components/xtensa/include/xt_utils.h:36

(inlined by) esp_cpu_get_core_id at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include/esp_cpu.h:126

(inlined by) xPortGetCoreID at D:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:629

(inlined by) xPortInIsrContext at D:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:280

 

A2      : 0x00000000  A3      : 0x0000000a  A4      : 0x00000000  A5      : 0x00000000

A6      : 0x00000000  A7      : 0x3fcabb23  A8      : 0x80048c6c  A9      : 0x3fcabb40

A10     : 0x00060425  A11     : 0x0000000a  A12     : 0x00000000  A13     : 0x00000000

A14     : 0x00000000  A15     : 0x3fcabb05  SAR     : 0x00000004  EXCCAUSE: 0x00000005

EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff

 

 

Backtrace: 0x4038681b:0x3fcabbe0 0x403771a1:0x3fcabc00 0x4038060a:0x3fcabc20 0x4038067c:0x3fcabc40 0x40375e1b:0x3fcabc60 0x42004bc0:0x3fcabc80 0x42005536:0x3fcabcb0 0x42004b11:0x3fcabd30 0x40375ec9:0x3fcabd80 0x40375cec:0x3fcabda0 0x00040022:0x3fcabe60 |<-CORRUPTED

0x4038681b: xPortSetInterruptMaskFromISR at D:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:558

(inlined by) xPortInIsrContext at D:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:279

 

0x403771a1: xPortEnterCriticalTimeoutSafe at D:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:584

(inlined by) vPortEnterCriticalSafe at D:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:594

(inlined by) regi2c_ctrl_write_reg_mask at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/regi2c_ctrl.c:45

 

0x4038060a: rtc_clk_cpu_freq_to_xtal at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/port/esp32s3/rtc_clk.c:345 (discriminator 4)

 

0x4038067c: rtc_clk_cpu_freq_set_xtal at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/port/esp32s3/rtc_clk.c:330

 

0x40375e1b: esp_restart_noos_dig at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/esp_system.c:39

 

0x42004bc0: panic_restart at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/panic_handler.c:250

 

0x42005536: esp_panic_handler at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/panic.c:388

 

0x42004b11: panic_handler at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/panic_handler.c:192

 

0x40375ec9: panicHandler at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/panic_handler.c:217

 

0x40375cec: xt_highint4 at D:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/soc/esp32s3/highint_hdl.S:108
```

diplfranzhoepfinger, 2023-05-01:
Hello Franz,

I am writing you because we have a problem with interrupt watchdog and hope that you can help me find a solution 

[image]

When I try to activate deep sleep by calling

```
esp_deep_sleep_start();
```

Below is the log.

Immediately after this I get an ECU reset caused by the interrupt watchdog.

I expected that esp\_deep\_sleep\_start() will stop all active watchdogs, but it seems that it doesn’t do it.

I also can’t find an interface for stopping the interrupt watchdog manually. 

[image]

Do you know if it is possible to stop the interrupt watchdog manually?

Is there any configuration for automatic WDG disabling when entering deep sleep?

diplfranzhoepfinger, 2023-05-01:
[https://github.com/espressif/esp-idf/tree/master/examples/system/deep_sleep](https://github.com/espressif/esp-idf/tree/master/examples/system/deep_sleep)

[https://github.com/espressif/esp-idf/tree/master/examples/system/deep_sleep_wake_stub](https://github.com/espressif/esp-idf/tree/master/examples/system/deep_sleep_wake_stub)

[https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep](https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep)

I tested all 3, about 8 Months ago, and they all worked seamlessly.

diplfranzhoepfinger, 2023-05-01:
Thank you for your help, Franz.

These examples are working because watchdog is disabled for all of them.

When I disable interrupt watchdog from menu config and disable RTC watchdog before entering deep sleep

everything works fine.

Btw. I think that RTC watchdog shall be enough so we can disable interrupt WDG, but let’s understand if this is an issue in ESP-IDF.

sudeep-mohanty, 2023-05-03:
Hi @diplfranzhoepfinger,
Can you share more details about the problem you are facing? Specifically -
- Which SoC are you using?
- Can you share a sample code to reproduce the problem at our end?
- Can you share the `sdkconfig` file?

closed 2023-05-04 by sudeep-mohanty