espressif/esp-idf #11307: Flash issue: after Flash Encryption development to switch in release mode

AvinashSinghPES, 2023-05-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

some question regarding flash encryption:
1. first we disable the encryption then after Flash Encryption in development to switch in release mode  but  EFUSE_CRYPT_CNT  value not change. why?
2. is this possible to reprogram after flash encryption in development to release mode ?

mahavirj, 2023-05-02:
- Once development mode is configured for the flash enc then it can be disabled by updating the `FLASH_CRYPT_CNT` field as highlighted in the docs [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#disabling-flash-encryption) (please see command from step 4)
- Odd number of set bits in the `FLASH_CRYPT_CNT` field indicates that flash enc is enabled, and even number indicates that flash enc is disabled. For release mode configuration, either all bits in `FLASH_CRYPT_CNT` must be exhausted or field be write protected. In summary, it is possible to switch from development to release mode but care must be taken w.r.t. the EFuse configuration and write protection of different config options.

AvinashSinghPES, 2023-05-02:
@mahavirj  one thing you tell me in case of after Flash Encryption in Release Mode, when we have to update firmware image through OTA. then we have to do plaintext binary image or encrypted binary image ?

mahavirj, 2023-05-02:
@AvinashSinghPES Irrespective of the flash encryption mode, the firmware over OTA update channel is always plaintext. Encryption happens on the device while the image is being written into the device flash storage. Please refer to this [technical post](https://blog.espressif.com/ota-updates-framework-ab5438e30c12) for more details.

AvinashSinghPES, 2023-05-03:
can you explain this , why FLASH_CRYPT_CNT count not increase but all the bit are set   : I choose only release mode Flash Encryption through Espressif IDE . I couldn't understand what happens this.

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                                   Flash encryption key

mahavirj, 2023-05-05:
@AvinashSinghPES Can you please share all details and steps that you are trying out? If you are facing this problem while the bootloader was enabling the flash encryption workflow then it is likely that device was power cycled interim of the encryption process.

mahavirj, 2023-05-15:
@AvinashSinghPES In the above setup, is problem specific to `FLASH_CRYPT_CNT` field only? Are you able to get console log from the custom dev board? Are you able to program other EFuse bits from the custom dev board?

AvinashSinghPES, 2023-05-15:
yeah, problem in specific to FLASH_CRYPT_CNT field only. no I am not able to get console from the custom dev board. not yet, program other EFuse bits from the custom dev board.

kewalmshah, 2023-05-19:
@AvinashSinghPES The above connections will work for console as well as programming.
The simplest way to test the console would be to connect the USB cable to DEV kit and put the other board in download mode as mentioned [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader) You should see the below prints.
```
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
```
Sometimes if this does not work, try interchanging RX and TX pins(since some modules name them interchangeably).

FYI, the image of Development board mentioned does not look like Espressif's `DevKitC`, so not sure if the CP2102 connections are done in a way as they are in `DevKit`

mahavirj, 2023-06-06:
@AvinashSinghPES Closing this issue as it is not directly related to IDF. Please follow suggestions from the above comment to verify the hardware connections.

closed 2023-06-06 by mahavirj