espressif/esp-idf #11309: virtual efuses for programming public key and enabling secure boot  :no bootable partition issue

Anzal3c3, 2023-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to enable secureboot . so inorder to identify the signed firmware the efuses must be programmed with the public key. inorder to do that i generated a  key pair using openssl  and hashed the public key and programmed it using the efuse API s
(i dont know if it is the right approach)

here is the code snippett to write the efuses.

char pub_key[]= "81d07907729111ffddbaf8d1f8043c6e7524b136d105c9fccdddcb3585c0275d";

void app_main(void)
{
    esp_efuse_write_block(EFUSE_BLK4,pub_key,0,sizeof(pub_key));
   
    
}

i also enabled the virtual efuses inorder to not to brick the esp.
i successfully programmed the efuses as i could see that the efuse blocks were stored in the flash using cmd

This is the output log from esp monitor after uploading the public key hash:

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6278,len:0x1b78
load:0x4004c000,len:0xad0
load:0x40050000,len:0x3744
entry 0x4004c1c4
W (20) boot.esp32s2: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (38) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (38) boot: compile time 11:14:17
I (38) boot: chip revision: v0.0
I (41) boot.esp32s2: SPI Speed      : 80MHz
I (45) boot.esp32s2: SPI Mode       : DIO
I (50) boot.esp32s2: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (79) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  3 factory          factory app      00 00 00010000 00100000
I (101) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (109) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (116) boot:  6 nvs_key          NVS keys         01 04 00310000 00001000
W (124) efuse: Loading virtual efuse blocks from flash
EFUSE_BLKx:
0) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
1) 0xa10587e0 0x00007cdf 0x00000000 0x00000000 0x00000000 0x00000000
2) 0xd9400ee8 0x9335d9b8 0x503c694a 0xe2bf2602 0x9da2b990 0x8e93979a 0xcbf4888a 0x96ab719a
3) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4) 0x30643138 0x37303937 0x31393237 0x66663131 0x61626464 0x31643866 0x34303866 0x65366333
5) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
6) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
7) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
8) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
9) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

I (216) boot:  7 efuse_em         efuse            01 05 00311000 00002000
I (223) boot: End of partition table
I (228) boot: Defaulting to factory image
I (232) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=07d90h ( 32144) map
I (247) esp_image: segment 1: paddr=00017db8 vaddr=3ffbd650 size=015f4h (  5620) load
I (251) esp_image: segment 2: paddr=000193b4 vaddr=40022000 size=06c64h ( 27748) load
I (265) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=1381ch ( 79900) map
I (282) esp_image: segment 4: paddr=00033844 vaddr=40028c64 size=049e4h ( 18916) load
I (293) boot: Loaded app from partition at offset 0x10000
I (293) boot: Disabling RNG early entropy source...
I (304) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (305) cpu_start: Pro cpu up.
I (325) cpu_start: Pro cpu start user code
I (325) cpu_start: cpu freq: 160000000 Hz
I (325) cpu_start: Application information:
I (328) cpu_start: Project name:     blink
I (333) cpu_start: App version:      1
I (337) cpu_start: Compile time:     May  2 2023 11:14:02
I (344) cpu_start: ELF file SHA256:  b7e861c8a73d5b27...
I (350) cpu_start: ESP-IDF:          v5.0.1
I (354) cpu_start: Min chip rev:     v0.0
I (359) cpu_start: Max chip rev:     v1.99
I (364) cpu_start: Chip rev:         v0.0
I (369) heap_init: Initializing. RAM available for dynamic allocation: 

now after this i tried uploading another code signed using the corresponding private key. 
the code was built successfully. But the flash monitor gave out some error and i am not sure what it is .

i am attaching that log here:

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6278,len:0x1b78
load:0x4004c000,len:0xad0
load:0x40050000,len:0x3744
entry 0x4004c1c4
W (20) boot.esp32s2: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (38) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (38) boot: compile time 11:14:17
I (38) boot: chip revision: v0.0
I (41) boot.esp32s2: SPI Speed      : 80MHz
I (45) boot.esp32s2: SPI Mode       : DIO
I (50) boot.esp32s2: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (79) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  3 factory          factory app      00 00 00010000 00100000
I (101) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (109) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (116) boot:  6 nvs_key          NVS keys         01 04 00310000 00001000
W (124) efuse: Loading virtual efuse blocks from flash
EFUSE_BLKx:
0) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
1) 0xa10587e0 0x00007cdf 0x00000000 0x00000000 0x00000000 0x00000000
2) 0xd9400ee8 0x9335d9b8 0x503c694a 0xe2bf2602 0x9da2b990 0x8e93979a 0xcbf4888a 0x96ab719a
3) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4) 0x30643138 0x37303937 0x31393237 0x66663131 0x61626464 0x31643866 0x34303866 0x65366333
5) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
6) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
7) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
8) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
9) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

I (216) boot:  7 efuse_em         efuse            01 05 00311000 00002000
I (223) boot: End of partition table
I (228) boot: Defaulting to factory image
I (232) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=07d90h ( 32144) map
I (247) esp_image: segment 1: paddr=00017db8 vaddr=3ffbd650 size=015f4h (  5620) load
I (251) esp_image: segment 2: paddr=000193b4 vaddr=40022000 size=06c64h ( 27748) load
I (265) esp_image: segment 3: paddr=00020020 vaddr=3f000020 size=0a090h ( 41104) map
I (274) esp_image: segment 4: paddr=0002a0b8 vaddr=3ffbe3a0 size=01764h (  5988) load
E (276) esp_image: Checksum failed. Calculated 0xa3 read 0x40
E (281) boot: Factory app partition is not bootable
E (287) esp_image: image at 0x110000 has invalid magic byte (nothing flashed here?)
E (295) boot: OTA app partition slot 0 is not bootable
E (301) esp_image: image at 0x210000 has invalid magic byte (nothing flashed here?)
E (309) boot: OTA app partition slot 1 is not bootable
E (315) boot: No bootable app partitions in the partition table

if somebody can point me in the right direction that will be a huge help. i dont know if this mistake is by me (obviosly its going to be ) or some other reasons as i said  i am a noob . if you could help it would be much appreciated

Anzal3c3, 2023-05-02:
[factory.txt](https://github.com/espressif/esp-idf/files/11372159/factory.txt)

this is the contents of the flash i got using read flash

KonstantinKondrashov, 2023-05-15:
Hi @Anzal3c3!
You use esp32s2, this chip supports secure boot V2. To enable secure boot you do not need to write any code on your own. You need to enable the SECURE boot option in menuconfig, read about it here - https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html#secure-boot-v2. It can work with virtual efuse option so the real efuses will not be burned (for testing ONLY). 

If you want to go on then you can take a look at how it was implemented in bootloader - https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/secure_boot_v2/secure_boot.c#L201, it extracts the boot_key_digests.key_digests from bootloader and writes into efuses.

I see the following issues: char pub_key[]= "81.....
>4.  0x30643138 0x37303937 0x31393237 0x66663131 0x61626464 0x31643866 0x34303866 0x65366333
0x38 - "8"
0x31 - "1"
...

The key was bured as string representation, the secure boot expects byte representation in the efuse block for secure boot.
I see a few issues in your code, see  [esp_efuse_write_block()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html#_CPPv421esp_efuse_write_block17esp_efuse_block_tPKv6size_t6size_t), it requires a byte array instead of string and number of bits instead of bytes.
This function does not set a key purpose for EFUSE_BLK4. So I suggest using the [esp_efuse_write_key()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html#_CPPv419esp_efuse_write_key17esp_efuse_block_t19esp_efuse_purpose_tPKv6size_t), see an example below.

```c
    const uint8_t key_data[32] = {
        1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,
        25,26,27,28,29,30,31,32
    };
    esp_err_t status = esp_efuse_write_key(key_block, ESP_EFUSE_KEY_PURPOSE_SECURE_BOOT_DIGEST0, key_data, sizeof(key_data));
```

One more thing is that you use virtual efuses but each time after reboot you have it empty again. To keep it between reboots use [CONFIG_EFUSE_VIRTUAL_KEEP_IN_FLASH](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-efuse-virtual-keep-in-flash) (see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html#virtual-efuses).

closed 2023-05-22 by KonstantinKondrashov