espressif/esp-idf #11310: uru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

aggaddam, 2023-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-DEVKIT

### Power Supply used.

USB

### What is the expected behavior?

I am getting crash with esp IDF v5.1`Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled` while downloading firmware binary from https server having chunked transfer encoded data. 

This issue was not observed with v4.x.x idf version

### What is the actual behavior?

The crash should not happen during firmware image binary download from server

### Steps to reproduce.

Download the firmware ota image binary from any https server with chunked transfer encoded data.

### Debug Logs.

```plain
D (83541) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:4 posted to loop 0x3fc9c0e0
D (83550) HTTP_CLIENT: data_process=96343, content_length=-1
D (83555) transport_base: remain data in cache, need to read again
D (83562) HTTP_CLIENT: http_on_body 256
D (83565) OTA_CLIENT: [ota_event_handler 110]  Free memory: 185024 bytes 
E (83572) OTA_CLIENT: HTTP_EVENT_ON_DATA, len=256 content len=-1
D (83579) ota_core: Input 256 bytes
D (83582) ota_core: Write 256 total 96599
D (83587) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:4 posted to loop 0x3fc9c0e0
D (83596) HTTP_CLIENT: data_process=96599, content_length=-1
D (83602) transport_base: remain data in cache, need to read again
D (83608) HTTP_CLIENT: http_on_body 256
D (83612) OTA_CLIENT: [ota_event_handler 110]  Free memory: 185024 bytes 
E (83618) OTA_CLIENT: HTTP_EVENT_ON_DATA, len=256 content len=-1
D (83625) ota_core: Input 256 bytes
D (83629) ota_core: Write 256 total 96855
D (83634) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:4 posted to loop 0x3fc9c0e0
D (83642) HTTP_CLIENT: data_process=96855, content_length=-1
D (83648) transport_base: remain data in cache, need to read again
D (83654) HTTP_CLIENT: http_on_body 256
D (83658) OTA_CLIENT: [ota_event_handler 110]  Free memory: 185024 bytes 
E (83665) OTA_CLIENT: HTTP_EVENT_ON_DATA, len=256 content len=-1
D (83671) ota_core: Input 256 bytes
[... 309 lines cut ...]
3fc9f9c0: 0x3fc92438 0x3fce0000 0x3fcb3c04 0x4038ab9a 0x3fcb3c04 0x3fce0000 0x3fca4e80 0x00000000
3fc9f9e0: 0xffffffc7 0x00000000 0x3fc91800 0x40382ee4 0x3fcb3c04 0x3fce0000 0x3fce0000 0x00000000
3fc9fa00: 0x3fcdf934 0x3fce0000 0x3ff1b000 0x3fce0000 0x3fce0000 0x3fce0000 0x000000f5 0x3fce0000
3fc9fa20: 0x3fcb3c64 0x3fce0000 0x3fcb3c04 0x400415ba 0x00000000 0x3fce0000 0x3fce0000 0x3ff1b5a8
3fc9fa40: 0x00000000 0x3fcdf918 0x3fce0000 0x40040902 0x00000000 0x00000000 0x00000011 0x00000000
3fc9fa60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fa80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9faa0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fc9f950 0x3fc942e4
3fc9fac0: 0x3fc8fd00 0x3fc8fd00 0x3fc9fab8 0x3fc8fcf8 0x00000002 0x3fc9d9f0 0x3fc9d9f0 0x3fc9fab8
3fc9fae0: 0x00000000 0x00000017 0x3fc9e0b4 0x69666977 0x4d80be00 0x24844785 0x001855ed 0x00000000
3fc9fb00: 0x3fc9fab0 0x00000017 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc94a84 0x3fc94aec
3fc9fb20: 0x3fc94b54 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x420a352c
3fc9fb40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fb60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fb80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fba0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fbc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fbe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9fc00: 0x00000000 0x00000000 0x3f000000 0x00000020 0x0000f40d 0x00000000 0x00000000 0x00000000
3fc9fc20: 0x00010001 0x1e0000c8 0x3fcb1ff8 0x00000000 0x00000030 0x42047bbc 0x42047b82 0x42047abc
3fc9fc40: 0x42047f4c 0x42047c92 0x42047f7a 0x42047ff4 0x42047b3c 0x42047ac0 0x42047b02 0x00000000
3fc9fc60: 0x3fc9fc2c 0x00000b94 0x6f6d706f 0x00006564 0x2e617473 0x64697373 0x00000000 0x2e617473
3fc9fc80: 0x68747561 0x65646f6d 0x00010000 0x00000000 0x00000003 0x00000002 0x3fc91ca4 0x2e617473
3fc9fca0: 0x64697373 0x00000000 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x00240701
```

ESP-Marius, 2023-05-04:
These kind of errors from the heap component are usually caused by memory corruption, some suggestions:
* See [heap memory debugging](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-corruption-detection) for how to enable some more debugging options.
* It seems like you are running with asserts disabled? Try enabling them and see if they can catch the error at an earlier stage.

If you are able to provide a minimal example which I could run to reproduce this issue?

aggaddam, 2023-05-04:
@ESP-Marius Here is the sample code you can use to replicate the problem.
Make sure the OTA binary image downloaded from the HTTPS server supports the configuration  Transfer-Encoding: chunked for file download

You should see this https server response 

`Transfer-Encoding: chunked`

`static esp_err_t ota_event_handler(esp_http_client_event_t *evt)
{
    int32_t ret               = ESP_OK;
    static int32_t output_len = 0;
    
    if (NULL != evt)
    {
        switch (evt->event_id)
        {
            case HTTP_EVENT_ERROR:
                ESP_LOGD(TAG, "HTTP_EVENT_ERROR");
                break;

            case HTTP_EVENT_ON_CONNECTED:
                ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
                break;

            case HTTP_EVENT_HEADER_SENT:
                ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
                break;

            case HTTP_EVENT_ON_HEADER:
                ESP_LOGD(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
                update_len = 0;
                break;
                
            case HTTP_EVENT_ON_DATA:
              if (!esp_http_client_is_chunked_response(evt->client))
              {
                  // If user_data buffer is configured, copy the response into the buffer
                  output_len += evt->data_len;
                  ESP_LOGD(TAG, "No chunk data: [%d, %d, %lld]",
                                 evt->data_len, output_len, esp_http_client_get_content_length(evt->client));

              }
              else
              {
                  output_len += evt->data_len;
                  ESP_LOGD(TAG, "chunk data: [%d, %d, %lld]",
                                 evt->data_len, output_len, esp_http_client_get_content_length(evt->client));
              }
              break;
              
            case HTTP_EVENT_ON_FINISH:
                ESP_LOGD(TAG, "HTTP_EVENT_ON_FINISH");
                break;
                
            case HTTP_EVENT_DISCONNECTED:
                ESP_LOGD(TAG, "HTTP_EVENT_DISCONNECTED");
                break;
                
            default:
                ESP_LOGD(TAG, "Default case, %d", evt->event_id);
                break;
        }
    }
    return ESP_OK;
}

esp_http_client_config_t  config = {
                                       .url            = "url for .bin image",
                                       .event_handler  = ota_event_handler,
                                       .transport_type = HTTP_TRANSPORT_OVER_SSL,
                                       .cert_pem       = server_root_ca_cert,
                                       .buffer_size    = 256,    
                                    };

esp_http_client_handle_t client = esp_http_client_init(&config);
if (NULL != client)
{
    // GET
    esp_err = esp_http_client_perform(client);
    if (ESP_OK == esp_err)
    {
        int32_t rsp_code = esp_http_client_get_status_code(client);
        ESP_LOGI(TAG, "HTTP GET Status = %d, content_length = %d",
                       rsp_code, esp_http_client_get_content_length(client));

        if (200 == rsp_code)
        {
            ESP_LOGI(TAG,"OTA IMAGE RECEIVED SUCCESSFULLY");
        }
        else
        {
            esp_err = ESP_FAIL;
            ESP_LOGI(TAG,"OTA IMAGE RECEIVED FAILED");
        }
    }
    else
    {
        esp_err = ESP_FAIL;
        ESP_LOGE(TAG, "HTTP GET request failed: %d", esp_err);
    }

    esp_http_client_cleanup(client);
}
else
{
    esp_err = ESP_FAIL;
    ESP_LOGE(TAG, "client handle is null");
}
`

AxelLin, 2023-05-04:
@aggaddam 
I think you shold use esp_http_client_get_chunk_length() for chunk data?

mahavirj, 2023-05-05:
@aggaddam 

- Can you please try [native_ota_example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/native_ota_example) and see if you could recreate this issue?
- We do have chunked encoding scenario covered in our tests, please see [here](https://github.com/espressif/esp-idf/blob/56123c52aaa08f1b53350c7af30c91320b352ef4/examples/system/ota/native_ota_example/pytest_native_ota.py#L118).

aggaddam, 2023-05-12:
@mahavirj  I tried with native example code as well but the same issue is happening with example code as well. 

I just replaced the URL for image download SDK config and associated certificate. After downloading few chunks of data device is crashing assert msg heap mem fail. 

The same error was observed with our project application as well

Here is the log

`assert failed: remove_free_block tlsf.c:330 (prev && "prev_free field can not be null")`

mahavirj, 2023-05-12:
@aggaddam Can you please share the patch that you created on top of native_ota_example?

aggaddam, 2023-05-13:
@mahavirj Attached the code

[native_ota_example.c.txt](https://github.com/espressif/esp-idf/files/11468871/native_ota_example.c.txt)

aggaddam, 2023-05-16:
@mahavirj  Did you get a chance to look into this issue? Are you able to recreate it?

hmalpani, 2023-05-17:
Hello @aggaddam 
I tried out native_ota_example with the patch you shared above. I didn't observe any crashes. The application successfully downloaded the new firmware.

aggaddam, 2023-05-17:
@hmalpani  Have you tried with esp-idf version from branch release/v5.1 ?

hmalpani, 2023-05-17:
@aggaddam 
Yes I tried it on branch `release/v5.1`

AxelLin, 2023-05-17:
@hmalpani 
Did you test https server with **chunked transfer encoded data**?

aggaddam, 2023-05-17:
@hmalpani Could you please share the commit of idf v5.1 used for testing ? I will try with the same commit id.

hmalpani, 2023-05-17:
@AxelLin @aggaddam 
> Did you test https server with chunked transfer encoded data?

 Yes I used chunked server for testing.

> @hmalpani Could you please share the commit of idf v5.1 used for testing ? I will try with the same commit id.

I am on the latest commit present on release/v5.1 branch: 6cfc6f53beb9ef0800e8ec534babc603a8836177

aggaddam, 2023-05-18:
@hmalpani  
I tested the native example code with release/v5.1 branch commit - 6cfc6f53beb9ef0800e8ec534babc603a8836177. I am still seeing the same issue with https server with chunked transfer encoded data. 

I got same assert error after receiving few chunks of data. 

```
assert failed: remove_free_block tlsf.c:330 (prev && "prev_free field can not be null")

Core  0 register dump:
MEPC    : 0x403806f2  RA      : 0x40386c5c  SP      : 0x3fca0210  GP      : 0x3fc90e00  
TP      : 0x3fc745e0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000070  S1      : 0x00000001  A0      : 0x3fca024c  A1      : 0x3fc92051  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc97000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fca0377  
S4      : 0x3fc92050  S5      : 0x3fca6004  S6      : 0x000005fe  S7      : 0x3fca6b18  
S8      : 0x0000001a  S9      : 0x00000000  S10     : 0x3fcb44fa  S11     : 0x3fcb44f0  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x0000000

```

Jacques-Zhao, 2023-05-22:
Hi @aggaddam are you still concerned about this problem? tslf assert usually occurs when a pointer oversteps one byte. I am very interested in debugging this problem. Could you please send me a demo that can reproduce the problem

hmalpani, 2023-05-22:
Hello @aggaddam 
I am unable to reproduce this issue. Can you please use https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html?highlight=heap%20tracing to enable heap debugging to trace the heap corruption?
Can you also share the full logs here?

aggaddam, 2023-05-22:
@Jacques-Zhao @hmalpani Attached is the log for your reference and the native ota example code used.  
Please ensure the server configuration is https with the transfer type as encoded chunked data

[ota_example_https_chunk_data_log.txt](https://github.com/espressif/esp-idf/files/11529034/ota_example_https_chunk_data_log.txt)
[native_ota_example.c.txt](https://github.com/espressif/esp-idf/files/11529215/native_ota_example.c.txt)

aggaddam, 2023-05-26:
@Jacques-Zhao Are you able to recreate the problem that I mentioned?

Jacques-Zhao, 2023-05-29:
@aggaddam  I'm sorry that I can't repeat the problem you mentioned, but I was dealing with a similar problem last week, and I haven't found the root cause yet, but there is a workaround you can try. Just disable hardware AES acceleration(MBEDTLS_HARDWARE_AES) in the menuconfig. Please test with this configuration item and let me know the results.

hmalpani, 2023-05-29:
Hello @aggaddam 
Can you also share sdkconfig file? It might help in reproducing the issue.

aggaddam, 2023-05-29:
@hmalpani Here is the SDK config file 

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/11589139/sdkconfig.txt)

Jacques-Zhao, 2023-05-29:
@aggaddam   Have you tested the method I'm talking about? I have found the root cause of this problem, if you disable AES hardware acceleration can fix the problem, you can try to modiy [lldesc_setup_link_constrained](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/aes/dma/esp_aes.c#L390) change `LLDESC_MAX_NUM_PER_DESC_16B_ALIGNED` to  `LLDESC_MAX_NUM_PER_DESC` and open AES hardware acceleration test again.

mahavirj, 2023-05-29:
@aggaddam 

Just to extend on the @Jacques-Zhao comment, please try following patch and see if it can help to fix your issue:

```
diff --git components/mbedtls/port/aes/dma/esp_aes.c components/mbedtls/port/aes/dma/esp_aes.c
index efc531a55a..ec4f3ae739 100644
--- components/mbedtls/port/aes/dma/esp_aes.c
+++ components/mbedtls/port/aes/dma/esp_aes.c
@@ -371,8 +371,8 @@ static int esp_aes_process_dma(esp_aes_context *ctx, const unsigned char *input,
             return esp_aes_process_dma_ext_ram(ctx, input, output, len, stream_out, input_needs_realloc, output_needs_realloc);
         }
 
-        /* Set up dma descriptors for input and output */
-        lldesc_num = lldesc_get_required_num(block_bytes);
+        /* Set up dma descriptors for input and output considering the 16 byte alignment requirement for EDMA */
+        lldesc_num = lldesc_get_required_num_constrained(block_bytes, LLDESC_MAX_NUM_PER_DESC_16B_ALIGNED);
 
         /* Allocate both in and out descriptors to save a malloc/free per function call */
         block_desc = heap_caps_calloc(lldesc_num * 2, sizeof(lldesc_t), MALLOC_CAP_DMA);
```

hmalpani, 2023-06-02:
Hello @aggaddam 
Does the above patch help you solve the issue?

aggaddam, 2023-06-02:
@hmalpani I didn't get a chance to test this out this week. I will test this change next week and let you know the feedback.

zjykymt, 2023-06-02:
> @aggaddam
>
> Just to extend on the @Jacques-Zhao comment, please try following patch and see if it can help to fix your issue:
>
> ```
> diff --git components/mbedtls/port/aes/dma/esp_aes.c [...]

After applying this fix, a new error appeared, the log is as follows:

```
E (315169) AES: Timed out waiting for completion of AES Interrupt

abort() was called at PC 0x420c1ec2 on core 1
0x420c1ec2: esp_aes_dma_wait_complete at C:/esp/esp-idf-v5.0.2/components/mbedtls/port/aes/dma/esp_aes.c:219 (discriminator 3)

Backtrace: 0x4037637a:0x3fcc8490 0x40383489:0x3fcc84b0 0x4038b0c2:0x3fcc84d0 0x420c1ec2:0x3fcc8540 0x420c2143:0x3fcc8560 0x420c26fe:0x3fcc85b0 0x420c3dae:0x3fcc85e0 0x420c3ec0:0x3fcc8640 0x420c3f07:0x3fcc8670 0x420c3f3f:0x3fcc86b0 0x420b80c9:0x3fcc8710 0x420b87ae:0x3fcc8740 0x420b2a19:0x3fcc8780 0x420b2d56:0x3fcc8860 0x420b343a:0x3fcc8880 0x420b34b0:0x3fcc88c0 0x420b35e2:0x3fcc88e0 0x420ae8b9:0x3fcc8900 0x4211162a:0x3fcc8920 0x4203586b:0x3fcc8940 0x42036836:0x3fcc8970 0x4203740c:0x3fcc8990 0x403871fd:0x3fcc89c0
0x4037637a: panic_abort at C:/esp/esp-idf-v5.0.2/components/esp_system/panic.c:423

0x40383489: esp_system_abort at C:/esp/esp-idf-v5.0.2/components/esp_system/esp_system.c:153

0x4038b0c2: abort at C:/esp/esp-idf-v5.0.2/components/newlib/abort.c:38

0x420c1ec2: esp_aes_dma_wait_complete at C:/esp/esp-idf-v5.0.2/components/mbedtls/port/aes/dma/esp_aes.c:219 (discriminator 3)

0x420c2143: esp_aes_process_dma at C:/esp/esp-idf-v5.0.2/components/mbedtls/port/aes/dma/esp_aes.c:439

0x420c26fe: esp_aes_crypt_ctr at C:/esp/esp-idf-v5.0.2/components/mbedtls/port/aes/dma/esp_aes.c:1036

0x420c3dae: esp_aes_gcm_update at C:/esp/esp-idf-v5.0.2/components/mbedtls/port/aes/esp_aes_gcm.c:478

0x420c3ec0: esp_aes_gcm_crypt_and_tag_partial_hw at C:/esp/esp-idf-v5.0.2/components/mbedtls/port/aes/esp_aes_gcm.c:574

0x420c3f07: esp_aes_gcm_crypt_and_tag at C:/esp/esp-idf-v5.0.2/components/mbedtls/port/aes/esp_aes_gcm.c:697

0x420c3f3f: esp_aes_gcm_auth_decrypt at C:/esp/esp-idf-v5.0.2/components/mbedtls/port/aes/esp_aes_gcm.c:718

0x420b80c9: mbedtls_cipher_aead_decrypt at C:/esp/esp-idf-v5.0.2/components/mbedtls/mbedtls/library/cipher.c:1440

0x420b87ae: mbedtls_cipher_auth_decrypt_ext at C:/esp/esp-idf-v5.0.2/components/mbedtls/mbedtls/library/cipher.c:1582

0x420b2a19: mbedtls_ssl_decrypt_buf at C:/esp/esp-idf-v5.0.2/components/mbedtls/mbedtls/library/ssl_msg.c:1439

0x420b2d56: ssl_prepare_record_content at C:/esp/esp-idf-v5.0.2/components/mbedtls/mbedtls/library/ssl_msg.c:3864

0x420b343a: ssl_get_next_record at C:/esp/esp-idf-v5.0.2/components/mbedtls/mbedtls/library/ssl_msg.c:4791

0x420b34b0: mbedtls_ssl_read_record at C:/esp/esp-idf-v5.0.2/components/mbedtls/mbedtls/library/ssl_msg.c:4029

0x420b35e2: mbedtls_ssl_read at C:/esp/esp-idf-v5.0.2/components/mbedtls/mbedtls/library/ssl_msg.c:5692

0x420ae8b9: esp_mbedtls_read at C:/esp/esp-idf-v5.0.2/components/esp-tls/esp_tls_mbedtls.c:222

0x4211162a: esp_tls_conn_read at C:/esp/esp-idf-v5.0.2/components/esp-tls/esp_tls.c:100

0x4203586b: nfp_task_ssl_recv_process at C:/Users/YMT/Documents/GitHub/HW-S3-Project/eFG/1_APP/main/netif/u_netif_poll.c:2518

0x42036836: nfp_task_socket_recv_loop at C:/Users/YMT/Documents/GitHub/HW-S3-Project/eFG/1_APP/main/netif/u_netif_poll.c:2554

0x4203740c: vTaskNetifPoll at C:/Users/YMT/Documents/GitHub/HW-S3-Project/eFG/1_APP/main/netif/u_netif_poll.c:2723

0x403871fd: vPortTaskWrapper at C:/esp/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: fd64b9b46b3e7840

Rebooting...
```

zjykymt, 2023-06-03:
> @aggaddam I'm sorry that I can't repeat the problem you mentioned, but I was dealing with a similar problem last week, and I haven't found the root cause yet, but there is a workaround you can try. [...]

My test method is to use HTTPS GET to download the bin file (128K bytes * 13) multiple times in a row. It seems that "**Just disable hardware AES acceleration (MBEDTLS_HARDWARE_AES)**" can solve my problem ( **assert failed: remove_free_block tlsf.c:330 (prev && "prev_free field can not be null")** ). The conclusion of this test is for reference.

Thanks.

AxelLin, 2023-06-05:
@mahavirj 

Your change in https://github.com/espressif/esp-idf/issues/11310#issuecomment-1566891892 is different from https://github.com/espressif/esp-idf/issues/11310#issuecomment-1566642037.

I'm not sure which one is correct.
Could you clarify if both places need change?

i.e. The diff would be below if apply both changes.

```
diff --git a/components/mbedtls/port/aes/dma/esp_aes.c b/components/mbedtls/port/aes/dma/esp_aes.c
index efc531a55a..a57cb7a4ae 100644
--- a/components/mbedtls/port/aes/dma/esp_aes.c
+++ b/components/mbedtls/port/aes/dma/esp_aes.c
@@ -371,8 +371,8 @@ static int esp_aes_process_dma(esp_aes_context *ctx, const unsigned char *input,
             return esp_aes_process_dma_ext_ram(ctx, input, output, len, stream_out, input_needs_realloc, output_needs_realloc);
         }
 
-        /* Set up dma descriptors for input and output */
-        lldesc_num = lldesc_get_required_num(block_bytes);
+        /* Set up dma descriptors for input and output considering the 16 byte alignment requirement for EDMA */
+        lldesc_num = lldesc_get_required_num_constrained(block_bytes, LLDESC_MAX_NUM_PER_DESC_16B_ALIGNED);
 
         /* Allocate both in and out descriptors to save a malloc/free per function call */
         block_desc = heap_caps_calloc(lldesc_num * 2, sizeof(lldesc_t), MALLOC_CAP_DMA);
@@ -387,7 +387,7 @@ static int esp_aes_process_dma(esp_aes_context *ctx, const unsigned char *input,
 
         lldesc_setup_link(block_in_desc, input, block_bytes, 0);
         //Limit max inlink descriptor length to be 16 byte aligned, require for EDMA
-        lldesc_setup_link_constrained(block_out_desc, output, block_bytes, LLDESC_MAX_NUM_PER_DESC_16B_ALIGNED, 0);
+        lldesc_setup_link_constrained(block_out_desc, output, block_bytes, LLDESC_MAX_NUM_PER_DESC, 0);
 
         out_desc_tail = &block_out_desc[lldesc_num - 1];
     }
```

AxelLin, 2023-06-06:
> @AxelLin
> 
> The version of the patch I had posted in my comment was correct. I am also attaching the full version of the fix here which includes a test case that can easily recreate the failure [aes_fix_for_DMA_descriptors_calculations.tar.gz](https://github.com/espressif/esp-idf/files/11653166/aes_fix_for_DMA_descriptors_calculations.tar.gz)
> 

Confirm the patch fixes the panic on ESP32C3.
Thanks, @mahavirj

closed 2023-06-08 by espressif-bot

aggaddam, 2023-06-28:
@mahavirj I applied the patch, and now the device is not crashing while downloading the image from the https server but there is still a problem while downloading the image from the HTTPS server with chunked transfer encoded data. The issue is after downloading a few packets, I am not seeing a callback event for HTTP_EVENT_ON_DATA and finally ended up receiving partial data instead of full data

I tried with the native ota example and HTTP streaming example. Same behavior observed 

Attached log file for your reference
[ota_log.txt](https://github.com/espressif/esp-idf/files/11898036/ota_log.txt)

AxelLin, 2023-07-04:
v5.1 does not include this fix. @mahavirj

fpgax, 2023-11-02:
Is this a security vulnerability?