espressif/esp-idf #11313: root节点如何切换？

Leadrive, 2023-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

组网是用esp_mesh_fix_root(true);无网关模式，允许多个root存在。
1、root A和root B，都是通过WIFI连上互联网，轮流把A, B断开、接上，子节点搜索大约几十秒后，可以自动切换根节点；
2、root A通过Wifi连公网，root B通过以太网连公网，轮流把A, B断开、接上，子节点无法自动切换根节点；子节点调用esp_restart()后，可以连上根节点。

尝试过mwifi_restart(), esp_mesh_disconnect(), esp_mesh_connect(), 都无法连上新的根节点。
请教一下，有什么办法切换新的根节点？

谢谢！

I want to know how to switch layer 2 nodes to another root, when the current root is down or disappear.
I tried more methods, but it did not work ok.
So I must restart the layer 2 nodes to swtich to another root.

I used esp_mesh_fix_root(true); make a routerless mesh.
In the mesh, maybe include some root(wifi or ethernet or usb modem), if the current root is down, I want the layer2 can swtich to another automatic.

I tested it.
1. When the root A and root B are wifi mode to connect the Internet, A down, the layer 2 node can switch to B.
2. When the root A is wifi, root B is ethernet, A down, the layer 2 node cannot switch to B, if restart layer 2 node, it can connect to B.

I tried the esp_mesh_disconnect(), esp_mesh_connect(), esp_mesh_stop(), esp_mesh_start(), it did not work ok.
How to switch root auto?

zhangyanjiaoesp, 2023-05-06:
@Leadrive 

> 组网是用esp_mesh_fix_root(true);无网关模式，允许多个root存在。
> 1、root A和root B，都是通过WIFI连上互联网，轮流把A, B断开、接上，子节点搜索大约几十秒后，可以自动切换根节点；

这个描述是说 root A 断开Wi-Fi 连接的时候，子节点会自动断开与 root A 的连接，然后连接到 root B；rooB 断开 Wi-Fi 连接，root A 连接上 Wi-Fi，子节点会自动断开与 root B 的连接，然后连接到 root A 吗？

> 2、root A通过Wifi连公网，root B通过以太网连公网，轮流把A, B断开、接上，子节点无法自动切换根节点；子节点调用esp_restart()后，可以连上根节点。

这个描述是说 root A 断开Wi-Fi 连接的时候，子节点自动断开无法连接到 root B？子节点的行为是什么？是一直尝试寻找新的 root 吗？

你可以提供一些相关的 log 吗？或者可以提供可复现的 demo 吗？

zhangyanjiaoesp, 2023-05-06:
@Leadrive 请参考相关的文档，看是否对你有帮助：https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/esp-wifi-mesh.html#id23

Leadrive, 2023-05-11:
> @Leadrive
>
>
> 这个描述是说 root A 断开Wi-Fi 连接的时候，子节点会自动断开与 root A 的连接，然后连接到 root B；rooB 断开 Wi-Fi 连接，root A 连接上 Wi-Fi，子节点会自动断开与 root B 的连接，然后连接到 root A 吗？
>
>
> 这个描述是说 root A 断开Wi-Fi 连接的时候，子节点自动断开无法连接到 root [...]

谢谢回复。
其实很容易复现的，就是用esp-idf带的no-route例子。
一个root改为wifi连外网，一个改为通过ethernet连外网，
轮流断开、连接root A, root B，子节点无法自动连另外的root。

我测试的结果，如果两个root都是通过wifi连外网，子节点是可以自动切换的，
这就是有疑问的地方，因为mesh部分不开源，我也不知道原因在哪。

Leadrive, 2023-05-11:
多次尝试连不上，调用esp_restart()，重启后是可以连的，
个人猜测，这中间有个什么设置，影响子节点寻找新的root。

这个功能，其实挺重要的，一个组网中，如果备有多个root，其中一个
上不了网，如果能自动切换到另外的root，对用户的影响是最小的。

Leadrive, 2023-05-11:
> @Leadrive 请参考相关的文档，看是否对你有帮助：https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/esp-wifi-mesh.html#id23

这文档看了，没找出有哪些需要注意的地方。我没做多余的操作，就是用的mdf例子，mwifi_init, mwifi_start。
我估计你们平常，也全都是用wifi来组网，wifi+ethernet+4g的混合组网，可能没实际测试过。

Leadrive, 2023-05-11:
@zhangyanjiaoesp 我做了个demo，https://gitee.com/luocanqi/ip_internal_network
自己测试验证了，确实有问题，麻烦你有时间帮看看。谢谢

Leadrive, 2023-05-11:
修改说明：
代码拷贝自 esp_idf/examples/mesh/ip_internal_network
主要修改是添加了ethernet的支持，手工指定root/node节点

测试说明：

1. 采用esp32模块，idf 4.4.4
2. 运行idf.py menuconfig，选择 "Mesh Support Ethernet" 根据自己的情况，设置以太网的其它选项（重要）。编译后生成支持以太网的root，烧到板子上
3. 运行idf.py menuconfig，选择 "Mesh Support Wifi" 注意：不要再选 "Mesh Support Ethernet" 并且设置 路由器的ssid, password 编译后生成支持WIFI的root，烧到板子上
4. 运行idf.py menuconfig，上面两项，都不要选择，编译后生成是node，烧到板子上
5. 三块板子启动后，node可能会连任意一个root，此时轮流断开、重连正在连接的root，node不会自动连到另外的root
6. 如果node节点重启，会自动连到另外的root。

node连不上时的报错信息
````
I (36552) mesh: 2004<arm>parent monitor, my layer:2(cap:6)(node), interval:347092ms, retries:3<>
I (38662) [mesh_main, 89]: Received Routing table [0] 58:bf:25:38:d2:c8
I (38662) [mesh_main, 89]: Received Routing table [1] 94:b5:55:f8:fa:fc
W (39732) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (40002) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (40272) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (40552) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (40822) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (41092) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
~W (43012) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (43282) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (43552) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (43832) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (44102) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (44372) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (44652) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
N$HeW (46562) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (46832) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (47112) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (47382) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (47652) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (47932) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (48202) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
W (48472) wifi:(->sleep)busy, waked:1, dream:0, sleep:0
I (48662) wifi:bcn_timeout,ap_probe_send_start
I (48662) wifi:pm stop, total sleep time: 24412722 us / 40069489 us

I (48662) wifi:(mesh)pm stop, sleep:23 secs, total:38 secs, current active set:10, running:0.61
I (51172) wifi:ap_probe_send over, resett wifi status to disassoc
I (51172) wifi:state: run -> init (c800)
I (51182) wifi:<ba-del>idx
I (51182) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (51182) mesh: [wifi]disconnected reason:200(beacon timeout), continuous:1/max:12, non-root, vote(,)<><>
I (51182) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:200
I (51192) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
E (51202) mesh_netif: Send with err code 16395 ESP_ERR_MESH_DISCONNECTED
I (51202) wifi:state: init -> auth (b0)
I (52712) wifi:state: auth -> init (200)
I (52722) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (52722) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2
I (52722) mesh: [wifi]disconnected reason:2(auth expire), continuous:2/max:12, non-root, vote(,)<><>
I (52732) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (52742) wifi:state: init -> auth (b0)
I (54242) wifi:state: auth -> init (200)
I (54242) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (54242) mesh: [wifi]disconnected reason:2(auth expire), continuous:3/max:12, non-root, vote(,)<><>
I (54242) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2
I (54252) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (54262) wifi:state: init -> auth (b0)
I (55762) wifi:state: auth -> init (200)
I (55762) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (55772) mesh: [wifi]disconnected reason:2(auth expire), continuous:4/max:12, non-root, vote(,)<><>
I (55772) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2
I (55772) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (55782) wifi:state: init -> auth (b0)
I (57292) wifi:state: auth -> init (200)
I (57292) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (57292) mesh: [wifi]disconnected reason:2(auth expire), continuous:5/max:12, non-root, vote(,)<><>
I (57292) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2
I (57302) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (57312) wifi:state: init -> auth (b0)
I (58812) wifi:state: auth -> init (200)
I (58812) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (58812) mesh: [wifi]disconnected reason:2(auth expire), continuous:6/max:12, non-root, vote(,)<><>
I (58822) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2
I (58822) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (58832) wifi:state: init -> auth (b0)
I (60342) wifi:state: auth -> init (200)
I (60342) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (60342) mesh: [wifi]disconnected reason:2(auth expire), continuous:0/max:12, non-root, vote(,)<><>
I (60342) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2
I (60342) mesh: [auth expire]looking for a new parent, [L:2]try layer:1
I (60442) [mesh_main, 346]: <MESH_EVENT_NETWORK_STATE>is_rootless:1
I (61042) mesh: [SCAN][ch:1]AP:8, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<>
I (61042) mesh: [FAIL][1]root:0, fail:1, normal:0

I (61652) mesh: [SCAN][ch:1]AP:8, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<>
I (61652) mesh: [FAIL][2]root:0, fail:2, normal:0

I (62262) mesh: [SCAN][ch:1]AP:6, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<>
I (62262) mesh: [FAIL][3]root:0, fail:3, normal:0
````

后面就是一直SCAN，计数到一定程度，出现PARENT_NOT_FOUND, PARENT_DISCONNECTED消息

Leadrive, 2023-05-11:
两个root的时候，子节点的启动信息如下：
````
I (5646) mesh: find root:ESPM_38D2C8, root_cap:1(max:300), new channel:1, old channel:0
I (5646) mesh: find root:ESPM_30FC9C, root_cap:1(max:300), new channel:11, old channel:1
I (5646) mesh: [FIND][ch:0]AP:43, otherID:0, MAP:2, idle:0, candidate:0, root:2[00:00:00:00:00:00][FIXED-ROOT]
I (5656) mesh: [FIND:1]find a network, channel:1, cfg<channel:0, router:, 00:00:00:00:00:00>

I (5666) wifi:mode : sta (c8:f0:9e:2e:19:9c) + softAP (c8:f0:9e:2e:19:9d)
W (5676) wifi:<MESH AP>adjust channel:1, secondary channel offset:1(40U)
I (5676) wifi:Total power save buffer number: 16
I (5666) [mesh_main, 364]: <MESH_EVENT_FIND_NETWORK>new channel:1, router BSSID:00:00:00:00:00:00
I (5996) mesh: [SCAN][ch:1]AP:12, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (5996) mesh: 7369[selection]try rssi_threshold:-78, backoff times:0, max:5<-78,-82,-85>
I (6006) mesh: [DONE]connect to parent:ESPM_38D2C8, channel:1, rssi:-67, 58:bf:25:38:d2:c9[layer:1, assoc:0], my_vote_num:0/voter_num:0, rc[00:00:00:00:00:00/-120/0]
I (6106) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (6626) wifi:state: init -> auth (b0)
I (6636) wifi:state: auth -> assoc (0)
I (6646) wifi:state: assoc -> run (10)
I (6646) mesh: <MESH_NWK_MIE_CHANGE><><><><ROOT ADDR><><><>
I (6656) mesh: <MESH_NWK_ROOT_ADDR>from assoc, layer:2, root_addr:58:bf:25:38:d2:c9, root_cap:1
I (6656) mesh: <MESH_NWK_ROOT_ADDR>idle, layer:2, root_addr:58:bf:25:38:d2:c9, conflict_roots.num:0<>
I (6666) wifi:connected with ESPM_38D2C8, aid = 1, channel 1, 40U, bssid = 58:bf:25:38:d2:c9
I (6676) wifi:security: WPA2-PSK, phy: bgn, rssi: -67
I (6656) [mesh_main, 374]: unknown id:24
I (6736) [mesh_main, 289]: <MESH_EVENT_ROOT_ADDRESS>root address:58:bf:25:38:d2:c9
I (6756) wifi:Set ps type: 1

I (6756) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (6756) mesh: 2004<arm>parent monitor, my layer:2(cap:6)(node), interval:10381ms, retries:1<normal connected>
I (6766) mesh_main: <MESH_EVENT_PARENT_CONNECTED>layer:0-->2, parent:58:bf:25:38:d2:c9<layer2>, ID:77:77:77:77:77:76
I (6766) mesh: [scan]new scanning time:300ms, beacon interval:100ms
I (6776) mesh_netif: It was a wifi station removing stuff
I (6796) wifi:<ba-add>idx:0 (ifx:0, 58:bf:25:38:d2:c9), tid:5, ssn:0, winSize:64
I (6796) [mesh_main, 322]: <MESH_EVENT_TODS_REACHABLE>state:0
I (6796) [mesh_main, 289]: <MESH_EVENT_ROOT_ADDRESS>root address:58:bf:25:38:d2:c9
I (6836) mesh: [scan]new scanning time:600ms, beacon interval:300ms
I (8316) [mesh_main, 89]: Received Routing table [0] 58:bf:25:38:d2:c8
I (8326) [mesh_main, 89]: Received Routing table [1] c8:f0:9e:2e:19:9c
I (9176) wifi:pm start, type: 1
````
提炼了几个信息：
````
I (5646) mesh: find root:ESPM_38D2C8, root_cap:1(max:300), new channel:1, old channel:0
I (5646) mesh: find root:ESPM_30FC9C, root_cap:1(max:300), new channel:11, old channel:1
I (5646) mesh: [FIND][ch:0]AP:43, otherID:0, MAP:2, idle:0, candidate:0, root:2[00:00:00:00:00:00][FIXED-ROOT]
I (5656) mesh: [FIND:1]find a network, channel:1, cfg<channel:0, router:, 00:00:00:00:00:00>
I (5666) [mesh_main, 364]: <MESH_EVENT_FIND_NETWORK>new channel:1, router BSSID:00:00:00:00:00:00
I (5996) mesh: [SCAN][ch:1]AP:12, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (5996) mesh: 7369[selection]try rssi_threshold:-78, backoff times:0, max:5<-78,-82,-85>
I (6006) mesh: [DONE]connect to parent:ESPM_38D2C8, channel:1, rssi:-67, 58:bf:25:38:d2:c9[layer:1, assoc:0], my_vote_num:0/voter_num:0, rc[00:00:00:00:00:00/-120/0]
````
设置的channel=0，连上第一个root，channel自动转为了1。
那个待连接的root，channel=11。

当第一个root断电后，子节点不断在搜索channel 1。
估计就是在这里，一直无法切换到第二个root。
````
I (48186) mesh: [SCAN][ch:1]AP:9, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<>
I (48186) mesh: [FAIL][2]root:0, fail:2, normal:0
````

有什么办法，让子节点能重新进入搜索root？
我试了esp_mesh_set_self_organized，esp_mesh_connect都不行。

Leadrive, 2023-05-12:
Root A, Root B，都是采用wifi连外网，但是连的不同的ssid，轮流断开root A, B时，子节点也无法自动切换。

如果A, B，连的是相同的ssid，channel是一样的，子节点就可以自动切换。
````
I (1435) [mesh_main, 201]: <MESH_EVENT_MESH_STARTED>ID:77:77:77:77:77:76
I (1435) [mesh_main, 511]: mesh starts successfully, heap:187124, root fixed
                                                                                                                                                                                                                                   
I (5635) mesh: find root:ESPM_30FC9C, root_cap:1(max:300), new channel:1, old channel:0
I (5645) mesh: find root:ESPM_F8FAFC, root_cap:1(max:300), new channel:1, old channel:1
I (5645) mesh: [FIND][ch:0]AP:40, otherID:0, MAP:2, idle:0, candidate:0, root:2[00:00:00:00:00:00][FIXED-ROOT]
I (5655) mesh: [FIND:1]find a network, channel:1, cfg<channel:0, router:, 00:00:00:00:00:00>
````
都是channel 1
应该就是这个channel的关系，请问子节点如何自动切换channel？mesh配置的channel是0，不知道这个mesh是如何改变channel。
@zhangyanjiaoesp

zhangyanjiaoesp, 2023-05-15:
@Leadrive 我会根据你的 demo 测试一下，然后尽快回复你，谢谢

zhangyanjiaoesp, 2023-05-17:
@Leadrive 
Please replace the wifi libs under the `components/esp_wifi/lib/esp32` folder with the attached contents.
[wifi_lib_0517.zip](https://github.com/espressif/esp-idf/files/11493784/wifi_lib_0517.zip)
**wifi firmware version: 3b859ec**

I have tested locally, after changing, the node can change root with different channel. Here is my test log:
[node_log_0517.zip](https://github.com/espressif/esp-idf/files/11493793/node_log_0517.zip)

By the way, the calling of `esp_mesh_set_type(MESH_NODE)` doesn't need, this will return error `I (871) mesh: [CONFIG]fail to set node, err:0x4008`

[image]

Leadrive, 2023-05-18:
@zhangyanjiaoesp 测试了ethernet+wifi+wifi，3个根节点，一个子节点，能顺利切换，正是我想要的结果。非常感谢。
后续我再在各种情况下多测试一下。
能否再编译一个4.4.4, S3版本，我再测试一下4G模式下的。

zhangyanjiaoesp, 2023-05-18:
@Leadrive 附件包含了4.4.4 下面支持的所有芯片的 wifi lib，wifi firmware version: 3b859ec.

[wifi_lib_0518.zip](https://github.com/espressif/esp-idf/files/11505022/wifi_lib_0518.zip)

Leadrive, 2023-05-31:
@zhangyanjiaoesp 
mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:98, no_wnd_count:0, timeout_count:0
目前是碰到这个问题，节点有时候一直出这个消息，连不上根结点。我测试了，另外一个子节点是正常的，根节点也是正常，就这个子节点，一直报这个信息。超过12小时，还是不行。
如果重启，就可以正常连上。

省电模式，根节点和子节点，都改成了0

zhangyanjiaoesp, 2023-06-02:
@Leadrive 
当node 节点想要发送数据到 root 的节点的时候，会检查本地的发送窗口 （xon window）是否为 0 ， 如果为 0 就要发送 xon request 给 root 节点，请求分配新的窗口。当 root 节点在指定时间内没有回复 node 节点时，node 节点就会打印 `mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:98, no_wnd_count:0, timeout_count:0` 这个log。应该是节点一直没有连上根节点，才会有这样的 log。

> 省电模式，根节点和子节点，都改成了0，都开了蓝牙blufi，是在已成功连网后，才设置的ps=0

如果从一开始就设置 ps=0 会出现这样的情况吗？

Leadrive, 2023-06-02:
@zhangyanjiaoesp 因为要开蓝牙，所以一开始要设置为1，否则报错重启。
你们一直没解决wifi+ble coexist 要设置ps=1的问题。
现在只能是等连好了，再改ps=0，但测试中，偶尔也会ps=0报错重启。

是子节点连上了根节点，但在某种情况下，又断开了，然后就不断报这个错误。
又没什么标志性错误，只能检查，如果一段时间还是连不上，只好esp_restart

zhangyanjiaoesp, 2023-06-02:
@Leadrive 
那我们可以看一下是什么情况下连接断开了，为什么没有办法重连。你那边有更详细的log吗？或者你可以提供复现的方法吗？

Leadrive, 2023-06-02:
@zhangyanjiaoesp 
[error.zip](https://github.com/espressif/esp-idf/files/11631952/error.zip)
这是其中的一种情况，这个设备是一开始就没正常连上，它是layer3，可能它连的layer2目前不正常。
更多的情况是，一开始连上的，在工作时，突然掉了，然后就一直连不上。这个我再观察，稍后提供日志。

另外有个疑问，如果节点当前没有正常连接，如果禁止有节点要连它？

Leadrive, 2023-06-02:
[error2.txt](https://github.com/espressif/esp-idf/files/11632094/error2.txt)
@zhangyanjiaoesp 这是运行5个小时后，突然连不上了，报错信息又不同。

Leadrive, 2023-06-02:
[error3.txt](https://github.com/espressif/esp-idf/files/11633256/error3.txt)
@zhangyanjiaoesp 
这个日志是启动后，做了一次ps=0，但后面不知道在哪个环节，又被置为ps=1（不是我程序置的，可能是esp的其它模块）
然后就一直连不上

[error4.txt](https://github.com/espressif/esp-idf/files/11633264/error4.txt)
这个，其它都没动，就是改为定时10秒ps=0，然后就顺利连上了。

个人总结，可能是跟ps有关，父、子节点，这个ps要一致才行。

但这里有个问题，就是wifi+ble coexist，要开启ble，必须设置ps=1，其实我不希望ps=1，因为休眠会影响wifi的连接。
但在wifilib中，会时不时报错说ps!=1，并且会导致重启，所以问题比较严重。

你们给出的意见是，先初始化wifi，ps=1，再开启ble，再wifi_start，但这个，并不是保险的做法。

mwifi中有个关闭省电模式，但我现在没用mdf，用的是wifi mesh

Leadrive, 2023-06-02:
[error5.txt](https://github.com/espressif/esp-idf/files/11633342/error5.txt)
@zhangyanjiaoesp 

这个是虽然设ps=0，但始终连不上

Leadrive, 2023-06-02:
[error6.txt](https://github.com/espressif/esp-idf/files/11633454/error6.txt)
@zhangyanjiaoesp 
这个是最典型的了，重启后连上，也能跑业务了，过不了几秒，就开始报seqno的错误，然后就一直这样

Leadrive, 2023-06-03:
````
I (11329) wifi:Set ps type: 0

E (11329) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!

abort() was called at PC 0x42117633 on core 0
0x42117633: pm_set_sleep_type at ??:?
                                                                                                                                                                                                                                   

Backtrace: 0x40375d8e:0x3fcf53c0 0x40382b0d:0x3fcf53e0 0x40388ec2:0x3fcf5400 0x42117633:0x3fcf5470 0x4210606c:0x3fcf5490 0x42104452:0x3fcf54b0 0x4214a275:0x3fcf54d0
0x40375d8e: panic_abort at /home/develop/esp/esp-idf/components/esp_system/panic.c:408
                                                                                                                                                                                                                                   
0x40382b0d: esp_system_abort at /home/develop/esp/esp-idf/components/esp_system/esp_system.c:137
                                                                                                                                                                                                                                   
0x40388ec2: abort at /home/develop/esp/esp-idf/components/newlib/abort.c:46
                                                                                                                                                                                                                                   
0x42117633: pm_set_sleep_type at ??:?
                                                                                                                                                                                                                                   
0x4210606c: wifi_set_ps_process at ??:?
                                                                                                                                                                                                                                   
0x42104452: ieee80211_ioctl_process at ??:?
                                                                                                                                                                                                                                   
0x4214a275: ppTask at ??:?
````

Leadrive, 2023-06-03:
ps异常，有些板子不报错，有些板子就一直报错，原因不明。
把这个设置取消，就不报异常。

Leadrive, 2023-06-03:
[error7.txt](https://github.com/espressif/esp-idf/files/11641652/error7.txt)
@zhangyanjiaoesp 这个是启动后一直没连上的日志

Leadrive, 2023-06-03:
ps，非root，在启动mesh后设置不会报异常。root节点，等连上外网，时间同步后再设置，就不报错了。

zhangyanjiaoesp, 2023-06-05:
@Leadrive 

> 另外有个疑问，如果节点当前没有正常连接，如果禁止有节点要连它？

`esp_mesh_post_toDS_state(bool reachable)` 可以把节点的连网状态发送给子节点，但是这个 API 只适用于 root 节点

> I (11329) wifi:Set ps type: 0
E (11329) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!
>
> abort() was called at PC 0x42117633 on core 0
0x42117633: pm_set_sleep_type at ??:?                                                                                                                                                                                                          
>
> Backtrace: 0x40375d8e:0x3fcf53c0 0x40382b0d:0x3fcf53e0 0x40388ec2:0x3fcf5400 0x42117633:0x3fcf5470 0x4210606c:0x3fcf5490 0x42104452:0x3fcf54b0 0x4214a275:0x3fcf54d0
0x40375d8e: panic_abort at /home/develop/esp/esp-idf/components/esp_system/panic.c:408
> 0x40382b0d: esp_system_abort at /home/develop/esp/esp-idf/components/esp_system/esp_system.c:137
 >                                                                                                                                                                                                                                  
> 0x40388ec2: abort at [...]

这个 `abort`  是在 log `E (11329) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!` 之后主动调用的。

Leadrive, 2023-12-13:
@zhangyanjiaoesp 这个lib包，好像没更新到4.4.6正式版本上，我测试是没能重连上。

zhangyanjiaoesp, 2023-12-13:
@Leadrive The v4.4.6 already contains the corresponding changes https://github.com/espressif/esp-idf/commit/8756d9545f2208ce4a56718eceea6f4d91b27502
[image]

Leadrive, 2024-01-09:
@zhangyanjiaoesp 我测试了，4.4.6版本，root在以太网和WIFI之间切换时，node都无法自动连。
用你上面提供的4.4.4的lib，就可以正常自动连。

zhangyanjiaoesp, 2024-01-25:
> 附件包含了4.4.4 下面支持的所有芯片的 wifi lib，wifi firmware version: 3b859ec.
> 
> [wifi_lib_0518.zip](https://github.com/espressif/esp-idf/files/11505022/wifi_lib_0518.zip)

@Leadrive 再次检查了 wifi code，这个 wifi lib 和 4.4.6 中 https://github.com/espressif/esp-idf/commit/8756d9545f2208ce4a56718eceea6f4d91b27502 的改动完全一致。你在测试 4.4.6 的时候，和 4.4.4 的测试环境和配置完全一样吗？测试 code  是否一致？ node 无法自动重连的现象是什么？有相关的log吗？

Leadrive, 2024-01-26:
最近访问不了github，只能在邮件里回复了。
如果需要直接联系我，请发邮件 ***@***.***

esp-idf用的是4.4.6最新版本，从 gitee下载的。
两个root设备，配的同一个meshId，一个是RootA，esp32+以太网，一个是RootB，esp32s3+wifi
node设备是 esp32s3

附件日志，是 node连的RootA，此时手工断开RootA电源，node无法自动连到RootB
手工重启node，能顺利连上RootB

这个问题，在半年前的回贴中，也说过，当时你发了lib后，测试是root可以切换的。

  

 
  
@Leadrive 再次检查了 wifi code，这个 wifi lib 和 4.4.6 中 8756d95 的改动完全一致。你在测试 4.4.6 的时候，和 4.4.4 的测试环境和配置完全一样吗？测试 code  是否一致？ node 无法自动重连的现象是什么？有相关的log吗？

Leadrive, 2024-01-26:
@zhangyanjiaoesp 
没测试你提供的lib，用的是gitee上的4.4.6
就算不做root切换，光是root关掉，过一会再开，
node也无法再连上，一直在报这个错误
W (1203021) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:23, no_wnd_count:0, timeout_count:639
这个错误，至少从2018年开始，就一直存在，到今天还是不能很好地解决。

非常影响芯片在工程项目的实施。
因为无法连上，又不能频繁重启设备。

Leadrive, 2024-01-26:
还是上面的配置，这次加了
    esp_mesh_send_block_time(5000);
    esp_err_t ret=esp_mesh_start();

断开RootB，node始终无法连上RootA，如果重新开启RootB，node则会自动连上RootB
估计是跟RootA是esp32, RootB是esp32s3有关，之前也反馈过这问题，你提供的lib，我测试了也是可以切换的，
但4.4.6中的lib，还是无法切换

不报上面的错误 W (82158948) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1800 ms timeout, seqno:0, xseqno:27860, no_wnd_count:0, timeout_count:44

改报这个错误
I (67090) mesh: [SCAN][ch:6]AP:6, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<&gt;

Leadrive, 2024-01-27:
esp_mesh_send_block_time(5000);
    esp_mesh_set_xon_qsize(128);
    esp_err_t ret=esp_mesh_start();

加了这两句，粗略测试了，目前可以切换。

wifi_lib_0518.zip 这个库，mesh在重连时，会导致 multi heap异常，原因不明，应该不是我程序原因，换了4.4.6没出现这种情况。
所以没再测试这个 lib

Leadrive, 2024-02-21:
> @Leadrive
> 
> > 另外有个疑问，如果节点当前没有正常连接，如果禁止有节点要连它？
> 
> `esp_mesh_post_toDS_state(bool reachable)` 可以把节点的连网状态发送给子节点，但是这个 API 只适用于 root 节点

@zhangyanjiaoesp 请教一下，root节点，检测到自己联不上互联网后，想断开所有的子节点，禁止子节点连上来，该如何操作？
如果检测到可以联网后，再想打开root节点供子节点连接，该如何写？

````
        mesh_netifs_destroy();
        esp_mesh_deinit();
//        mesh_netifs_stop();
        esp_mesh_stop();
````
mesh is not started

zhangyanjiaoesp, 2024-03-04:
@Leadrive root 节点可以调用 `esp_mesh_post_toDS_state(bool reachable)` API 把节点的连网状态发送给子节点，子节点可以根据这个状态决定要不要去连接 root。root 节点去 deinit/stop mesh 不是好的选择。

Leadrive, 2024-03-15:
@zhangyanjiaoesp 现在碰到最大的问题是，node与root，大约10秒就断开一次，然后就是重连。
日志见附件，里面的正常通讯字样，是我省略了不相关的内容。
大约10秒断一次，root节点是esp32+ethernet, node节点是esp32s3
麻烦帮看看问题所在，idf版本是最新的4.4.6，程序是 ip_internal_network 改的
[mesh_log.txt](https://github.com/espressif/esp-idf/files/14617012/mesh_log.txt)

Leadrive, 2024-03-16:
ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
是否此值有影响？子node有每隔3秒发一次心跳包，按理有数据通讯的。

Leadrive, 2024-03-18:
可能是我的程序导致的，还在测试观察中。

zhangyanjiaoesp, 2024-04-23:
@Leadrive Can this issue be closed?

closed 2024-04-23 by Leadrive