espressif/esp-idf #11314: Maximum number of client connections through LAN8720

banbeiduyao, 2023-05-02:
### Is your feature request related to a problem?

通过 LAN8720做客户端 连接服务端最大可以连接几个?
How many connections can the server connect to the client through LAN8720?

kostaond, 2023-05-05:
@banbeiduyao could you please be more specific about your question? I am not sure if I fully understand it. Could you please describe your scenario in details?

kostaond, 2023-06-19:
@banbeiduyao could you please provide more information about your issue as requested previously? Otherwise, I will have to close the issue for lack of feedback. Thank you for understanding.

kostaond, 2023-07-25:
I'm closing the issue due to lack of feedback.

closed 2023-07-25 by kostaond