espressif/esp-idf #11315: v4.4.x - Intermittent crashes with BLE + Classic + Light Sleep (RTCWDT_RTC_RESET)

expresspotato, 2023-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

LyraT v4.3 | Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

Hello,

We're running on ESP32 here on the development board, LyraT v4.3 along with a custom board developed around the pico-v3-02. With light sleep enabled, neither board is able to maintain both a BLE & Bluetooth Classic A2DP connection without crashing with `RTCWDT_RTC_RESET`.

Disabling the interrupt watchdog timer via sdkconfig `CONFIG_ESP_INT_WDT=n` had no effect , clearly the CPU is in fact locked up when exiting light sleep, as disabling any wdt protect via commenting out portions of `components/esp_hw_support/sleep_modes.c` just cause the group timer (`TG1WDT`) to timeout instead moments later.

Something definitely feels wrong exiting from light sleep, as without light sleep the problem does not occur, even after hours of extended testing (8+ hours). The only close ticket I could find is [this one](https://github.com/espressif/esp-idf/issues/8184), where the op was trying attempt to disable to watchdog to solve issues related to BLE & WiFi at the same time and noted their project still crashes with `RTCWDT_RTC_RESET` when its disabled via sdkconfig / menuconfig -> bootloader.

Downgrading to v4.3.5 seems to make the problem go away (> 120 minutes no crashes) but I would like to be off a branch that will not receive any updates starting in December 2023, but cannot do so without it being rock solid. v.4.4.3 also exhibits the same issue, other v4.4 versions might.

### What is the actual behavior?

Crashes with RTCWDT_RTC_RESET anywhere from 5 to 120 minutes later.

### Steps to reproduce.

Untested these exact steps, since I'm too busy to share a large portion of source code right now this might work:
1. Download the coex example
2. Adjust settings to match MCLK / WS / DSOUT, etc.
3. Enable light sleep through menuconfig
4. Flash to a compatible board with a 32.768 kHz crystal attached
5. Connect to the device using nRF connect for BLE
6. Connect to the device using your normal process from iOS / Android (only tested with my S22+)
7. Crash eventually

### More Information.

- Tried a fresh `sdkconfig`
- Tried multiple LytraT v4.3 boards
- Tried custom board
- Tried v4.4.3 and v4.4.4

esp-lis, 2023-05-04:
@expresspotato Add the following code below this line and test it.
`REG_SET_FIELD(RTC_CNTL_BIAS_CONF_REG, RTC_CNTL_DBG_ATTEN, 3);`

expresspotato, 2023-12-05:
I've added some additional `ets_printf` statements and it appears the CPU does come out of light sleep, but then locks up. Can anyone at espressif comment on this issue? It's critical for us to have BLE & Light Sleep.

`
SWW2W3W4W5W6W7W8W9SWW2W3W4W5W6W7W8W9Sets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7004
ho 0 tail 12 room 4
load:0x40078000,len:16364
load:0x40080400,len:4552
0x40080400: _init at ??:?

entry 0x400806bc
`

`
diff --git a/components/esp_hw_support/sleep_modes.c b/components/esp_hw_support/sleep_modes.c
index ea8fb5b557..24794acf17 100644
--- a/components/esp_hw_support/sleep_modes.c
+++ b/components/esp_hw_support/sleep_modes.c
@@ -88,7 +88,7 @@
 #define VDD_SDIO_POWERUP_TO_FLASH_READ_US 700
 
 // Cycles for RTC Timer clock source (internal oscillator) calibrate
-#define RTC_CLK_SRC_CAL_CYCLES      (10)
+#define RTC_CLK_SRC_CAL_CYCLES      (16)
 
 #ifdef CONFIG_IDF_TARGET_ESP32
 #define DEFAULT_CPU_FREQ_MHZ                CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ
@@ -663,19 +663,25 @@ static esp_err_t esp_light_sleep_inner(uint32_t pd_flags,
                                        rtc_vddsdio_config_t vddsdio_config)
 {
     // Enter sleep
+    ets_printf("S");
     uint32_t reject = esp_sleep_start(pd_flags);
+    ets_printf("W");
 
     // If VDDSDIO regulator was controlled by RTC registers before sleep,
     // restore the configuration.
     if (vddsdio_config.force) {
         rtc_vddsdio_set_config(vddsdio_config);
     }
+    
+    ets_printf("W2");
 
     // If SPI flash was powered down, wait for it to become ready
     if (pd_flags & RTC_SLEEP_PD_VDDSDIO) {
         // Wait for the flash chip to start up
         esp_rom_delay_us(flash_enable_time_us);
     }
+    
+    ets_printf("W3");
 
     return reject ? ESP_ERR_SLEEP_REJECT : ESP_OK;
 }
@@ -696,7 +702,7 @@ static inline bool can_power_down_vddsdio(const uint32_t vddsdio_pd_sleep_durati
     return (s_config.pd_options[ESP_PD_DOMAIN_VDDSDIO] == ESP_PD_OPTION_OFF) ? force_pd : safe_pd;
 }
 
-esp_err_t esp_light_sleep_start(void)
+esp_err_t IRAM_ATTR esp_light_sleep_start(void)
 {
     s_config.ccount_ticks_record = cpu_ll_get_cycle_count();
     static portMUX_TYPE light_sleep_lock = portMUX_INITIALIZER_UNLOCKED;
@@ -820,7 +826,7 @@ esp_err_t esp_light_sleep_start(void)
     bool wdt_was_enabled = wdt_hal_is_enabled(&rtc_wdt_ctx);    // If WDT was enabled in the user code, then do not change it here.
     if (!wdt_was_enabled) {
         wdt_hal_init(&rtc_wdt_ctx, WDT_RWDT, 0, false);
-        uint32_t stage_timeout_ticks = (uint32_t)(1000ULL * rtc_clk_slow_freq_get_hz() / 1000ULL);
+        uint32_t stage_timeout_ticks = 65535; //(uint32_t)(1000ULL * rtc_clk_slow_freq_get_hz() / 1000ULL);
         wdt_hal_write_protect_disable(&rtc_wdt_ctx);
         wdt_hal_config_stage(&rtc_wdt_ctx, WDT_STAGE0, stage_timeout_ticks, WDT_STAGE_ACTION_RESET_RTC);
         wdt_hal_enable(&rtc_wdt_ctx);
@@ -844,6 +850,8 @@ esp_err_t esp_light_sleep_start(void)
 
     // light sleep wakeup flag only makes sense after a successful light sleep
     s_light_sleep_wakeup = (err == ESP_OK);
+    
+    ets_printf("W4");
 
     // FRC1 has been clock gated for the duration of the sleep, correct for that.
 #ifdef CONFIG_IDF_TARGET_ESP32C3
@@ -866,21 +874,36 @@ esp_err_t esp_light_sleep_start(void)
      * for very small values of sleep_duration. Ignore those to keep esp_timer
      * monotonic.
      */
+    
+    ets_printf("W5");
+    
     if (time_diff > 0) {
         esp_timer_private_advance(time_diff);
     }
+    
     esp_set_time_from_rtc();
+    
+    ets_printf("W6");
 
     esp_clk_private_unlock();
     esp_timer_private_unlock();
     esp_ipc_isr_release_other_cpu();
+    
+    ets_printf("W7");
+    
     if (!wdt_was_enabled) {
         wdt_hal_write_protect_disable(&rtc_wdt_ctx);
         wdt_hal_disable(&rtc_wdt_ctx);
         wdt_hal_write_protect_enable(&rtc_wdt_ctx);
     }
+    
+    ets_printf("W8");
+    
     portEXIT_CRITICAL(&light_sleep_lock);
     s_config.sleep_time_overhead_out = (cpu_ll_get_cycle_count() - s_config.ccount_ticks_record) / (esp_clk_cpu_freq() / 1000000ULL);
+    
+    ets_printf("W9");
+    
     return err;
 }
`

expresspotato, 2023-12-05:
Updated to 5.1.2 and the issue still occurs. Any ideas?

`
I (10311082) LIBERTY: 8 Bit --> 4052095
I (10311082) LIBERTY: Internal --> 43607
I (10311092) LIBERTY: SPI RAM --> 4008488
I (10311092) LIBERTY: Largest free block in Internal --> 38900
I (10311102) LIBERTY: Largest free block in SPI RAM --> 3997684
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7428
load:0x40078000,len:17100
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:4296
entry 0x40080668
I (162) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (163) boot: compile time Dec  5 2023 13:16:28
I (163) boot: Multicore bootloader
I (166) boot: chip revision: v3.0
I (170) qio_mode: Enabling default flash chip QIO
I (175) boot.esp32: SPI Speed      : 80MHz
I (180) boot.esp32: SPI Mode       : QIO
I (185) boot.esp32: SPI Flash Size : 8MB
I (189) boot: Enabling RNG early entropy source...
I (195) boot: Partition Table:
I (198) boot: ## Label            Usage          Type ST Offset   Length
I (206) boot:  0 phy_init         RF data          01 01 0000a000 00001000
I (213) boot:  1 nvs_keys         NVS keys         01 04 0000b000 00001000
I (221) boot:  2 nvs              WiFi data        01 02 0000c000 00008000
I (228) boot:  3 nvs_factory      WiFi data        01 02 00014000 00008000
I (236) boot:  4 nvs_system       WiFi data        01 02 0001c000 00020000
I (244) boot:  5 nvs_user         WiFi data        01 02 0003c000 00040000
I (251) boot:  6 otadata          OTA data         01 00 0007c000 00002000
I (259) boot:  7 ota_0            OTA app          00 10 00080000 0028a000
I (266) boot:  8 ota_1            OTA app          00 11 00310000 0028a000
I (274) boot:  9 spiffs_factory   Unknown data     01 82 0059a000 000c0000
I (281) boot: 10 spiffs_system    Unknown data     01 82 0065a000 000c0000
I (289) boot: 11 spiffs_user      Unknown data     01 82 0071a000 000c0000
I (296) boot: 12 coredump         Unknown data     01 03 007da000 00018000
I (304) boot: 13 efuse            efuse            01 05 007f2000 00002000
I (312) boot: 14 emul_efuse       efuse            01 05 007f4000 00002000
I (319) boot: End of partition table
I (323) esp_image: segment 0: paddr=00080020 vaddr=3f400020 size=60f08h (397064) map
I (441) esp_image: segment 1: paddr=000e0f30 vaddr=3ff80063 size=00008h (     8) load
I (441) esp_image: segment 2: paddr=000e0f40 vaddr=3ffbdb60 size=08d10h ( 36112) load
I (458) esp_image: segment 3: paddr=000e9c58 vaddr=40080000 size=063c0h ( 25536) load
I (467) esp_image: segment 4: paddr=000f0020 vaddr=400d0020 size=1824f4h (1582324) map
I (901) esp_image: segment 5: paddr=0027251c vaddr=400863c0 size=213d8h (136152) load
I (946) esp_image: segment 6: paddr=002938fc vaddr=400c0000 size=00064h (   100) load
I (967) boot: Loaded app from partition at offset 0x80000
I (967) boot: Disabling RNG early entropy source...
I (979) cpu_start: Multicore app
I (979) quad_psram: This chip is ESP32-D0WD
I (979) esp_psram: Found 8MB PSRAM device
I (981) esp_psram: Speed: 40MHz
I (985) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (992) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (1001) cpu_start: Pro cpu up.
I (1005) cpu_start: Starting app cpu, entry point is 0x40081eac
0x40081eac: call_start_cpu1 at /Users/kevin/esp/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.

`

esp-lis, 2023-12-11:
> I've updated to 4.4.6 and the problem remains after varying times of being connected... Is there a better way to debug the RTC Watchdog reset? Because the bug takes so long to reproduce, I'm hoping that someone at @espressif knows what this could be.
> 
> I've found another user on the forums is also experiencing the same issue after updating to the 4.4.x branch. I'm on a pico-v3-02: https://www.esp32.com/viewtopic.php?t=20316
> 
> After 338 minutes of being connected.

I suggest modify [RTC_CNTL_XTL_BUF_WAIT_SLP_US](https://github.com/espressif/esp-idf/blob/v5.1.2/components/soc/esp32/include/soc/rtc.h#L539) to 2000, and modify [FREERTOS_IDLE_TIME_BEFORE_SLEEP](https://github.com/espressif/esp-idf/blob/v5.1.2/components/freertos/Kconfig#L256) to 6 at the same time.

expresspotato, 2023-12-18:
It still crashes... And unfortunately both those changes slightly narrow the available light sleep.

This bug is 100% reproducible... Do you want me to zip you a sample project?

esp-wzh, 2023-12-28:
@expresspotato  Sorry for the late reply,

> This bug is 100% reproducible... Do you want me to zip you a sample project?

Of course, I would be much grateful if you could provide a project that can reproduce the error.

AhmedSalem636, 2024-01-05:
Hi, I am facing the exact same issue. Following your discussion for a possible solution

hoping you can find a solution for this issue

AhmedSalem636, 2024-01-07:
Hi, I am zipping and dropping a simple project that causes this issue.
[toESP_test.zip](https://github.com/espressif/esp-idf/files/13852385/toESP_test.zip)

Wish we can find a solution for this issue.

esp-wzh, 2024-01-07:
@AhmedSalem636 Thank you for the information! I reproduced your problem, but I don't think the problem you're experiencing is the same thing as what this thread reports.

The Arduino SDK only provides the basic lightsleep interface, once it is called, the Bluetooth hardware state will be corrupted, and it is also not recommended to go to sleep without deinit Bluetooth, which may be the cause of the problem, refer [Wi-Fi/Bluetooth and Sleep Modes](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#wi-fi-bluetooth-and-sleep-modes).

So, if you want to implement lightsleep when using Bluetooth, you probably have to use esp-idf, which provides a [power management component](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html) that allows bluetooth services and system sleep to co-exist,  you can refer the [nimble powersave example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/power_save)

AhmedSalem636, 2024-01-10:
> @AhmedSalem636 Thank you for the information! I reproduced your problem, but I don't think the problem you're experiencing is the same thing as what this thread reports.
>
> The Arduino SDK only [...]

thanks a lot for your reply and the info. Also sorry if I confused between topics. I am downloading and reading about how to use ESP_IDF

AhmedSalem636, 2024-01-25:
Hi,

I have used the esp-idf and I ran into this same issue of light-sleep crashing with ble.

did you find a solution for this above mentioned issue?

xiongweichao, 2024-01-29:
@AhmedSalem636 Please provide idf commit ID and steps to reproduce. Thanks!

xiongweichao, 2024-01-31:
Hi @AhmedSalem636 

Calling `esp_light_sleep_start` is not allowed when Bluetooth is enabled. Because Bluetooth may still be working, this will cause the status of some registers to be incorrect.

If you want to control the sleep time, you need to deinitialize Bluetooth. Reinitialize Bluetooth when exiting light sleep.

Thanks

AhmedSalem636, 2024-02-02:
thanks for the info, I went for the automatic light-sleep, and it works fine so far.

xiongweichao, 2024-02-23:
@expresspotato May I ask if the problem still exists?

expresspotato, 2024-02-23:
Yes I beleive so, I haven't had a chance to migrate back upwards from v4.3.7 on this large and buys project.

xiongweichao, 2024-02-28:
@expresspotato  Could you please tell me what the CPU frequency you are using? I will try to reproduce this issue, thanks!
`pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED`

expresspotato, 2024-02-28:
240 Mhz, 200 Hz or 100 Hz (no difference) tick rate, PM enabled. Attached sdkconfig.defaults

[sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/14428737/sdkconfig.defaults.txt)

closed 2024-07-01 by Sherry616