espressif/esp-idf #11316: lcd _st7789v unable to drive

yushuaijun, 2023-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

st7789v screen can not use st7789 to drive the screen, what is the solution.

suda-morris, 2023-05-04:
@yushuaijun Thanks for reporting.  But have you read the issue template when you open this issue? The issue template will kindly guide you on what information you should prepare and what alternative solutions might help you. If you open the issue in this way, I'm afraid we can't get enough information to help you to continue your debugging.

AFAICT, I see no problem using the st7789 driver in the esp-idf if you follow the initialization that is documented in our example.

yushuaijun, 2023-05-04:
@suda-morris Thanks for the reply, I have debugged it now。There is no lcd_init_cmd_t in the st7789 driver under idf, and then I added it in the driver of st7789 through the format of ili9341, and it is ok.

atanisoft, 2023-05-04:
@yushuaijun Can you share the modification you did so it can be incorporated into a proper fix?

yushuaijun, 2023-05-04:
> typedef struct {
    uint8_t cmd;
    uint8_t data[16];
    uint8_t data_bytes; // Length of data in above data array; 0xFF = end of cmds.
} lcd_init_cmd_t;

static const lcd_init_cmd_t vendor_specific_init[] = {
    /* Memory Data Access Control, MX=MV=1, MY=ML=MH=0, RGB=0 */
    {0x36, {0x00}, 1},
    /* Interface Pixel Format, 16bits/pixel for RGB/MCU interface */
    {0x3A, {0x55}, 1},
    /* Porch Setting */
    {0xB2, {0x0c, 0x0c, 0x00, 0x33, 0x33}, 5},
    /* Gate Control, Vgh=13.65V, Vgl=-10.43V */
    {0xB7, {0x45}, 1},
    /* VCOM Setting, VCOM=1.175V */
    {0xBB, {0x2B}, 1},
    /* LCM Control, XOR: BGR, MX, MH */
    {0xC0, {0x2C}, 1},
    /* VDV and VRH Command Enable, enable=1 */
    {0xC2, {0x01, 0xff}, 2},
    /* VRH Set, Vap=4.4+... */
    {0xC3, {0x11}, 1},
    /* VDV Set, VDV=0 */
    {0xC4, {0x20}, 1},
    /* Frame Rate Control, 60Hz, inversion=0 */
    {0xC6, {0x0f}, 1},
    /* Power Control 1, AVDD=6.8V, AVCL=-4.8V, VDDS=2.3V */
    {0xD0, {0xA4, 0xA1}, 1},
    /* Positive Voltage Gamma Control */
    {0xE0, {0xD0, 0x00, 0x05, 0x0E, 0x15, 0x0D, 0x37, 0x43, 0x47, 0x09, 0x15, 0x12, 0x16, 0x19}, 14},
    /* Negative Voltage Gamma Control */
    {0xE1, {0xD0, 0x00, 0x05, 0x0D, 0x0C, 0x06, 0x2D, 0x44, 0x40, 0x0E, 0x1C, 0x18, 0x16, 0x19}, 14},
    /* Sleep Out */
    {0x11, {0}, 0x80},
    /* Display On */
    {0x29, {0}, 0x80},
    {0, {0}, 0xff}
};
static esp_err_t panel_st7789v_init(esp_lcd_panel_t *panel)
{
    st7789v_panel_t *st7789v = __containerof(panel, st7789v_panel_t, base);
    esp_lcd_panel_io_handle_t io = st7789v->io;

    // LCD goes into sleep mode and display will be turned off after power on reset, exit sleep mode first
    esp_lcd_panel_io_tx_param(io, LCD_CMD_SLPOUT, NULL, 0);
    vTaskDelay(pdMS_TO_TICKS(100));
    esp_lcd_panel_io_tx_param(io, LCD_CMD_MADCTL, (uint8_t[]) {
        st7789v->madctl_val,
    }, 1);
    esp_lcd_panel_io_tx_param(io, LCD_CMD_COLMOD, (uint8_t[]) {
        st7789v->colmod_cal,
    }, 1);

    // vendor specific initialization, it can be different between manufacturers
    // should consult the LCD supplier for initialization sequence code
    int cmd = 0;
    while (vendor_specific_init[cmd].data_bytes != 0xff) {
        esp_lcd_panel_io_tx_param(io, vendor_specific_init[cmd].cmd, vendor_specific_init[cmd].data, vendor_specific_init[cmd].data_bytes & 0x1F);
        cmd++;
    }

    return ESP_OK;
}
### 
The above is a modification I added in st7789 @atanisoft

suda-morris, 2023-05-04:
@yushuaijun Thanks for sharing the code. AFAIK, most of the values defined in the table are default values.
```c
static const lcd_init_cmd_t vendor_specific_init[] = {
/* Memory Data Access Control, MX=MV=1, MY=ML=MH=0, RGB=0 /
{0x36, {0x00}, 1},
/ Interface Pixel Format, 16bits/pixel for RGB/MCU interface /
{0x3A, {0x55}, 1},
/ Porch Setting /
{0xB2, {0x0c, 0x0c, 0x00, 0x33, 0x33}, 5},
/ Gate Control, Vgh=13.65V, Vgl=-10.43V /
{0xB7, {0x45}, 1},
/ VCOM Setting, VCOM=1.175V /
{0xBB, {0x2B}, 1},
/ LCM Control, XOR: BGR, MX, MH /
{0xC0, {0x2C}, 1},
/ VDV and VRH Command Enable, enable=1 /
{0xC2, {0x01, 0xff}, 2},
/ VRH Set, Vap=4.4+... /
{0xC3, {0x11}, 1},
/ VDV Set, VDV=0 /
{0xC4, {0x20}, 1},
/ Frame Rate Control, 60Hz, inversion=0 /
{0xC6, {0x0f}, 1},
/ Power Control 1, AVDD=6.8V, AVCL=-4.8V, VDDS=2.3V /
{0xD0, {0xA4, 0xA1}, 1},
/ Positive Voltage Gamma Control /
{0xE0, {0xD0, 0x00, 0x05, 0x0E, 0x15, 0x0D, 0x37, 0x43, 0x47, 0x09, 0x15, 0x12, 0x16, 0x19}, 14},
/ Negative Voltage Gamma Control /
{0xE1, {0xD0, 0x00, 0x05, 0x0D, 0x0C, 0x06, 0x2D, 0x44, 0x40, 0x0E, 0x1C, 0x18, 0x16, 0x19}, 14},
/ Sleep Out /
{0x11, {0}, 0x80},
/ Display On */
{0x29, {0}, 0x80},
{0, {0}, 0xff}
};
```
Some of the common commands having a dedicated control function. e.g. the display on command: `0x29` is just what the function `esp_lcd_panel_disp_on_off` does.

What's more, some configurations are LCD module-specific, like the gamma control, different manufacturers may set different ones, that is why we didn't include all of them in the st7789 driver but use the default values of the LCD. But esp-idf leaves the flexibility for you to set extra configurations after the general initialization. The general steps can be:

```c
esp_lcd_panel_reset(panel_handle);
esp_lcd_panel_init(panel_handle);
esp_lcd_panel_invert_color(panel_handle, true);
// extra configurations e.g. gamma control
esp_lcd_panel_io_tx_param(io_handle, GAMMA_CMD, (uint8_t[]) {
       GAMMA_ARRAY
    }, N);
// turn on display
esp_lcd_panel_disp_on_off(panel_handle, true);
```

huming2207, 2023-05-05:
Maybe a bit off-topic to this, but I remember there's a 1.54-inch ST7789V panel I've used earlier actually requires different VCOM (or VGH/VGL voltages, can't remember the details) and gamma LUT. I think it might be safer to let users use their own `lcd_init_cmd_t[]` as inappropriate VCOM/VGL/VGH may kill the panel, or at least end up in some irreversible ghosting.

closed 2023-05-06 by yushuaijun