espressif/esp-idf #11319: Error when using inttypes.h for PRImacros for esp_err_t

aggaddam, 2023-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-DEVKIT

### Power Supply used.

USB

### What is the expected behavior?

Unsilenced the warning for -Wno-format and replaced %d with %"PRId32" for esp_err_t and size_t. It should work with out typecast

### What is the actual behavior?

Unsilenced the warning for -Wno-format and replaced %d with %"PRId32" for esp_err_t. But still, I am getting build errors with the below codes

Example 1: Getting error
#include <inttypes.h>
void app_main(void)
{
    esp_err_t esp_error = ESP_OK;
    ESP_LOGD(TAG, "esp_err = %"PRId32"", esp_err);
}

Example 2: No error after typecast to int32_t
#include <inttypes.h>
void app_main(void)
{
    esp_err_t esp_error = ESP_OK;
    ESP_LOGD(TAG, "esp_err = %"PRId32"", (int32_t)esp_err);
}

Example 3: Getting error
#include <inttypes.h>
void app_main(void)
{
    size_t required_size = 0; 
    ESP_LOGD(TAG, "required_size = %"PRIu32"", required_size);  
}

Example 4: No error
#include <inttypes.h>
void app_main(void)
{
    size_t required_size = 0; 
    ESP_LOGD(TAG, "required_size = %"PRIu32"", (uint32_t)required_size);  
}

What data types are defined for esp_err_t and size_t in esp idf v5.1? Does the application need to typecast for these values to fix the build errors?

### Steps to reproduce.

1. Step
2. Step
3. Step
...

igrr, 2023-05-03:
I think `esp_err_t` is defined as `int` in all IDF versions, so it doesn't need the PRI- (exact width) integer format specifiers in this case. You can simply print it as `%d`.

igrr, 2023-05-03:
Sorry if the recommendation in the docs wasn't formulated clearly. We don't recommend replacing all the format specifiers with the PRI-based ones.

inttypes.h specifiers should only be used for [fixed-width integer types](https://en.cppreference.com/w/c/types/integer). See the "Format macro constants" section over at that page, various PRI* macros are used for different types defined in stdint.h, such as uint32_t, uintmax_t, intptr_t, and so on.

For other types, refer to the [documentation of printf family of functions](https://en.cppreference.com/w/c/io/fprintf). For example, for `size_t` the format specifier should be `%zu`.

One note regarding `%zu` is that when "nano" formatting implementation is used in Newlib (CONFIG_NEWLIB_NANO_FORMAT=y), there is no support for `z` format modifier. For that reason, we don't use `z` modifier a lot in IDF, since we generally have to stay compatibly with "nano" formatting implementation. If you aren't using it in your project, though, you can use `z` modifier for `size_t`, which is better for portability.

AxelLin, 2023-05-04:
> One note regarding `%zu` is that when "nano" formatting implementation is used in Newlib (CONFIG_NEWLIB_NANO_FORMAT=y), there is no support for `z` format modifier. For that reason, we don't use `z` modifier a lot in IDF, since we generally have to stay compatible with "nano" formatting implementation. If you aren't using it in your project, though, you can use `z` modifier for `size_t`, which is better for portability.

So you are aware that this issue needs fix: https://github.com/espressif/esp-idf/issues/9631

igrr, 2023-05-04:
@AxelLin yes, we are aware of that one and a few others.

0xjakob, 2023-05-12:
@aggaddam We clarified the migration guide a bit. It will be available on master soon.

zikalino, 2023-05-25:
closing considering migration guide was udated

closed 2023-05-25 by zikalino